evcc-io/evcc #12444: battery mode: modbus: exception '4' (server device failure), function '16'

StevieC121176, 2024-02-24:
### Describe the bug

Seit der Version  (0.124.4 (fe8ec4fb)) bekomme ich Modbus Fehler.
bei eingeschalteter Batteriesteuerung:  battery mode: modbus: exception '4' (server device failure), function '16'
bei ausgeschalteter Batteriesteuerung: modbus: exception '4' (server device failure), function '16'
An meiner Konfiguration habe ich nichts geändert. wechsle ich auf Version 0.124.4 (cd21eb15) bekomme ich keine Fehlermeldung.
Mein System: Kostal WR und BYD Batterie

### Steps to reproduce

evcc starten, Fehler kommen sofort

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxx

interval: 3s # control cycle interval

sponsortoken: xxxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: battery  
  modbus: tcpip
  capacity: 12.8 
  minsoc: 20 
  maxsoc: 100  
  watchdog: 60s 
  name: battery2
- type: template
  template: kostal-ksem 
  id: 71  
  host: 192.168.178.116  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid3
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.117  
  port: 1502  
[... 473 lines cut ...]
[modbus] TRACE 2024/02/24 16:44:55 modbus: send 00 09 00 00 00 06 47 03 9c 87 00 69
[modbus] TRACE 2024/02/24 16:44:55 modbus: recv 00 09 00 00 00 d5 47 03 d2 80 00 00 50 00 db 00 a2 ff fe 80 00 5b 59 5a 7b 5b 95 80 00 80 00 80 00 80 00 ff fe 13 82 ff fe 00 01 00 04 00 0d ff f1 00 01 00 57 00 0e 00 2c ff de 00 01 ff a9 ff f2 ff d6 ff e2 00 01 00 15 01 33 01 2a 01 cd ff fd ff b3 4e 00 ff e1 d7 cf ff e5 f1 7a ff dd 33 6f 00 17 a5 c8 00 0d 65 c0 00 11 08 e2 00 07 88 6d 00 00 ff 94 7d 91 ff db 81 4b ff e1 9c 3a ff d8 a6 4f 00 34 58 41 00 15 b1 35 00 28 2f 22 00 09 aa 37 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 00 00 00 00 00 00
[modbus] TRACE 2024/02/24 16:44:55 modbus: send 00 0a 00 00 00 06 47 03 9c 87 00 69
[modbus] TRACE 2024/02/24 16:44:55 modbus: recv 00 0a 00 00 00 d5 47 03 d2 80 00 00 50 00 db 00 a2 ff fe 80 00 5b 59 5a 7b 5b 95 80 00 80 00 80 00 80 00 ff fe 13 82 ff fe 00 01 00 04 00 0d ff f1 00 01 00 57 00 0e 00 2c ff de 00 01 ff a9 ff f2 ff d6 ff e2 00 01 00 15 01 33 01 2a 01 cd ff fd ff b3 4e 00 ff e1 d7 cf ff e5 f1 7a ff dd 33 6f 00 17 a5 c8 00 0d 65 c0 00 11 08 e2 00 07 88 6d 00 00 ff 94 7d 91 ff db 81 4b ff e1 9c 3a ff d8 a6 4f 00 34 58 41 00 15 b1 35 00 28 2f 22 00 09 aa 37 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 00 00 00 00 00 00
[modbus] TRACE 2024/02/24 16:44:55 modbus: send 00 0b 00 00 00 06 47 03 9c 87 00 69
[modbus] TRACE 2024/02/24 16:44:55 modbus: recv 00 0b 00 00 00 d5 47 03 d2 80 00 00 50 00 db 00 a2 ff fe 80 00 5b 59 5a 7b 5b 95 80 00 80 00 80 00 80 00 ff fe 13 82 ff fe 00 01 00 04 00 0d ff f1 00 01 00 57 00 0e 00 2c ff de 00 01 ff a9 ff f2 ff d6 ff e2 00 01 00 15 01 33 01 2a 01 cd ff fd ff b3 4e 00 ff e1 d7 cf ff e5 f1 7a ff dd 33 6f 00 17 a5 c8 00 0d 65 c0 00 11 08 e2 00 07 88 6d 00 00 ff 94 7d 91 ff db 81 4b ff e1 9c 3a ff d8 a6 4f 00 34 58 41 00 15 b1 35 00 28 2f 22 00 09 aa 37 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 00 00 00 00 00 00
[modbus] TRACE 2024/02/24 16:44:55 modbus: send 00 0c 00 00 00 06 47 03 9c 87 00 69
[modbus] TRACE 2024/02/24 16:44:55 modbus: recv 00 0c 00 00 00 d5 47 03 d2 80 00 00 50 00 db 00 a2 ff fe 80 00 5b 59 5a 7b 5b 95 80 00 80 00 80 00 80 00 ff fe 13 82 ff fe 00 01 00 04 00 0d ff f1 00 01 00 57 00 0e 00 2c ff de 00 01 ff a9 ff f2 ff d6 ff e2 00 01 00 15 01 33 01 2a 01 cd ff fd ff b3 4e 00 ff e1 d7 cf ff e5 f1 7a ff dd 33 6f 00 17 a5 c8 00 0d 65 c0 00 11 08 e2 00 07 88 6d 00 00 ff 94 7d 91 ff db 81 4b ff e1 9c 3a ff d8 a6 4f 00 34 58 41 00 15 b1 35 00 28 2f 22 00 09 aa 37 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 00 00 00 00 00 00
[modbus] TRACE 2024/02/24 16:44:55 modbus: send 00 0d 00 00 00 06 47 03 9c 87 00 69
[modbus] TRACE 2024/02/24 16:44:55 modbus: recv 00 0d 00 00 00 d5 47 03 d2 80 00 00 50 00 db 00 a2 ff fe 80 00 5b 59 5a 7b 5b 95 80 00 80 00 80 00 80 00 ff fe 13 82 ff fe 00 01 00 04 00 0d ff f1 00 01 00 57 00 0e 00 2c ff de 00 01 ff a9 ff f2 ff d6 ff e2 00 01 00 15 01 33 01 2a 01 cd ff fd ff b3 4e 00 ff e1 d7 cf ff e5 f1 7a ff dd 33 6f 00 17 a5 c8 00 0d 65 c0 00 11 08 e2 00 07 88 6d 00 00 ff 94 7d 91 ff db 81 4b ff e1 9c 3a ff d8 a6 4f 00 34 58 41 00 15 b1 35 00 28 2f 22 00 09 aa 37 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 00 00 00 00 00 00
Power:          10W
Energy:         1549.8kWh
Current L1..L3: 0.8A 2.19A 1.62A
Power L1..L3:   40W 130W -150W

pi@raspberrypi:~ $
```

### What type of operating system are you running?

Linux

### Version

0.124.4 (fe8ec4fb)

StevieC121176, 2024-02-24:
```
● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2024-02-24 16:21:52 CET; 16min ago
   Main PID: 3264 (evcc)
      Tasks: 11 (limit: 3933)
        CPU: 22.844s
     CGroup: /system.slice/evcc.service
             └─3264 /usr/bin/evcc

Feb 24 16:38:18 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:38:18 charge voltages: [234 232 232]V
Feb 24 16:38:18 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:38:18 detected connected phases: 3p
Feb 24 16:38:18 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:38:18 charge currents: [0 0 0]A
Feb 24 16:38:18 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:38:18 charge total import: 4704.800kWh
Feb 24 16:38:18 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:38:18 charger status: B
Feb 24 16:38:19 raspberrypi evcc[3264]: [site  ] ERROR 2024/02/24 16:38:19 battery mode: modbus: exception '4' (server device failure), function '16'
Feb 24 16:38:19 raspberrypi evcc[3264]: [site  ] DEBUG 2024/02/24 16:38:19 ----
Feb 24 16:38:19 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:38:19 charge power: 0W
Feb 24 16:38:19 raspberrypi evcc[3264]: [lp-2  ] DEBUG 2024/02/24 16:38:19 charge power: 0W
Feb 24 16:38:20 raspberrypi evcc[3264]: [site  ] DEBUG 2024/02/24 16:38:20 pv power: 466W
pi@raspberrypi:~ $ sudo journalctl -u evcc --since "16:20:00"
-- Journal begins at Mon 2024-02-19 14:51:00 CET, ends at Sat 2024-02-24 16:41:56 CET. --
Feb 24 16:21:42 raspberrypi evcc[543]: [main  ] FATAL 2024/02/24 16:21:42 evcc was stopped. OS should restart the service. Or restart manually.
Feb 24 16:21:42 raspberrypi systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Feb 24 16:21:42 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
Feb 24 16:21:42 raspberrypi systemd[1]: evcc.service: Consumed 1.095s CPU time.
Feb 24 16:21:52 raspberrypi systemd[1]: evcc.service: Scheduled restart job, restart counter is at 1.
Feb 24 16:21:52 raspberrypi systemd[1]: Stopped evcc.
Feb 24 16:21:52 raspberrypi systemd[1]: evcc.service: Consumed 1.095s CPU time.
Feb 24 16:21:52 raspberrypi systemd[1]: Started evcc.
Feb 24 16:21:52 raspberrypi evcc[3264]: [main  ] INFO 2024/02/24 16:21:52 evcc 0.124.4 (3bf57f87)
Feb 24 16:21:52 raspberrypi evcc[3264]: [main  ] INFO 2024/02/24 16:21:52 using config file: /etc/evcc.yaml
Feb 24 16:21:52 raspberrypi evcc[3264]: [main  ] INFO 2024/02/24 16:21:52 starting ui and api at :7070
Feb 24 16:21:53 raspberrypi evcc[3264]: [db    ] INFO 2024/02/24 16:21:53 using sqlite database: /var/lib/evcc/evcc.db
Feb 24 16:21:53 raspberrypi evcc[3264]: [mqtt  ] INFO 2024/02/24 16:21:53 connecting evcc-1482118671 at tcp://192.168.178.80:1883
Feb 24 16:21:53 raspberrypi evcc[3264]: [mqtt  ] DEBUG 2024/02/24 16:21:53 tcp://192.168.178.80:1883 connected
Feb 24 16:21:53 raspberrypi evcc[3264]: [proxy-5200] DEBUG 2024/02/24 16:21:53 modbus proxy for 192.168.178.74:502 listening at :5200
Feb 24 16:22:00 raspberrypi evcc[3264]: [lp-1  ] WARN 2024/02/24 16:22:00 deprecated: minCurrent setting is ignored, please remove
Feb 24 16:22:00 raspberrypi evcc[3264]: [lp-1  ] WARN 2024/02/24 16:22:00 deprecated: maxcurrent setting is ignored, please remove
Feb 24 16:22:00 raspberrypi evcc[3264]: [lp-1  ] WARN 2024/02/24 16:22:00 deprecated: phases setting is ignored, please remove
Feb 24 16:22:00 raspberrypi evcc[3264]: [lp-2  ] WARN 2024/02/24 16:22:00 deprecated: minCurrent setting is ignored, please remove
Feb 24 16:22:00 raspberrypi evcc[3264]: [lp-2  ] WARN 2024/02/24 16:22:00 deprecated: maxcurrent setting is ignored, please remove
Feb 24 16:22:00 raspberrypi evcc[3264]: [lp-2  ] WARN 2024/02/24 16:22:00 deprecated: phases setting is ignored, please remove
Feb 24 16:22:01 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:22:01 charge total import: 4704.700kWh
Feb 24 16:22:01 raspberrypi evcc[3264]: [lp-2  ] DEBUG 2024/02/24 16:22:01 charge total import: 0.000kWh
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] WARN 2024/02/24 16:22:01 vehicle 'E-Bike' does not support automatic detection
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01 site config:
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01   meters:      grid ✓ pv ✓ battery ✓
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     grid:      power ✓ energy ✓ currents ✓
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     pv 1:      power ✓ energy ✗ currents ✗
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     pv 2:      power ✓ energy ✗ currents ✗
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     pv 3:      power ✓ energy ✓ currents ✗
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01   vehicles:
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     vehicle 3: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[... cut ...]
Feb 24 16:22:39 raspberrypi evcc[3264]: [lp-2  ] DEBUG 2024/02/24 16:22:39 charge total import: 0.000kWh
Feb 24 16:22:39 raspberrypi evcc[3264]: [lp-2  ] DEBUG 2024/02/24 16:22:39 charger status: A
Feb 24 16:22:39 raspberrypi evcc[3264]: [site  ] ERROR 2024/02/24 16:22:39 battery mode: modbus: exception '4' (server device failure), function '16'
Feb 24 16:22:40 raspberrypi evcc[3264]: [site  ] DEBUG 2024/02/24 16:22:40 ----
Feb 24 16:22:40 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:22:40 charge power: 0W
```

Sollten noch Logs fehlen, bitte um Info

andig, 2024-02-24:
Kannst Du eine Portweiterleitung auf den Kostal für info@evcc.io einrichten? Alternativ SSH für 

    ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIGi8rUJarKZuWnZNOnhtcXZCrPskbF7HOcig6nkjWlU2

Das müssen wir mal direkt anschauen.

StevieC121176, 2024-02-24:
Mail ist raus

andig, 2024-02-24:
Wir geschlossen bis Debugmöglichkeit besteht.

andig, 2024-02-25:
Es ist ein Modbusfehler, braucht also den Modbusport.

andig, 2024-02-26:
@StevieC121176 nochmal im Klartext: Du hast Port 80 weiter geleitet. Es braucht Port 1502.

StevieC121176, 2024-02-28:
> @StevieC121176 nochmal im Klartext: Du hast Port 80 weiter geleitet. Es braucht Port 1502.

Sorry, jetzt ist es Port 1502. Ich hoffe es klappt.

andig, 2024-02-28:
Grad probiert mit 0.124.4 kein Fehler:

    evcc meter --log error,modbus:trace -c steve.yaml

oder 

    evcc meter --log error,modbus:trace -c steve.yaml -b hold

Jetzt ist guter Rat teuer- es fehlt eine Möglichkeit, das Verhalten nachzuvollziehen :O

andig, 2024-02-28:
@StevieC121176 mach bitte mal ein `modbus: trace` und reproduziere den Fehler damit man sieht welche Abfrage da ursächlich ist.

andig, 2024-02-28:
Der Fehler kommt aus https://github.com/evcc-io/evcc/pull/11904 /cc @MarcusWichelmann

In #11904- Solaredge- ist es ein `int` Wert der, als `float32s` durch einen `IntSetter` (little endian) geschrieben werden soll.
Hier in #12444- bei Kostal- ist es ebenfalls ein `int` Wert, der als `float32s` durch einen `FloatSetter` (little endian) geschrieben werden soll.

Warum ein `IntSetter` in #11904 einen Floatwert schreiben soll? Dass es ein `IntSetter` ist liegt daran, dass er mit einem Integer Wert befüllt wird (minsoc).
Warum das hier in #12444 anders ist? Weil der `soc` in `limitsoc` kopiert wird und `limitsoc` als `FloatSetter` definiert ist.

Der Code sieht so aus:

```go
val = m.scale * val

var uval uint64
switch m.op.Length {
case 2:
	uval = uint64(math.Float32bits(float32(val)))
case 4:
	uval = math.Float64bits(val)
}

var err error
switch m.op.FuncCode {
case gridx.FuncCodeWriteMultipleRegisters:
	err = m.writeMultipleRegisters(uval)
default:
	err = fmt.Errorf("invalid write function code: %d", m.op.FuncCode)
}
```

und in `m.writeMultipleRegisters` werden dann die Bytes vertauscht in dem `op.Encode` aufgerufen wird:

```go
case "float32s", "ieee754s":
	op.Encode = func(v uint64) uint64 {
		fmt.Printf("litte endian: % x\n", v)
		b := make([]byte, 4)
		encoding.PutFloat32LswFirst(b, float32(v))
		return uint64(binary.BigEndian.Uint32(b))
	}
```

Nur was bei `int` richtig ist- erstmal `float32` draus zu machen- ist bei float32 falsch weil der Wert dann 2x umgewandelt wird.

Die Logik hier im Modbus Plugin ist also kaputt, je nachdem ob int->float oder float->int geschrieben werden soll.

@MarcusWichelmann mit fehl heute abend die Fantasie- hast Du eine gute Idee, wie wir das elegant lösen können?

StevieC121176, 2024-02-28:
> @StevieC121176 mach bitte mal ein `modbus: trace` und reproduziere den Fehler damit man sieht welche Abfrage da ursächlich ist.

```
pi@raspberrypi:~ $ evcc meter --log error,modbus:trace 
[main  ] INFO 2024/02/28 23:53:08 evcc 0.124.4 (ab877571)
[main  ] INFO 2024/02/28 23:53:08 using config file: /etc/evcc.yaml
pv1
---
[modbus] TRACE 2024/02/28 23:53:11 modbus: send 00 3b 00 00 00 06 47 03 9d 30 00 08
[modbus] TRACE 2024/02/28 23:53:12 modbus: recv 00 3b 00 00 00 13 47 03 10 ff fe ff ff 00 00 00 00 00 00 00 00 00 03 00 00
[modbus] TRACE 2024/02/28 23:53:12 modbus: send 00 3c 00 00 00 06 47 03 9d 38 00 14
[modbus] TRACE 2024/02/28 23:53:12 modbus: recv 00 3c 00 00 00 2b 47 03 28 00 01 44 43 5f 53 54 52 49 4e 47 5f 31 20 20 20 20 20 00 01 00 38 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[modbus] TRACE 2024/02/28 23:53:12 modbus: send 00 3d 00 00 00 06 47 03 9d 30 00 08
[modbus] TRACE 2024/02/28 23:53:12 modbus: recv 00 3d 00 00 00 13 47 03 10 ff fe ff ff 00 00 00 00 00 00 00 00 00 03 00 00
[modbus] TRACE 2024/02/28 23:53:12 modbus: send 00 3e 00 00 00 06 47 03 9d 4c 00 14
[modbus] TRACE 2024/02/28 23:53:12 modbus: recv 00 3e 00 00 00 2b 47 03 28 00 02 44 43 5f 53 54 52 49 4e 47 5f 32 20 20 20 20 20 00 02 00 48 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[modbus] TRACE 2024/02/28 23:53:12 modbus: send 00 3f 00 00 00 06 47 03 9d 30 00 08
[modbus] TRACE 2024/02/28 23:53:12 modbus: recv 00 3f 00 00 00 13 47 03 10 ff fe ff ff 00 00 00 00 00 00 00 00 00 03 00 00
[modbus] TRACE 2024/02/28 23:53:12 modbus: send 00 40 00 00 00 06 47 03 9d 60 00 14
[modbus] TRACE 2024/02/28 23:53:12 modbus: recv 00 40 00 00 00 2b 47 03 28 00 03 44 43 5f 53 54 52 49 4e 47 5f 33 20 20 20 20 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Power: 0W

battery2
--------
[modbus] TRACE 2024/02/28 23:53:12 modbus: send 00 41 00 00 00 06 47 03 9d 76 00 3e
[modbus] TRACE 2024/02/28 23:53:12 modbus: recv 00 41 00 00 00 7f 47 03 7c 00 18 39 08 3a 66 3a 66 00 00 00 00 00 00 00 00 00 00 00 3f 00 00 00 00 00 00 00 83 00 00 00 00 00 00 00 00 00 01 00 04 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 14 85 16 d0 00 00 00 21 00 00 00 00 00 20 00 00 00 00 00 00 00 4d 0a 00 0a 00 01 a5 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff fe ff fe 00 00
[modbus] TRACE 2024/02/28 23:53:12 modbus: send 00 42 00 00 00 06 47 03 9d 76 00 3e
[modbus] TRACE 2024/02/28 23:53:13 modbus: recv 00 42 00 00 00 7f 47 03 7c 00 18 39 08 3a 66 3a 66 00 00 00 00 00 00 00 00 00 00 00 3f 00 00 00 00 00 00 00 83 00 00 00 00 00 00 00 00 00 01 00 04 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 14 85 16 d0 00 00 00 21 00 00 00 00 00 20 00 00 00 00 00 00 00 4d 0a 00 0a 00 01 a5 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff fe ff fe 00 00
Power: 421W
Soc:   63%

grid3
-----
[modbus] TRACE 2024/02/28 23:53:13 modbus: send 00 43 00 00 00 06 47 03 00 fc 00 02
[modbus] TRACE 2024/02/28 23:53:13 modbus: recv 00 43 00 00 00 07 47 03 04 00 00 42 a6
Power: 83W

pv4
---
Power: 0W
```

und hier mit hold

```
pi@raspberrypi:~ $ evcc meter --log error,modbus:trace -b hold
[main  ] INFO 2024/02/28 23:56:14 evcc 0.124.4 (ab877571)
[main  ] INFO 2024/02/28 23:56:14 using config file: /etc/evcc.yaml
[modbus] TRACE 2024/02/28 23:56:17 modbus: send 00 3b 00 00 00 06 47 03 9d 76 00 3e
[modbus] TRACE 2024/02/28 23:56:17 modbus: recv 00 3b 00 00 00 7f 47 03 7c 00 18 39 08 3a 66 3a 66 00 00 00 00 00 00 00 00 00 00 00 3f 00 00 00 00 00 00 00 83 00 00 00 00 00 00 00 00 00 01 00 04 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 14 84 16 d0 00 00 00 21 00 00 00 00 00 20 00 00 00 00 00 00 00 57 0a 00 0a 00 01 cc 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff fe ff fe 00 00
[modbus] TRACE 2024/02/28 23:56:17 modbus: send 00 3c 00 00 00 0b 47 10 04 12 00 02 04 f8 00 4e 84
[modbus] TRACE 2024/02/28 23:56:17 modbus: recv 00 3c 00 00 00 03 47 90 04
[main  ] FATAL 2024/02/28 23:56:17 set battery mode: modbus: exception '4' (server device failure), function '16'
pi@raspberrypi:~ $ 
```

StevieC121176, 2024-02-28:
mit der Version 0.124.5 und 0.124.6 kann ich den Fehler nicht mehr reproduzieren.

MarcusWichelmann, 2024-02-28:
@andig Ah, jetzt verstehe ich. Das lässt sich vermutlich nur mit einem kleinen Refactoring vernünftig lösen, meine Lösung ist ja auch eher als "Hack" anzusehen. ;) Müsste mir da aber auch erstmal mit frischem Hirn einen Überblick verschaffen, wie das aktuell gestrickt ist.

Falls du (mit besserem Überblick über die Codebase) keine Idee haben solltest, kann ich die Tage aber auch wohl mal reinschauen.

andig, 2024-02-29:
Gestern ist mir nix Schlaues eingefallen. In der Tat braucht das noch einen Umbau für die Schreibzugriffe. Im nightly ist es jetzt wieder "falsch", betroffen ist ausschließlich Batteriesteuerung. Jede Hilfe ist willkommen :)

closed 2024-03-02 by andig

doctordsingh, 2024-07-30:
gibt es mittlerweile eine Lösung für? ich wusste auch nicht was für eine Konsequenz diesen Fehler überhaupt hat?

andig, 2024-07-30:
Seit 6 Monaten gelöst…