evcc-io/evcc #12460: Min.Ladung% lässt sich nicht ändern wenn Fahrzeug Offline

gennergy, 2024-02-25:
### Describe the bug

Der Porsche Server schmeißt aktuell wieder Internal Server Error:
[main  ] ERROR 2024/02/25 11:52:33 creating vehicle taycan failed: cannot create vehicle type 'template': cannot create vehicle 
type 'porsche': unexpected status: 500 (Internal Server Error)                                                                  
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'porsche': unexpected status:
 500 (Internal Server Error)                                                                                                    
Capacity: 84.0kWh                                                                                                               
Features: [Offline]

Das Problem ist, dass nun das Fahrzeug sofort beim einstecken geladen wird, weil ein Wert für die mindest-Ladung eingestellt ist. Stand der Wert vor dem Offline Modus des Fahrzeugs auf --, könnte nun PV Regeln geladen werden. Leider kann der "Min. Ladung %" Wert nicht verändert werden, wenn das Fahrzeug Offline ist.

### Steps to reproduce

1. Fahrzeug Offline
2. Min. Ladung % ändern

### Configuration details

```yaml
nebensächlich
```

### Log details

```text
kein Log, Gui Problem
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.124.4

andig, 2024-02-25:
> Das Problem ist, dass nun das Fahrzeug sofort beim einstecken geladen wird, weil ein Wert für die mindest-Ladung eingestellt ist.

Das ist Absicht. Die Alternative wäre nicht zu laden- damit stehst Du dann schlimmstenfalls im leerem Auto da.

> Das Problem ist, dass nun das Fahrzeug sofort beim einstecken geladen wird

Das kann nur passieren, wenn Du ein Defaultfahrzeug konfiguriert hast oder dieses per RFID (oder ISO) einloggst- anderenfalls würde es beim Anstecken mangels API gar nicht erkannt.

Workaround: Defaultfahrzeug aus der Config nehmen oder Fahrzeug auf Gast stellen.

/cc @naltatis @MarkusGH

MarkusGH, 2024-02-25:
@andig: Du hast das "wontfix" Label hinzugefügt. Wollt Ihr das UI Problem nicht korrigieren?

StefanSchoof, 2024-02-25:
Ich gebe andig Recht, das im Fehlerfall laden besser ist. Aber ich verstehe nicht warum man nicht in der Lage sein sollte den minSoc wieder per GUI raus zu nehmen. Wenn man auf Gastfahrzeug wechselt, wird die Lademenge in der Zwischenzeit nicht auf das richtige Fahrzeug gespeichert.

gennergy, 2024-02-25:
Ich nutze die mindest Ladung nur um ein tiefentladenen Akku wieder auf auf eine wie gesagt Mindest Ladung zu bringen, damit er nicht solange entladen rumsteht bis die Energie günstig ist. Den Ladestand habe ich immer soweit im Auge, dass ich weiß, wenn ich "Schnell" laden muss.
Das Fahrzeug ist im Monent 60% geladen.  Damit komme ich in der Regel 2-3 Tage hin. Am Wochenende ist Tibber günstig und ich lade meist für die ganze Woche. Ich sehe jetzt keine Chance, automatisiert den günstigen Preis abzugreifen. Es sei denn, ich könnte jetzt bei dem Offline Fahrzeug die mindest Ladung wegdrehen. Ich würde eher auf die Mindest Ladung verzichten und es vorsorglich abstellen. 
Denn gerade bei dem Porsche ist doch dauernd ein Problem mit dem Server oder dem Login. (Deswegen ist wahscheinlich das Porsche "Power Up" seit über einem Jahr nicht gefixt)

andig, 2024-02-25:
Das mag alles sein, ist aber Dein UC. Wir handeln *grundsätzlich* so, dass laden Vorrang vor minimalem Energieverbrauch hat.

MarkusGH, 2024-02-25:
> Das mag alles sein, ist aber Dein UC. Wir handeln _grundsätzlich_ so, dass laden Vorrang vor minimalem Energieverbrauch hat.

Das ist doch unbestritten - aber in dem Issue hier geht es doch nur um das UI, das einen eine einmal eingestellte und persistierte Mindestladung nicht mehr ändern lässt wenn ein Fahrzeug "offline" geht.

andig, 2024-02-25:
Da sind wir uns seit ungefähr 10 Kommentaren einig...

> Es sei denn, ich könnte jetzt bei dem Offline Fahrzeug die mindest Ladung wegdrehen. Ich würde eher auf die Mindest Ladung verzichten und es vorsorglich abstellen.

Wie das geht hab ich ja oben beschrieben. Keine Ahnung, warum wir uns da im Kreis drehen?

MarkusGH, 2024-02-25:
> > Es sei denn, ich könnte jetzt bei dem Offline Fahrzeug die mindest Ladung wegdrehen. Ich würde eher auf die Mindest Ladung verzichten und es vorsorglich abstellen.
> 
> Wie das geht hab ich ja oben beschrieben. Keine Ahnung, warum wir uns da im Kreis drehen?

Kann sein ich bin begriffsstutzig, aber wie kann ich bei einem "offline" Fahrzeug einen in der Datenbank gespeicherten Wert für die Mindestladung wieder löschen?

MarkusGH, 2024-02-25:
@gennergy: Korrigiere bitte mal Deine "Steps to reproduce:"

1. Fahrzeug online
2. Min. Ladung % auf Wert >0% setzen
3. Fahrzeug offline
4. Min. Ladung % kann nicht mehr geändert werden.

naltatis, 2024-02-28:
@andig Wir müssen hier nochmal überdenken, wie wir mit Online-Fahrzeugen umgehen, die aus irgendeinem Grund nicht mehr erreicht werden können. Passiert das während der Laufzeit von evcc (Hersteller-Server nicht erreichbar) haben wir aktuell ein anderes Verhalten, als wenn das bei einem Neustart auftritt. Diese Unterschiede sind ja auch der Grund für den hier beschriebenen komischen Zustand.

Der Fahrzeug-bei-Neustart-Offline Fall führt dazu, dass der Nutzer ein Fahrzeug sieht, was zu meinen einen anderen Titel hat (+ Offline), das macht auch Quatscheinträge im Ladelog. Das größere Problem ist aber, dass wir in diesem Zustand einen Daten-Zwitter aus Offline-/Online-Fahrzeug in der UI und Steuerung bekommen. Offlinefahrzeuge haben weder MinSoc noch Plan-Soc. Ein Onlinefahrzeug das durch Neustart offline wird, kann diese beiden Werte aktuell aber haben. Hier müssen wir entweder diese Felder bei der Initialisierung überspringen oder (wäre mein Favorit) einen separaten "aktuell nicht erreichbar" Status am Fahrzeug setzen. Auf der Basis könnte die UI dann ne Markierung (bspw. Offline Label) hinzufügen. Die Funktionen in der UI würde aber weiterhin die eines Online-Fahrzeuges (evtl. mit Warnhinweis versehen) bleiben. Das Ladeverhalten (im zweifelsfall bei minsoc laden) würde dann genau so bleiben wie jetzt, nur dass man es weiterhin editieren kann.

andig, 2024-03-09:
> Der Fahrzeug-bei-Neustart-Offline Fall führt dazu, dass der Nutzer ein Fahrzeug sieht, was zu meinen einen anderen Titel hat (+ Offline), das macht auch Quatscheinträge im Ladelog.

Das ist blöd, lässt sich aber anschließend korrigieren.

>  Das größere Problem ist aber, dass wir in diesem Zustand einen Daten-Zwitter aus Offline-/Online-Fahrzeug in der UI und Steuerung bekommen. Offlinefahrzeuge haben weder MinSoc noch Plan-Soc. Ein Onlinefahrzeug das durch Neustart offline wird, kann diese beiden Werte aktuell aber haben. 

Das stimmt. Dieses Fahrzeug kann aber per Erkennung nie gefunden werden- denn es ist ja offline und hat keinen Status. In den Grenzfall kann man daher nur kommen wenns ein Defaultfahrzeug ist. Und dann halte ich es erstmal für korrekt- mangels besseren Wissens- das Fahrzeug zu laden.

> Hier müssen wir entweder diese Felder bei der Initialisierung überspringen oder (wäre mein Favorit) einen separaten "aktuell nicht erreichbar" Status am Fahrzeug setzen. 

Minsoc muss angewandt werden- sonst bleibt das Auto leer!

> Hier müssen wir entweder diese Felder bei der Initialisierung überspringen oder (wäre mein Favorit) einen separaten "aktuell nicht erreichbar" Status am Fahrzeug setzen. 

Ich denke mal laut. Z.B. ein "ich bin ein offline Fahrzeug aber nur weils grad dumm läuft" Feature? Das bräuchte dann kein neues Api.

naltatis, 2024-03-10:
> Ich denke mal laut. Z.B. ein "ich bin ein offline Fahrzeug aber nur weils grad dumm läuft" Feature? Das bräuchte dann kein neues Api.

Ja genau. Wir brauchen aber irgendein Merkmal (nicht " Offline" Endung im Namen) an der wir das erkennen können und dann die Logiken und Darstellungen daran anpassen.

Weiteres Thema: Dieses Merkmal sollte sich auch im Betrieb ändern können. Heißt ein beim Start nicht erkanntes Fahrzeug (kein Empfang, ...), sollten wir in regelmäßigen Abständen wieder auf Verfügbarkeit checken und im Erfolgsfall den vollen Funktionsumfang wieder freischalten. Aktuell muss der User manuelle Neustarts machen, um zu prüfen, ob es geht. Das betrifft allerdings nicht nur Fahrzeuge, sondern all unsere Devices, Tarife, .... Ich weiß, dass das ein größerer technischer Umbau ist (Konzept Healthcheck). Aber das ist das, wo wir aus Nutzersicht hin müssen.
Vielleicht Out-of-Scope für dieses konkrete Problem hier, aber geht mir eher um die Richtung.

StefanSchoof, 2024-03-10:
Müsste es dann auch während evcc läuft auch in die andere Richtung gehen? Also von Online auf Offline Fahrzeug, weil der sich über längere Zeit nicht abrufbar ist.

andig, 2024-03-10:
@naltatis “wieder zum Leben erwecken“ wäre nett, und könnte man auf der Basis vllt. Machen- denn dann erkennen wir die halbtoten Fahrzeuge ja. Eins nach dem Anderen. Wie sollte das „Feature“ heissen? Sowas wie `Repairable`?

naltatis, 2024-03-10:
> Wie sollte das „Feature“ heissen? Sowas wie Repairable?

Würde da eher aus einer Zustandssicht draufschauen: `status: ok|unreachable`, `available: false`, `healthy: false`, ...

andig, 2024-03-10:
So funktionieren die Features aber nicht. Wenn du das willst brauchts eine andere Lösung. Eins nach dem Anderen?

closed 2024-03-14 by andig