evcc-io/evcc #12461: Battery Control: SAX Homespeicher

premultiply, 2024-02-25:
### Discussed in https://github.com/evcc-io/evcc/discussions/12448

<div type='discussions-op-text'>

<sup>Originally posted by **Sebreiter** February 24, 2024</sup>
Hallo zusammen,
ist ggf. auch eine aktive Batteriesteuerung für den o.g. Akku möglich?
Wenn möglich, wird diese im entsprechenden Template integriert?
Lt. Hersteller ist der Standardeintrag im Register 43 4600 (in W)
Byte Inhalt Wert                                         Zugriffsart
0x2b 43 Leistungsgrenzwert für Entladung write only
Entsprechend müsste dann beim Modus "Schnell" der Registerwert auf 0 gesetzt werden und anschließend wieder auf 4600,
oder verstehe ich das falsch?

Nochmal vielen Dank für die Implementierung des SAX Power Akku in evcc.
Seit ich den Akku im März 2023 installiert habe, funktioniert alles Tadellos, soweit ich das beurteilen kann.
Viele Grüße, Reiner S.</div>

premultiply, 2024-02-25:
@Sebreiter Wie funktioniert das Zwangsladen?
Die Doku ist... sparsam.

Sebreiter, 2024-02-25:
@premultiply Dazu kann ich leider nichts sagen. Ich habe auch in der Doku keinen Hinweis auf "Zwangsladen" gefunden.
Gruß, Reiner

closed 2024-03-03 by andig

Sebreiter, 2024-03-08:
Hallo zusammen,
erst einmal Danke für die schnelle Umsetzung.
Ich habe natürlich sofort den Schalter auf "EIN" gestellt. Leider erscheint daraufhin folgende Fehlermeldung.
ERROR 2024/03/06 10:36:47 battery mode: read tcp 192.168.178.20:45390->192.168.178.39:3600: i/o timeout
192.168.178.20=HA
192.168.178.39=SAX
Was will er denn da lesen?
Bei mir läuft das Ganze unter HA. Kann das daran liegen?
Wenn ich den Schalter auf "AUS" stelle, erscheint keine Fehlermeldung.
Viele Grüße, Reiner

andig, 2024-03-08:
Ohne Logfile kann man da nur raten: bitte Issue mit Logfile

PR #12462 (2024-03-03): SAX Homespeicher: add battery control
Fixes #12461