evcc-io/evcc #12471: GoodWe ET (GW25K-ET): Leistungswerte Batterie und Verbrauch null

daftbugs, 2024-02-25:
### Describe the bug

evcc liest stets den Wert 0 für Leistung der Batterie (sowie, scheinbar, Energie Grid):
`grid1
-----
Power:  63W
Energy: 0.0kWh

pv2
---
Power:  0W
Energy: 390.7kWh

battery3
--------
Power:    0W
Energy:   244.6kWh
Soc:      69%
Capacity: 16.4kWh`

(Die 0W Power bei pv2 sind allerdings angesichts der Uhrzeit korrekt)

evcc meter -l trace:
`
[main  ] INFO 2024/02/25 20:10:28 evcc 0.124.4
[main  ] INFO 2024/02/25 20:10:28 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/25 20:10:28 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/02/25 20:10:28 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT * FROM `settings` 1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT     fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT     fk_devices_details      %") -1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%       device_id       %") -1 <nil>
[db    ] TRACE 2024/02/25 20:10:28 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
grid1
-----
[modbus] TRACE 2024/02/25 20:10:28 modbus: send 00 01 00 00 00 06 f7 03 8c b9 00 02
[modbus] TRACE 2024/02/25 20:10:28 modbus: recv 00 01 00 00 00 07 f7 03 04 00 00 00 1b
[modbus] TRACE 2024/02/25 20:10:28 modbus: send 00 02 00 00 00 06 f7 03 8c b1 00 02
[modbus] TRACE 2024/02/25 20:10:28 modbus: recv 00 02 00 00 00 07 f7 03 04 00 00 00 00
Power:  -27W
Energy: 0.0kWh

pv2
---
[modbus] TRACE 2024/02/25 20:10:28 modbus: send 00 03 00 00 00 06 f7 03 89 21 00 02
[modbus] TRACE 2024/02/25 20:10:28 modbus: recv 00 03 00 00 00 07 f7 03 04 00 00 00 00
[modbus] TRACE 2024/02/25 20:10:28 modbus: send 00 04 00 00 00 06 f7 03 89 25 00 02
[modbus] TRACE 2024/02/25 20:10:28 modbus: recv 00 04 00 00 00 07 f7 03 04 00 00 00 00
[modbus] TRACE 2024/02/25 20:10:28 modbus: send 00 05 00 00 00 06 f7 03 89 29 00 02
[modbus] TRACE 2024/02/25 20:10:28 modbus: recv 00 05 00 00 00 07 f7 03 04 00 00 00 00
[modbus] TRACE 2024/02/25 20:10:28 modbus: send 00 06 00 00 00 06 f7 03 89 2d 00 02
[modbus] TRACE 2024/02/25 20:10:28 modbus: recv 00 06 00 00 00 07 f7 03 04 00 00 00 00
[modbus] TRACE 2024/02/25 20:10:28 modbus: send 00 07 00 00 00 06 f7 03 89 77 00 02
[modbus] TRACE 2024/02/25 20:10:28 modbus: recv 00 07 00 00 00 07 f7 03 04 00 00 0f 43
Power:  0W
Energy: 390.7kWh

battery3
--------
[... 98 lines cut ...]
[lp-1  ] DEBUG 2024/02/25 20:26:39 charger disable
[lp-1  ] DEBUG 2024/02/25 20:26:39 wake-up timer: stop
[site  ] DEBUG 2024/02/25 20:26:49 ----
[lp-1  ] DEBUG 2024/02/25 20:26:49 charge power: 193W
[site  ] DEBUG 2024/02/25 20:26:49 pv power: 0W
[site  ] DEBUG 2024/02/25 20:26:49 battery soc: 68%
[site  ] DEBUG 2024/02/25 20:26:49 battery power: 0W
[site  ] DEBUG 2024/02/25 20:26:49 grid power: 193W
[site  ] DEBUG 2024/02/25 20:26:49 site power: 193W
[lp-1  ] DEBUG 2024/02/25 20:26:49 charge total import: 0.000kWh
[lp-1  ] DEBUG 2024/02/25 20:26:49 charger status: B
[lp-1  ] DEBUG 2024/02/25 20:26:49 final charge energy: 0kWh
[lp-1  ] INFO 2024/02/25 20:26:49 stop charging <-
[lp-1  ] DEBUG 2024/02/25 20:26:49 charge total import: 0.000kWh
[lp-1  ] DEBUG 2024/02/25 20:26:49 vehicle soc: 62%
[lp-1  ] DEBUG 2024/02/25 20:26:49 vehicle range: 181km`
```

### What type of operating system are you running?

Linux

### Version

0.124.4

andig, 2024-03-03:
Laut template lesen wir:

```
  power:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register: # manual non-sunspec register configuration
      address: 35183 # Pbattery1 First group battery power
      type: holding
      decode: int16
  soc:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register: # manual non-sunspec register configuration
      address: 37007 # SOC First group battery capacity
      type: holding
      decode: uint16
  energy:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register: # manual non-sunspec register configuration
      address: 35209 # E-BatteryDischarge Discharge energy
      type: holding
      decode: uint32
    scale: 0.1
```

Brauchts bei Dir vllt. andere Register? Was heisst "first group"?

andig, 2024-03-03:
Grid energy ist:

```
  energy:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register: # manual non-sunspec register configuration
      address: 36017 # E-Total-Buy
      type: holding
      decode: float32
    scale: 0.1
```

daftbugs, 2024-03-03:
[BESS Modbus protocol Map V1.1 20231030.pdf](https://github.com/evcc-io/evcc/files/14474214/BESS.Modbus.protocol.Map.V1.1.20231030.pdf)

Ich kann zwar nicht genau eingrenzen, bei welchen Modell es sich auftrennt. Bei dieser Frage hilft auch das Attachment nicht weiter. Allerdings scheint das zutreffend zu sein auf die akt. Inverter-Modellreihe "ET 15-30".

Demnach ist 35182 ein S32-Register mit der "battery power". 35183 gibt es da gar nicht.

Erklärt allerdings  nicht 36017. Der scheint weiterhin korrekt.

andig, 2024-03-08:
@premultiply die Implementierung kommt von Dir?

PR #12854 (2024-03-10): GoodWe-Hybrid: fix registers
Fix #12471
closed 2024-03-10 by andig