evcc-io/evcc #12473: GoodWe-WiFi causes busy/spin loop in test

GrimmiMeloni, 2024-02-26:
### Describe the bug

Executing tests locally, I see the log spammed due to a failing test for goodwe-wifi.
The log entries shown below originate from here:

https://github.com/evcc-io/evcc/blob/2df084afbe3f724a965422f13b76d367bd2d6cc9/meter/goodwe/server.go#L76

### Steps to reproduce

1. make test

### Configuration details

```yaml
N/A
```

### Log details

```text
... the following 3 lines are an example. The console gets bombarded with this until the testdriver times out.

[goodwe-wifi] ERROR 2024/02/26 08:03:47 write udp [::]:8899->:8899: sendto: no route to host
[goodwe-wifi] ERROR 2024/02/26 08:03:47 write udp [::]:8899->:8899: sendto: no route to host
[goodwe-wifi] ERROR 2024/02/26 08:03:47 write udp [::]:8899->:8899: sendto: no route to host
panic: test timed out after 30s
running tests:
        TestTemplates/volkszaehler-ws/grid (30s)

goroutine 543 [running]:
testing.(*M).startAlarm.func1()
        [goodwe-wifi] ERROR 2024/02/26 08:03:47 write udp [::]:8899->:8899: sendto: no route to host
/Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/testing/testing.go:2366 +0x30c
created by time.goFunc
[goodwe-wifi] ERROR 2024/02/26 08:03:47 write udp [::]:8899->:8899: sendto: no route to host
        /Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/time/sleep.go:177 +0x38

goroutine 1 [chan receive]:
testing.tRunner.func1()
        /Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/testing/testing.go:1650 +0x43c
testing.tRunner([goodwe-wifi] ERROR 2024/02/26 08:03:47 write udp [::]:8899->:8899: sendto: no route to host
[goodwe-wifi] ERROR 2024/02/26 08:03:47 write udp [::]:8899->:8899: sendto: no route to host
0x140007f8000[goodwe-wifi] ERROR 2024/02/26 08:03:47 write udp [::]:8899->:8899: sendto: no route to host
, 0x14000057c58)
        /Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/testing/testing.go:1695 +0x128
testing.runTests(0x14000b0a498, {0x101cc7800, 0x1, 0x1}, {0x14000057d18?, 0x100193d20?, 0x101d155a0?})
        /Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/testing/testing.go:2159 +0x3b0
testing.(*M).Run(0x1400078a140)
        /Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/testing/testing.go:2027 +0x5a4
main.main()
        _testmain.go:47 +0x16c

goroutine 22 [chan receive]:
testing.tRunner.func1()
        /Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/testing/testing.go:1650 +0x43c
testing.tRunner(0x140007f81a0, 0x1012e91f8)
        /Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/testing/testing.go:1695 +0x128
created by testing.(*T).Run in goroutine 1
        /Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/testing/testing.go:1742 +0x318

goroutine 478 [IO wait]:
internal/poll.runtime_pollWait(0x1021afec0, 0x72)
        /Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/runtime/netpoll.go:345 +0xa0
internal/poll.(*pollDesc).wait(0x140014fb680?, 0x1400143c000?, 0x0)
        /Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/internal/poll/fd_poll_runtime.go:84 +0x28
internal/poll.(*pollDesc).waitRead(...)
        /Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/internal/poll/fd_poll_runtime.go:89
internal/poll.(*FD).ReadFromInet6(0x140014fb680, {0x1400143c000, 0x400, 0x400}, 0x1400094ed88)
        /Users/mhess/go/pkg/mod/golang.org/toolchain@v0.0.1-go1.22.0.darwin-arm64/src/internal/poll/fd_unix.go:274 +0x1ec
net.(*netFD).readFromInet6(0x140014fb680, {0x1400143c000?, 0x30?, 0x28?}, 0x1400094edd8?)
[... 214 lines cut ...]
        /Users/mhess/go/pkg/mod/nhooyr.io/websocket@v1.8.10/read.go:175 +0x48
nhooyr.io/websocket.(*Conn).reader(0x1400134cb60, {0x1013046f0, 0x1400033e070})
        /Users/mhess/go/pkg/mod/nhooyr.io/websocket@v1.8.10/read.go:344 +0xc8
nhooyr.io/websocket.(*Conn).Reader(...)
        /Users/mhess/go/pkg/mod/nhooyr.io/websocket@v1.8.10/read.go:37
nhooyr.io/websocket/wsjson.read({0x1013046f0?, 0x1400033e070?}, 0x1400134cb60, {0x10112e1a0, 0x14000b18000})
        /Users/mhess/go/pkg/mod/nhooyr.io/websocket@v1.8.10/wsjson/wsjson.go:24 +0x80
nhooyr.io/websocket/wsjson.Read(...)
        /Users/mhess/go/pkg/mod/nhooyr.io/websocket@v1.8.10/wsjson/wsjson.go:18
github.com/hasura/go-graphql-client.(*WebsocketHandler).ReadJSON(0x1400071acc0, {0x10112e1a0, 0x14000b18000})
        /Users/mhess/go/pkg/mod/github.com/hasura/go-graphql-client@v0.11.0/subscription.go:1026 +0x6c
github.com/hasura/go-graphql-client.(*SubscriptionClient).Run.func1()
        /Users/mhess/go/pkg/mod/github.com/hasura/go-graphql-client@v0.11.0/subscription.go:772 +0xa4
created by github.com/hasura/go-graphql-client.(*SubscriptionClient).Run in goroutine 586
        /Users/mhess/go/pkg/mod/github.com/hasura/go-graphql-client@v0.11.0/subscription.go:765 +0x180
FAIL    github.com/evcc-io/evcc/meter   30.781s
```

### What type of operating system are you running?

Linux

### Version

master @ 3bf57f870ee66995c9a01784d93d9ac7e5ce7e02

andig, 2024-02-26:
Also Backoff mit einbauen?

GrimmiMeloni, 2024-02-26:
Gute Frage.
Dieser Fall tritt jetzt erst mal im Test auf, wo es keinen Endpoint gibt, der auf der Adresse lauscht. 

Es könnte auch in einem echten Szenario auftreten. Kurz gesagt, wenn ein Fehler beim ModBus Aufruf zurück kommt, hätten wir genau das gleiche Verhalten und würden auf dem ModBus rumhämmern. Das will vermutlich niemand.

Ich denke, ich habe mich gerade selbst überzeugt, dass Backoff da rein muss. 😆

closed 2024-02-26 by andig