evcc-io/evcc #12474: SunSpec Battery: Broken Test

GrimmiMeloni, 2024-02-26:
### Describe the bug

The sunspec-inverter/battery tests are failing when executing locally.

### Steps to reproduce

1. make test

### Configuration details

```yaml
N/A
```

### Log details

```text
--- FAIL: TestTemplates (0.00s)
    --- FAIL: TestTemplates/sunspec-inverter/battery (0.00s)
        render_testing.go:88: map[baudrate:9600 capacity:50 comset:8N1 device:/dev/ttyUSB0 host:localhost id:1 modbus: port:502 tcpip:true template:sunspec-inverter usage:battery]
        render_testing.go:88: cannot create meter type 'template': cannot create meter type 'custom': power: missing configuration
FAIL
FAIL	github.com/evcc-io/evcc/meter	31.895s
```

### What type of operating system are you running?

Linux

### Version

master @ 3bf57f870ee66995c9a01784d93d9ac7e5ce7e02

andig, 2024-02-26:
Da gibts tatsächlich nur soc weil Model 802 keine Leistung hat. Könnte man nur raus werfen.

premultiply, 2024-02-26:
Ja, das geht nur in Verbindung mit einem anderen Model welches die Leistung liefern würde.

andig, 2024-02-26:
@premultiply ein nicht-Hybrid Inverter hat seine Batterieleistung doch ganz normal als PV Leistung, oder? Er kann also nur entweder- oder verwendet werden. Damit müsste hier der PV Block nach Battery kopiert werden?

premultiply, 2024-02-26:
Ja, genau.

closed 2024-02-26 by andig