evcc-io/evcc #12480: Implement WARP2 charger wakeup using evse_control_pilot_disconnect API or start/stop

Kishi85, 2024-02-26:
**Is your feature request related to a problem? Please describe.**
The WARP2 charger allows direct control of the control pilot signal to the car, allowing simulation of a full unplug/replug of the charging cable via API (see https://www.warp-charger.com/api.html#features_section and https://www.warp-charger.com/api.html#evse_control_pilot_disconnect) but EVCC does not include this feature for it's charger wakeup function (no API.Resurrector for WARP2). 

**Describe the solution you'd like**
It would be great to see the use of the evse_control_pilot_disconnect API endpoint being used for charger car wakeup (as available for e.g. OpenWB) if it is available. If it is not available (in case of using WARPs load management) similar functionality is also available by stopping and starting the charge on WRAP (via API stop_charging/start_charging) especially when EV Wakeup is also configured in WARPs Charge settings.

**Describe alternatives you've considered**
None

**Additional context**
I have a Hyundai Kona EV 2020 that sometimes does not wakeup after it's been idle for too long. Bluelink integration triggers the wakeup and charge but is sometimes not available (503 Service unavailble) and also is only trigger once every 15 Minues (by default due to API Limits) hence why the charger wakeup using my WARP2 would be preferrable.

andig, 2024-02-26:
> Dieser Wert kann über evse/control_pilot_disconnect_update mit dem selben Payload aktualisiert werden.

Mhhhm was muss da wohl geschrieben werden?

Kishi85, 2024-02-26:
evse/control_pilot_disconnect via HTTP returns (without a car connected as I'm not home atm):
```
{"disconnect":false}
```
so presumably it is just a bool true but I don't know if it automatically resets after it's done or if you have to reset it manually back to false back again. I can check that once I have the car back on the charger tomorrow.

deadrabbit87, 2024-02-26:
I think this is only relevant if no WEM (warp energy manager) is installed. 

In case of that, the wem is controlling the CP disconnect.

andig, 2024-02-26:
No quite- that's the delay between our tries (up to 3). Sofar all boxes we've encountered were doing "atomic" wakeups. This is a bit unfortunate since we'd also need to handle the timing:

> You have to set it back to false to reconnect CP.

Since Warp already does this it seems we shouldn't. Or is CP disconnect not used at all without WEM?

rtrbt, 2024-02-26:
> Or is CP disconnect not used at all without WEM?

Only the API to control the disconnect is deactivated with the WEM. (To not interfere with the phase switching) With and without the WEM, the charger will disconnect CP and reconnect it four seconds later if the car does not react in 30 seconds.

Kishi85, 2024-02-26:
> ... the charger will disconnect CP and reconnect it four seconds later if the car does not react in 30 seconds.

But only if EV Wakeup is configured in Charge Settings, correct?

This doesn't seem to work at times for some cars (maybe the 4 second wait is indeed not enough). What seems to work for me in these cases is stopping and restarting the charge via Warp's WebUI but that might be a coincidence. Hence my fallback suggestion to use stop_charging, followed by an immediate start_charging for the wakeup feature (maybe as a configuration option for EVCC?).

rtrbt, 2024-02-26:
>But only if EV Wakeup is configured in Charge Settings, correct?

Correct.

>maybe the 4 second wait is indeed not enough

We are currently (as of half an hour ago) testing a new firmware that retrys the CP disconnect a second time, but for 30 seconds, if the 4 second disconnect did not wake up the car. If you want to, you can test this firmware:
[warp2_firmware-NIGHTLY_2_2_1_65dca55f_2aa5b51ad4b2067_merged.bin.zip](https://github.com/evcc-io/evcc/files/14406333/warp2_firmware-NIGHTLY_2_2_1_65dca55f_2aa5b51ad4b2067_merged.bin.zip)
(Extract it first. Github does not allow attaching .bin files, but .zip works)

deadrabbit87, 2024-02-26:
Is this firmware also working with the custom firmware for the wem you have published me at the e208 thread?

andig, 2024-02-27:
Seems this request can be replaced by better Warp firmware- tentatively closing.

closed 2024-02-27 by andig