evcc-io/evcc #12488: Huawei Luna2000 always shows battery charging

smoki3, 2024-02-26:
### Describe the bug

I am using a Huawei Sun2000 inverter with the Lun2000 battery and the Fusion Wallbox. The second configured PV is a 800Watt Mini-PV

The problem, that evcc is always showing battery charing from around 115 to 118 Watt:

[image]

The Total grid consumption is correct. But the house consumption is wrong because the always says the it battery is loading.

### Steps to reproduce

Going to UI and check the values

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 45s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

mqtt:
  broker: 192.168.10.50:1883
  topic: evcc
  user: smoki3
  password: hgfresq123

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# meter definitions
meters:
  - name: grid
    type: template
    template: huawei-dongle-powersensor
    usage: grid
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.10.50 # Hostname
    port: 502 # Port
    timeout: 45s # optional

  - name: my_pv
    type: template
    template: huawei-dongle-powersensor
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.10.50 # Hostname
    port: 502 # Port
    timeout: 45s # optional

  - name: balkon_pv
    type: custom
    energy:
      source: mqtt
      topic: solar/114483190201/0/yieldtotal
    power:
      source: mqtt
      topic: solar/114483190201/0/power

  - name: my_battery
    type: template
    template: huawei-dongle-powersensor
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.10.50 # Hostname
    port: 502 # Port
    timeout: 45s # optional

# charger definitions
chargers:
  - name: my_charger
    type: template
    template: huawei-ac
    # Modbus TCP
    modbus: tcpip
    id: 255
    host: 192.168.10.201 # Hostname
    port: 502 # Port

# site describes the EVU connection, PV and home battery
site:
  title: Home # display name for UI
  meters:
    grid: grid
    pv:
      - my_pv
      - balkon_pv
    battery: my_battery
  residualPower: 100

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Wallbox
    charger: my_charger
    mode: pv
    phases: 0

# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 0.3763 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # EUR/kWh
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/02/26 14:42:51 evcc 0.124.4
[main  ] INFO 2024/02/26 14:42:51 using config file: /config/evcc.yaml
[main  ] INFO 2024/02/26 14:42:51 starting ui and api at :7070
[mqtt  ] INFO 2024/02/26 14:42:51 connecting evcc-403656787 at tcp://192.168.10.50:1883
[site  ] INFO 2024/02/26 14:42:51 site config:
[site  ] INFO 2024/02/26 14:42:51   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/02/26 14:42:51     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/02/26 14:42:51     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/02/26 14:42:51     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/02/26 14:42:51     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[lp-1  ] INFO 2024/02/26 14:42:51 loadpoint 1:
[lp-1  ] INFO 2024/02/26 14:42:51   mode:        pv
[lp-1  ] INFO 2024/02/26 14:42:51   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/02/26 14:42:51   meters:      charge ✓
[lp-1  ] INFO 2024/02/26 14:42:51     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2024/02/26 14:42:51 phase timer inactive
[lp-1  ] DEBUG 2024/02/26 14:42:51 pv timer inactive
[lp-1  ] ERROR 2024/02/26 14:43:51 charger: modbus: response transaction id '285' does not match request '1'
[site  ] DEBUG 2024/02/26 14:43:51 ----
[lp-1  ] ERROR 2024/02/26 14:45:08 charge meter: modbus: response transaction id '136' does not match request '4'
[site  ] DEBUG 2024/02/26 14:45:10 pv power: 52W
[site  ] DEBUG 2024/02/26 14:45:13 battery soc: 16%
[site  ] DEBUG 2024/02/26 14:45:13 battery power: -117W
[site  ] DEBUG 2024/02/26 14:45:15 grid power: 271W
[site  ] DEBUG 2024/02/26 14:45:18 grid currents: [0.99 0.71 0.46]A
[site  ] DEBUG 2024/02/26 14:45:19 site power: 254W
[lp-1  ] ERROR 2024/02/26 14:45:34 charge meter: read tcp 192.168.10.50:34424->192.168.10.201:502: i/o timeout
[lp-1  ] ERROR 2024/02/26 14:45:34 charge meter: read tcp 192.168.10.50:55008->192.168.10.201:502: read: connection reset by peer
[lp-1  ] ERROR 2024/02/26 14:45:34 charger: dial tcp 192.168.10.201:502: connect: connection refused
[site  ] DEBUG 2024/02/26 14:45:35 ----
[lp-1  ] ERROR 2024/02/26 14:46:35 charge meter: modbus: response transaction id '136' does not match request '10'
[site  ] DEBUG 2024/02/26 14:46:41 pv power: 55W
[site  ] DEBUG 2024/02/26 14:46:43 battery soc: 16%
[site  ] DEBUG 2024/02/26 14:46:43 battery power: -115W
[site  ] DEBUG 2024/02/26 14:46:45 grid power: 265W
[site  ] DEBUG 2024/02/26 14:46:47 grid currents: [0.96 0.7 0.46]A
[site  ] DEBUG 2024/02/26 14:46:48 site power: 250W
[lp-1  ] ERROR 2024/02/26 14:47:09 charge meter: read tcp 192.168.10.50:48554->192.168.10.201:502: i/o timeout
[lp-1  ] ERROR 2024/02/26 14:47:09 charge meter: read tcp 192.168.10.50:43448->192.168.10.201:502: read: connection reset by peer
[lp-1  ] ERROR 2024/02/26 14:47:09 charger: dial tcp 192.168.10.201:502: connect: connection refused
[site  ] DEBUG 2024/02/26 14:47:09 ----
[lp-1  ] ERROR 2024/02/26 14:48:42 charge meter: read tcp 192.168.10.50:53342->192.168.10.201:502: i/o timeout
[site  ] DEBUG 2024/02/26 14:49:23 pv power: 61W
[site  ] DEBUG 2024/02/26 14:49:27 battery soc: 16%
[site  ] DEBUG 2024/02/26 14:49:27 battery power: -116W
[site  ] DEBUG 2024/02/26 14:49:27 grid power: 342W
[site  ] DEBUG 2024/02/26 14:49:30 grid currents: [0.97 0.68 0.7]A
[site  ] DEBUG 2024/02/26 14:49:31 site power: 326W
[lp-1  ] ERROR 2024/02/26 14:50:31 charge meter: modbus: response transaction id '136' does not match request '17'
[lp-1  ] ERROR 2024/02/26 14:50:47 charge meter: read tcp 192.168.10.50:33696->192.168.10.201:502: i/o timeout
[lp-1  ] ERROR 2024/02/26 14:50:47 charger: read tcp 192.168.10.50:38894->192.168.10.201:502: read: connection reset by peer
[site  ] DEBUG 2024/02/26 14:50:47 ----
[lp-1  ] ERROR 2024/02/26 14:50:47 charge meter: dial tcp 192.168.10.201:502: connect: connection refused
[site  ] DEBUG 2024/02/26 14:51:48 pv power: 66W
[site  ] DEBUG 2024/02/26 14:51:50 battery soc: 16%
[site  ] DEBUG 2024/02/26 14:51:50 battery power: -116W
[site  ] DEBUG 2024/02/26 14:51:51 grid power: 310W
[site  ] DEBUG 2024/02/26 14:51:56 grid currents: [0.93 0.61 0.68]A
[site  ] DEBUG 2024/02/26 14:51:58 site power: 294W
[lp-1  ] ERROR 2024/02/26 14:52:08 charge meter: read tcp 192.168.10.50:56582->192.168.10.201:502: i/o timeout
[lp-1  ] ERROR 2024/02/26 14:53:14 charge meter: read tcp 192.168.10.50:52574->192.168.10.201:502: i/o timeout
[lp-1  ] ERROR 2024/02/26 14:53:14 charger: read tcp 192.168.10.50:59102->192.168.10.201:502: read: connection reset by peer
[site  ] DEBUG 2024/02/26 14:53:14 ----
[lp-1  ] ERROR 2024/02/26 14:53:15 charge meter: dial tcp 192.168.10.201:502: connect: connection refused
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.4

smoki3, 2024-02-26:
scheint es nicht zu akzeptieren
[image]

smoki3, 2024-02-26:
[image]
sieht nach dem Batteriewert aus

smoki3, 2024-02-26:
ich finde auch sonst keinen Wert in Homeassistant der zu den -116W passen würde

mucki12, 2024-02-26:
Mal unabhängig von den evcc Daten:
Was macht der Luna denn zum Zeitpunkt der Screenshots ganz konkret? Mit der Standardeinstellung für einen Mindest SoC (5%) müsst er ja gerade den gesamten Hausverbrauch decken und somit eigentlich entladen.
Oder hast du den Mindest-SoC höher eingestellt? Dann könnte das der Standby Verbrauch der Luna sein. Wie ich das verstanden habe, weist evcc den separat aus (HA mit der Standard Integration nicht).
Hatte das im Diskussionsforum auch schon mal angefragt, aber um so länger ich darüber nachdenke, um so logischer finde ich die Darstellung von evcc.

andig, 2024-02-26:
Wäre Klasse, statt Screenshots **IMMER** den Text zu posten. Der lässt sich weiter verarbeiten...

smoki3, 2024-02-26:
> Mal unabhängig von den evcc Daten: Was macht der Luna denn zum Zeitpunkt der Screenshots ganz konkret? Mit der Standardeinstellung für einen Mindest SoC (5%) müsst er ja gerade den gesamten [...]

Min SOC ist auf 5 gestellt aber entlädt trotzdem nur bis 15% da keine Strings bisher angeschlossen sind. Standbyverbrauch kann es nicht sein, da sonst unser Haus stellenweise nur noch 100W benötigen würde und vor der Luna istallation war er immer zwischen 280 und 400W

smoki3, 2024-02-26:
> Wäre Klasse, statt Screenshots **IMMER** den Text zu posten. Der lässt sich weiter verarbeiten...

ja wäre auch mein Plan aber über dann ssh addon von HA kann ich leider nicht kopieren. :-(

andig, 2024-02-26:
Einer der vielen Gründe warum wir für HA keinen Support geben wollen :/

smoki3, 2024-02-26:
> Ich meinte den Standbyverbrauch des Luna selber (der leider nicht unerheblich ist).

Ja würde dann aber trotzdem nicht in die Berechnung passen. Sonst müsste ja der Gesamtverbrauch anders sein.

smoki3, 2024-02-26:
> Einer der vielen Gründe warum wir für HA keinen Support geben wollen :/

Ich kann gern morgen ne andere Installation aufsetzen vermute nur das wird an dem ganzen nichts ändern

mucki12, 2024-02-26:
Verstehe ich nicht. Haus verbraucht 287W, Luna benötigt 116W also gesamt 403W. 
Zeigt denn HA zum gleichen Zeitpunkt einen Hausverbrauch von 403W an? Bei mir kann ich das jedenfalls so beobachten.

smoki3, 2024-02-26:
ja klar aber aber 403W sind der Hausverbrauch ohne die Luna. Wenn ich die Luna/Inverter vom Haus trenne ändert sich ja mein Netzverbrauch nicht und bleibt dann bei 403W sonst müss ja aufeinmal der Verbrauch um 116W runter gehen

mucki12, 2024-02-26:
Ja, wenn du die Luna vom Netz trennst, sollte der Verbrauch insgesamt um 116W sinken.

smoki3, 2024-02-26:
ja und dass passiert aber nicht. Darum wundert es micht ja um so mehr warum es in evcc angezeigt wird

mucki12, 2024-02-26:
Das heißt du schaltest den Luna hart ab und der Verbrauch wird weiterhin angezeigt? Ok, das muss ich mal durchspielen.

smoki3, 2024-02-26:
Ja einfach den FI raus. Dann ist er AllPhasig vom Haus getrennt

mucki12, 2024-02-26:
Ohne FI dürfte das Auslesen der Huawei Komponenten schwierig werden :-)
Meinte natürlich Trennung vom Sun2000

smoki3, 2024-02-26:
Habe einen Digitalen Stromzähler dort seh ich ja genau was vom Netz gezogen wird

mucki12, 2024-02-26:
Kan hier beobachten, dass es nahezu aufs Watt passt. Wenn der Stromzähler 591W anzeigt meldet evcc folgendes:
[image]

mucki12, 2024-02-26:
Habs jetzt tatsächlich für dich getestet. Dann habe ich ziemlich genau die 64W Netzbezug weniger. Passt also aus meiner Sicht.

smoki3, 2024-02-26:
Was genau meinst du? Also die 116 W meiner Battery weiß ich nicht wo sie her kommen.
Ich hab nun den ganzen Tag versucht die Huawei wall box einzubinden die Funktioniert aber immer noch nicht. Somit kann evcc sowieso nicht nutzen. Wenn ich aber bei der Fehlerbehebung helfen kann steh ich bereit

smoki3, 2024-02-27:
Vollständigkeitshalber:

```
/app # evcc meter -c /config/evcc.yaml
[main  ] INFO 2024/02/26 19:14:48 evcc 0.124.4
[main  ] INFO 2024/02/26 19:14:48 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/02/26 19:14:48 connecting evcc-189233711 at tcp://192.168.10.50:1883
grid
----
Power:          440W
Energy:         1092.4kWh
Current L1..L3: 1.15A 1.15A 0.47A

my_pv
-----
Power:  0W
Energy: 21.3kWh

balkon_pv
---------
Power:  0W
Energy: 730.3kWh

my_battery
----------
Power:  -119W
Energy: 27.9kWh
Soc:    16%
```

RTTTC, 2024-02-27:
Does it correlate with Fusion Solar data?

smoki3, 2024-02-27:
No. Fusionsolar does show battery power = 0W

smoki3, 2024-02-27:
I saw there where some huawei fixes merged yesterday. Lets see if this fix something:

https://github.com/evcc-io/evcc/commit/2b4772c567a70b6a896b123dcebf50e2a5be8d9a

RTTTC, 2024-02-27:
It could be something simple such as not correct or not the best register address used for battery in the template..
Check Huawei Modbus definition pdf and the template https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/huawei-sun2000-dongle-powersensor.yaml

Geri-G93, 2024-02-27:
mal was anderes dazu
lt. FusionSolar passt die Anzeige?
ganz ganz häufig, also so 5x die Woche in der Facebook gruppe, ist der dtsu verkehrt eingebaut verkabelt!!

smoki3, 2024-02-27:
Nein in fusion Pass ja alles