evcc-io/evcc #12492: Alfen-Wallbox: Parameter "Validity time"

VolkerK62, 2024-02-26:
Zur Info: Ich habe in der Alfen-Wallbox noch den Parameter "Validity time" gefunden. Dieser Parameter bezieht sich auf den Parameter "Safe current". Dieser war hier auf 10s eingestellt. Hab die Zeit nun mal auf 300s eingestellt und bisher läuft es stabil. Vielleicht hilft es dem ein oder anderen.
[image]

_Originally posted by @Lenus2024 in https://github.com/evcc-io/evcc/discussions/12458#discussioncomment-8591828_

VolkerK62, 2024-02-26:
@premultiply 
kann man den evtl auslesen und heartbeat daran anpassen?
Alternativ würde ich einen HInweis in die Doku aufnehmen.
Heartbeat ist 25s?
https://github.com/evcc-io/evcc/blob/d8eb167333859b4a4b174b505d658694338519bf/charger/alfen.go#L108

andig, 2024-02-26:
@VolkerK62 welches Register ist das denn?

VolkerK62, 2024-02-26:
Bin persönlich nicht betroffen.
Ich denke es ist das hier
[image]

Laut Handbuch ist der default 60s. Hier in dem Fall war 10s eingestellt (warum auch immer)

andig, 2024-03-01:
> Ich denke es ist das hier

Denke nicht- das ist die aktuell verbleibende Zeit. Wir bräuchten die maximal gültige Zeit. Das müsste ein anderes Register sein.

VolkerK62, 2024-03-01:
Ein anderes konnte ich nicht finden.
[Implementation of Modbus Slave TCPIP for Alfen NG9xx platform (2).pdf](https://github.com/evcc-io/evcc/files/14463526/Implementation.of.Modbus.Slave.TCPIP.for.Alfen.NG9xx.platform.2.pdf)

andig, 2024-03-01:
Find auch nix :( Lass uns Hinweis für die Doku machen.

VolkerK62, 2024-03-01:
mach ich

closed 2024-03-02 by andig

Tombra1889, 2024-03-04:
Mir ist noch aufgefallen, dass trotz "fast"-Mode weiterhin das ActiveLoadBalancing getriggert wird.
[image]
Demnach würde die Wallbox trotz "fast"-Mode zunächst auf einer Phase starten.

Wenn ich EVCC stoppe, zählt das Register 41209 auf 0s runter und dann gehen 41207 und 41211 auf 16A.

Aus meiner Sicht könnte im "fast"-Mode das ActiveLoadBalancing deaktiviert werden.

andig, 2024-03-05:
Es kann nur einer steuern- entweder evcc oder Loadbalancing.

Tombra1889, 2024-03-05:
Evcc nutzt die Loadbalancing-Funktion der Alfen, also evcc steuert an dieser Stelle.

PR #12593 (2024-03-02): add ValidityTime info to Alfen.yaml
fix #12492