evcc-io/evcc #12493: Porsche: Zugriff auf e-mobility Endpoint schlägt mit "unexpected status: 500 (Internal Server Error)" fehl.

MarkusGH, 2024-02-26:
### Describe the bug

Das Porsche API scheint mal wieder kaputt zu sein, nach dem Neustart ist das Fahrzeug offline:

Seltsamer Weise funktioniert aber [pyporscheconnectapi](https://github.com/CJNE/pyporscheconnectapi) einwandfrei:

porschecli emobility -v VIN liefert Ergebnisse.

Irgendwas macht evcc anders und der Server stolpert darüber.

Siehe auch:

https://github.com/evcc-io/evcc/discussions/12445#discussioncomment-8593449

### Steps to reproduce

evcc neu starten.

### Configuration details

```yaml
vehicles:
- name: ev1
  type: template
  template: porsche
  title: xxx
  user: xxx
  password: xxx
  vin: WP1ZZxxx
  capacity: 17.9  # Akku-Kapazität in kWh
  phases: 2
  minCurrent: 6
  maxCurrent: 16
  cache: 15m
```

### Log details

```text
[porsche] TRACE 2024/02/25 15:44:02 GET https://identity.porsche.com/authorize?audience=https%3A%2F%2Fapi.porsche.com&client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&code_challenge=TttP7b5kkCTtlt3rp87C9m_d71ZhywM03byFkZXhKc4&code_challenge_method=S256&redirect_uri=https%3A%2F%2Fmy.porsche.com%2F&response_type=code&scope=openid+offline_access+profile+email+pid%3Auser_profile.vehicles%3Aread&state=eicp7A6QYaJ9PcAr&ui_locales=de-DE
[porsche] TRACE 2024/02/25 15:44:02 Found. Redirecting to /u/login/identifier?state=hKFo2SBTUHc3RmRheTBob3Y0SzFFMEtWemFwUEp2dkZaSmZpTaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGdvUU1wcHZRR0otTWNSYWtUaGFZRWxkandUWXlLQWxRo2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de
[porsche] TRACE 2024/02/25 15:44:02 POST https://identity.porsche.com/u/login/identifier?state=hKFo2SBTUHc3RmRheTBob3Y0SzFFMEtWemFwUEp2dkZaSmZpTaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGdvUU1wcHZRR0otTWNSYWtUaGFZRWxkandUWXlLQWxRo2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de
[porsche] TRACE 2024/02/25 15:44:03 action=default&is-brave=false&js-available=true&state=hKFo2SBTUHc3RmRheTBob3Y0SzFFMEtWemFwUEp2dkZaSmZpTaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGdvUU1wcHZRR0otTWNSYWtUaGFZRWxkandUWXlLQWxRo2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de&username=***&webauthn-available=false&webauthn-platform-available=false
[porsche] TRACE 2024/02/25 15:44:03 POST https://identity.porsche.com/u/login/password?state=hKFo2SBTUHc3RmRheTBob3Y0SzFFMEtWemFwUEp2dkZaSmZpTaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGdvUU1wcHZRR0otTWNSYWtUaGFZRWxkandUWXlLQWxRo2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de
[porsche] TRACE 2024/02/25 15:44:03 action=default&password=***&state=hKFo2SBTUHc3RmRheTBob3Y0SzFFMEtWemFwUEp2dkZaSmZpTaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGdvUU1wcHZRR0otTWNSYWtUaGFZRWxkandUWXlLQWxRo2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de&username=***
[porsche] TRACE 2024/02/25 15:44:03 GET https://identity.porsche.com/authorize/resume?state=goQMppvQGJ-McRakThaYEldjwTYyKAlQ
[porsche] TRACE 2024/02/25 15:44:03 Found. Redirecting to https://my.porsche.com/?code=Iw_aaZzlk_x1HDEIrUOUY-HHH811A2cWy3qRcQ6D9LIBe&state=eicp7A6QYaJ9PcAr
[porsche] TRACE 2024/02/25 15:44:03 POST https://identity.porsche.com/oauth/token
[porsche] TRACE 2024/02/25 15:44:03 client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&code=Iw_aaZzlk_x1HDEIrUOUY-HHH811A2cWy3qRcQ6D9LIBe&code_verifier=ef3Pqh7qNYZjInfOTdLZZTUrjpvjX-32exG8vfoWrwo&grant_type=authorization_code&redirect_uri=https%3A%2F%2Fmy.porsche.com%2F
[porsche] TRACE 2024/02/25 15:44:03 GET https://api.porsche.com/core/api/v3/de/de_DE/vehicles
[porsche] TRACE 2024/02/25 15:44:04 [ {
[porsche] TRACE 2024/02/25 15:44:04 GET https://api.porsche.com/core/api/v3/de/de_DE/vehicles/***/pairing
[porsche] TRACE 2024/02/25 15:44:04 {
[porsche] TRACE 2024/02/25 15:44:04 GET https://api.porsche.com/e-mobility/vcs/capabilities/***
[porsche] TRACE 2024/02/25 15:44:04 An unexpected error occurred.
[main  ] ERROR 2024/02/25 15:44:04 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'porsche': unexpected status: 500 (Internal Server Error)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.4 (3bf57f87)

andig, 2024-02-26:
> porschecli emobility -v VIN liefert Ergebnisse.

Gibts davon ein http trace zum Vergleich?

andig, 2024-02-26:
Durch einen MITM Proxy wie ProxyMan schicken

MarkusGH, 2024-02-26:
pyporscheconnect hat eine Debugoption.

Habe Dir gerade den Output von `pyporscheconnect -d emobility -v VIN` per E-Mail gesendet.

mitmproxy ist schwierig, da ich nicht wüsste, wie ich pyporscheconnect in Ubuntu einen Proxy unterschieben kann.

MarkusGH, 2024-02-27:
Proposed fix: https://github.com/evcc-io/evcc/pull/12497 - hope it is really that easy...

closed 2024-02-27 by andig