evcc-io/evcc #12495: battery mode: modbus: exception '3' (illegal data value), function '16' and modbus: exception '3' (illegal data value), function '16'

HolgerLexow, 2024-02-26:
### Describe the bug

after implementation battery SMA sunny Island I get this bug and as well bug 
"modbus: exception '3' (illegal data value), function '16'"

### Steps to reproduce

1. UI configuration - Grid config changes are not possible
2.add device config sma-si-modbus with IP and modes address 3 check - all is ok (green)
3. then restart - and issue is happened see above
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  port: 7070
  host: evcc # .local suffix announces the hostname on MDNS
#  host: localhost # .local suffix announces the hostname on MDNS

log: info
levels:
  cache: error

interval: 60s # control cycle interval

sponsortoken: xxx

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: shm2  
  name: grid1

chargers:
- type: template
  template: easee 
  user: xxx 
  password: xxx  
  charger: xxx  
  name: wallbox3
- type: template
  template: easee 
  user: xxx  
  password: xxx 
  charger: xxx  
  name: wallbox4

vehicles:
- type: template
  template: vw 
  title: Passat GTE  
  user: xxx  
  password: xxx  
  vin: xxx  
  capacity: 13  
  phases: 1  
  cache: 15m  
  mode: pv  
  name: ev2
- name: mercedes
  type: mercedes
  title: Mercedes GLE 350de
  vin: xxx
 # clientId: xxx # nicht mehr supported
 # clientSecret: xxx # nicht mehr supported

loadpoints:
- title: Carport
  charger: wallbox3
  vehicle: ev2
  mode: pv
#  phases: 0
#  mincurrent: 6
#  maxcurrent: 16
#  resetOnDisconnect: true # nicht mehr supported
  enable:
    threshold: 0
    delay: 5m
  disable:
    threshold: 1000 # maximum import power (W)
    delay: 15m
- title: Garage
  charger: wallbox4
  mode: pv
#  phases: 0
#  mincurrent: 6
#  maxcurrent: 20
#  resetOnDisconnect: true # nicht mehr supported
  enable:
    threshold: 0
    delay: 5m
  disable:
    threshold: 1000 # maximum import power (W)
    delay: 15m

site:
  title: Mein Zuhause
  meters:
    grid: grid1

hems:
  type: sma
  allowcontrol: false

tariffs:
  currency: EUR # (default EUR)
  grid:
  # either
    # static grid price
    type: tibber #fixed #tibber
   # price: 0.2874 # [currency]/kWh
    token: "xxx"

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.1187 # [currency]/kWh
```

### Log details

```text
not applicable
```

### What type of operating system are you running?

Docker container

### Version

0.124.4

closed 2024-02-27 by andig

CarlosEllan, 2024-03-03:
Same bug with my configuration - request via Home assistant with the same parameters works without any errors!

andig, 2024-03-03:
Issue closed due to missing logfiles. If still a problem with nightly version please open new issue.