evcc-io/evcc #12496: Allow standbypower, heating and integrateddevice for type:custom chargers

stumbaumr, 2024-02-26:
**Is your feature request related to a problem? Please describe.**
Tried to set up AC ELWA-E as charger to display temperature, power usage and status as type: custom charger. Unfortunately custom chargers are missing the funtionality from switchsockets (static-switch mode).

**Describe the solution you'd like**
Allow chargers of type custom to use
- standbypower
- heating
- integrateddevice
- icon

**Describe alternatives you've considered**
aux meter is working fine but missing UI functionality.

stumbaumr, 2024-02-27:
Example:
Interface Script:
```
#!/bin/sh
# /config/ac-elwa-e_interface.sh -u AC-ELWA-URL -p (maxcurrent|power|soc|status)

getopt -T
if [ "$?" -ne 4 ]; then
    echo "wrong version of 'getopt' installed, exiting..." >&2
    exit 1
fi 
params="$(getopt -o u:p: -- "$@")"
eval set -- "$params"
while [ "$#" -gt 0 ]; do
    case "$1" in
    -u)
        URL=${2}
        shift 2;;
    -p)
        PARAM=${2}
        shift 2;;
    --) 
        shift
        break;;
     *) 
        echo "something else: '$1'"
        shift;;
    esac
done

DATEFILE=$(date -ur /tmp/ac-elwa-data.json +'%s' || echo 0)
DATECURRENT=$(date -ud@"$(( `date +%s`-10))" +'%s')

if [ ${DATEFILE} -lt ${DATECURRENT} ]
then
  wget -qO /tmp/ac-elwa-data.json ${URL}/data.jsn
fi

case "${PARAM}" in
  maxcurrent)
    POWER=$(egrep -o "\"power\".*[^,]*," /tmp/ac-elwa-data.json  | cut -d ',' -f1 | cut -d ':' -f2)
    echo "${POWER}/230" | bc
    ;;
  power)
    POWER=$(egrep -o "\"power\".*[^,]*," /tmp/ac-elwa-data.json  | cut -d ',' -f1 | cut -d ':' -f2)
    echo ${POWER}
    ;;
  soc)
    TEMP1=$(egrep -o "\"temp1\".*[^,]*," /tmp/ac-elwa-data.json  | cut -d ',' -f1 | cut -d ':' -f2)
    echo "scale=2; ${TEMP1}/10" | bc
    ;;
  status)
    STATE=$(egrep -o "\"status\".*[^,]*," /tmp/ac-elwa-data.json  | cut -d ',' -f1 | cut -d ':' -f2)
    case "${STATE}" in
      2)
        echo "C"
        shift;;
      3)
        echo "B"
        shift;;
      22)
        echo "B"
        shift;;
      *) 
        echo "Wrong status parse: '${STATE}'"
        shift;;
    esac
    ;;
  *)
    echo "Wrong parameter: '$1'"
    ;;
esac
exit 0
```

evcc.yaml Config:
```
meters:
  - name: AC-ELWA-E_power
    type: custom
    power:
      source: script
      cmd: "/config/ac-elwa-e_interface.sh -u http://192.168.178.23 -p power"
chargers:
  - name: AC-ELWA-E
    type: custom
    status:
      source: script
      cmd: "/config/ac-elwa-e_interface.sh -u http://192.168.178.23 -p status"
    enabled:
      source: const
      value: false
    enable:
      source: script
      cmd: "echo 0"
    maxcurrent:
      source: script
      cmd: "/config/ac-elwa-e_interface.sh -u http://192.168.178.23 -p maxcurrent"
site:
  - title: Zuhause # display name for UI
    meters:
      grid: grid # grid meter reference
      pv:
        - pv # pv meter reference
      aux:
        - AC-ELWA-E_power
vehicles:
  - name: AC-ELWA-E_soc
    type: custom
    title: Heizstab
    icon: waterheater
    soc:
      source: script
      cmd: "/config/ac-elwa-e_interface.sh -u http://192.168.178.23 -p soc"
loadpoints:
  - title: Warmwasser
    charger: AC-ELWA-E
    meter: AC-ELWA-E_power
    vehicle: AC-ELWA-E_soc
```

andig, 2024-02-27:
Does Elwa have a rest api? Could we add a template?

andig, 2024-02-27:
These are already possible:

- heating
- integrateddevice
- icon

`standbypower` would require something like a generic switch socket. Seems we don't have that yet.

stumbaumr, 2024-02-27:
> Does Elwa have a rest api? Could we add a template?

Yes, they do have a ReST API.

https://github.com/zaubererty/homeassistant-mvpv/issues/3

Unfortunately My-PV seems not consistent with their API over all their devices:
AC ELWA-E uses
- power
- temp1

whereas other devices like AC Thor use
- power_act
and more than one temperature sensor.

So you would either need a template per My-PV device or decide by the "device" value of data.jsn.
I would only want the visualisation in EVCC - the heating element management via the ModBus connection of the device directly to the SmartMeter is very stable and fast. I would not want evcc to steer the AC ELWA-E...

VolkerK62, 2024-02-27:
with your config, the power will be count twice. As AUX and as Loadpoint.

andig, 2024-02-27:
@stumbaumr sieht https://github.com/evcc-io/evcc/pull/12505 plausibel aus?