evcc-io/evcc #12508: `evcc configure` Kostal

VolkerK62, 2024-02-27:
OK habe es mir so weit selbst beantwortet. Ich habe die Windows-Version runtergeladen und würde die config damit komplett neu anlegen. Leider klappt es nicht. Sobald ich das Energy-Meter angeben will mit Port 502 (so wie es bisher funktioniert hat), kommt folgender Fehler:

```
- Hausinstallation einrichten

Wähle eines der folgenden integrierten PV-Systeme aus, oder 'Mein Gerät ist nicht in der Liste' falls es kein integriertes PV-System vorhanden ist

 Wähle ein integriertes PV-System: Kostal Plenticore Hybrid

-------------------------------------------------
Das Gerät hat die folgenden Voraussetzungen:
Nur ein System kann und darf auf den Wechselrichter zugreifen! Für die aktive Batteriesteuerung muss die externe Batteriesteuerung über Modbus mit dem Handwerkerzugang aktiviert sein.

-------------------------------------------------

Führe folgende Einstellungen durch:

 Modbus ID 71
 IP-Adresse oder Hostname wechselrichter.fritz.box
 Port 1502

Teste die pv Konfiguration von Kostal Plenticore Hybrid ...

Teste die battery Konfiguration von Kostal Plenticore Hybrid ...

Kostal Plenticore Hybrid wurde erfolgreich hinzugefügt.
 Möchtest du ein 'Kostal Smart Energy Meter (über den Wechselrichter)' Gerät als einen Netz-Stromzähler hinzufügen? Yes

Führe folgende Einstellungen durch:

 Modbus ID 71
 IP-Adresse oder Hostname 192.168.1.11
 Port 502

Teste die grid Konfiguration von Kostal Smart Energy Meter (über den Wechselrichter) ...
   Fehler: read failed: modbus: exception '2' (illegal data address), function '3'
```

_Originally posted by @Tscherno in https://github.com/evcc-io/evcc/discussions/12463#discussioncomment-8607429_

VolkerK62, 2024-02-27:
Wenn man die Frage nach "KSEM über den WR" mit "Ja" beantortet, könnte doch die die IP und der Port vom WR gleich automatisch übernommen werden, oder?

andig, 2024-02-27:
In configure stecken wir zum jetzigen Zeitpunkt keine Arbeit mehr...

closed 2024-02-27 by andig