evcc-io/evcc #12521: Preisberechnung für Awattar korrigieren

Philipp23, 2024-02-28:
**Feature request**
Die Preisberechnung für Awattar Region AT sollte an die aktuelle gültige Formel angepasst werden.

**Describe the solution you'd like**
Berechnung (Netto) lt. Awattar-Website für Tarif Hourly: (https://www.awattar.at/tariffs/hourly)
Stündliche Preise EPEX Spot ® AT +
**| Stündliche Preise EPEX Spot ® AT | * 0.03 (3%)**
+ 1,500 Cent/kWh

**Additional context**
Anpassung ggf. in tariff/embed.go durch implementieren einer zusätzlichen Methode totalPrice und Erweiterung der evcc.yaml um Parameter für Abbilung dieses Teils der Formel: **| Stündliche Preise EPEX Spot ® AT | * 0.03 (3%)**

func (t *embed) **totalPriceAwattar**(price float64) float64 {
	return (price + **math.Abs(price) * 0.03** +  t.Charges) * (1 + t.Tax)
}

andig, 2024-02-28:
Findet man diese 0,03 im Api? Gibt das nur für AT?

Philipp23, 2024-02-29:
Die API liefert leider nur den Marketprice. 
Den Kundenpreis muss man daher manuell lt. Tarif berechnen.
Und blöderweise ist die Kundenpreis-Berechnung des Tarifs "Hourly" in Österreich und Deutschland unterschiedlich.

Die oben genannte Formel gilt nur für Österreich.

andig, 2024-02-29:
Für DE gilt:

> Für HOURLY Neukunden fallen zusätzlich zum Basispreis 3% Aufschlag, sowie 1,5 Cent/kWh, jeweils zzgl. MwSt. an.

...allerdings nur für Neukunden. Ich sehe nicht, wie wir das automatisiert berechnen können. Sinnvoll wäre, wenn Awattar den echten Kundenpreis nach ihren Regeln ins API mit aufnehmen würde.

Bitte an den Awattar Support wenden- in evcc würde ich das nicht umsetzen wollen da es sich um Awattar Logik handelt.

Philipp23, 2024-02-29:
OK, dann gilt wohl für DE und AT die selbe Logik.

Wenn Awattar den tatsächlichen Kundenpreis in ihre API aufnehmen würde, hilft das nur bedingt weiter, weil es offenbar verschiedene Varianten des Tarifs "Hourly" gibt (Neukunden, Bestandskunden, ...).
Heißt also, man müsste in der EVCC-Config wieder irgendwie spezifizieren, welchen Tarif man hat, damit der korrekte Kundenpreis gefunden wird.

Alternativ könnte man aber auch die Berechnungslogik in EVCC erweitern (zusätzliche, optionale Parameter in der Config oder ein Plugin) um diesen Gap zu schließen.

Letzte Variante wäre aus meiner Sicht jedenfalls einfacher / schneller umgesetzt.

andig, 2024-02-29:
> Wenn Awattar den tatsächlichen Kundenpreis in ihre API aufnehmen würde, hilft das nur bedingt weiter, weil es offenbar verschiedene Varianten des Tarifs "Hourly" gibt (Neukunden, Bestandskunden, ...).

Auch die Info hat Awattar! Das API sollte halt die Preise für "mich" zurück geben statt irgendwas.

schenlap, 2024-03-01:
Selbst kannst du es über die Config lösen:
```
    tax: 0.23 # 20% + 3%
    charges: 0.015  # da müsste man die 3% noch runterrechnen bzw. Netzgebühren dazu
```

closed 2024-03-03 by andig

jwedenig, 2024-09-27:
Gibt es auch eine Möglichkeit bei #Feedin Awattar einzusetzen oder geht das nur bei #Grid?

Ich würde gerne den Sunny Einspeisetarif abonnieren!
Liebe Grüße
Jürgen