evcc-io/evcc #12522: Keine Restanzeige beim Laden vom VW E-Golf

fL4sH3r, 2024-02-28:
Wenn der E-Golf an die Wallbox angesteckt und geladen wird, habe ich nicht mehr die Möglichkeit mir die Restzeit anzeigen zu lassen. 
[image]

In einem anderen Adapter wird mir die Restzeit angezeigt. VW hatte letztes Jahr die App gewechselt und vorher hat es funktioniert. Wurde der Datenpunkt eventuell geändert und über EVCC nicht mehr abgefragt.

[image]

Wenn log Daten benötigt werden kann ich diese gerne noch nachliefern.

andig, 2024-02-29:
Leider gabs bisher kein Logfile anhand dessen wir das Thema hätten nachvollziehen können. Das fehlt leider auch hier.

closed 2024-02-29 by andig

fL4sH3r, 2024-02-29:
Welches Logfile wird denn benötigt dann kann ich das nachliefern ?