evcc-io/evcc #12525: Ladeverlust Statistik

desq42, 2024-02-28:
**Is your feature request related to a problem? Please describe.**
Es existieren ja für jedes Fahrzeug entsprechende "Ladekurven", in denen man die Effizienz des Ladevorgangs ablesen kann.
(Diese scheinen aber sehr generisch zu sein und von vielen Faktoren abzuhängen.)
Als relativer Newbie der Elektromobilität war ich schon sehr erschrocken über die Ladeverluste. (teilweise bis zu ~25%)

Da EVCC die Hand direkt am "Lade"-Puls hat, müsste es doch möglich sein, anhand der tatsächlichen Ladung nach Abschluss und der beim Start fehlenden Ladekapazität die Effizienz des individuellen Ladevorgangs anzuzeigen. 

**Describe the solution you'd like**
Eine weitere Spalte im "Ladevorgänge"-Bereich, der den errechneten Ladeverlust in Prozent anzeigt.

**Additional context**
Bitte den entsprechenden Prozentwert via MQTT, zwischen den jeweiligen Ladevorgängen publizieren.

panzerdev, 2024-02-28:
Die von dir angesprochenen Ladekurven betreffen wohl eher nur DC Ladevorgänge.

evcc macht ja hauptsächlich AC Ladevorgänge und die meisten Autos halten da ihr Maximum dauerhauft durch. Normal 11kw.

Was du mit Ladeverlusten meinst ist wohl von der Wallbox bis im Akku drinnen. Also was die Kabel, die Elektronik des Autos, das Ladegerät dazwischen und der Akku an Wärme beim Laden verliert.

Wenn du jetzt aus Perspektive der Wallbox 10kwh investiert hat müsstest ja um zu wissen wieviel mehr im Akku drinnen sind ganz genau wissen um wieviel kwh sich der Energieinhalt der Autos geändert hat. Ich wüsste nicht wie man das herausfinden sollte ohne zugriff auf mehr Autospezifische Daten die zb. über Canbus aus dem Auto kommen müssten.

Der ADAC hatte mal verschiedene Autos auf Ladeverluste je nach Ladeleistung AC Seitig getestet und ich glaub du kannst mal pauschal mit 7 bis 15% rechnen und liegst nicht falsch. Das genau zu bestimmen in evcc context sehe ich nicht wirklich.

Die Ver

VolkerK62, 2024-02-28:
https://www.adac.de/rund-ums-fahrzeug/elektromobilitaet/laden/ladeverluste-elektroauto-studie/

desq42, 2024-02-28:
Ja, es ging mir um das Laden an der Wallbox. Der richtige Ausdruck ist wohl eher "Wirkleistung".
https://www.smart-emotion.de/article/13-effizienz-und-netzr%C3%BCckwirkung-ed3-und-ed4/](url)

Speziell beim "langsamen" 1-phasig etc. laden ist wohl konstruktionsbedingt die Wirkleistung schlecht.
(Ist wohl auch ein spezielles Phänomen bei Smart ED4 und der Zoe, sofern man also "Min+PV" lädt kann einem die
"Wirkleistung" auch egal sein.)  Trotzdem danke für die schnelle Reaktion.