evcc-io/evcc #12534: Nach einiger Zeit Weißer Bildschirm im Chrome Android Browser

chrissooo, 2024-02-28:
### Describe the bug

Nach einer gewissen Zeit, ohne dass man eine Ursache erkennen kann (Update usw.) erscheint beim Aufruf von evcc im Chrome Browser auf einem Android Gerät eine weiße Seite. Erst wenn man den Cache im Browser leert, geht es wieder eine Weile lang ohne Probleme. Evcc wird über eine Desktop Verknüpfung (Dektop App) verwendet. Wenn der Fehler passiert, geht es aber auch direkt im Chrome Browser nicht.

Habe über USB Debugging folgende Fehler in der Developer Tools:
[image]

### Steps to reproduce

Einfach eine Weile lang benutzer, bis der Fehler auftritt. Leider keine Anhaltspukte was die Ursache ist.

### Configuration details

```yaml
spielt keine Rolle
```

### Log details

```text
keine Logs da Client Problem
```

### What type of operating system are you running?

Linux

### Version

0.124.6

chrissooo, 2024-02-28:
EVCC wird übrigens nicht im Docker verwendet, sondern ist über apt auf einem Debian installiert.

chrissooo, 2024-02-28:
@naltatis wie gewünscht eigestellt.

naltatis, 2024-02-28:
Super, danke. Mit der Meldung kommen wir dem Problem sicher näher.

PR #12546 (2024-02-29): Chrome CORS fix for private http ressources
fixes #12534

This is not a pretty solution since it disables chromes new CORS check to prevent CSRF measures. This PR only adds exceptions for static assets. So it's not critical and hopefully sufficient for fixing the issue.

For a proper and secure solution, we need to think about switching to HTTPS (self-signed) by default and/or adding authentication (WIP).
closed 2024-02-29 by andig