evcc-io/evcc #12544: evcc charger uses wrong SQLite database

Tranquility76, 2024-02-28:
### Describe the bug

When running evcc charger command, the logfile shows "using sqlite database: /root/.evcc/evcc.db" instead of /var/lib/evcc/evcc.db

/var/lib/evcc/evcc.db seems to be used when running evcc as service as this has the newer modification date.

### Steps to reproduce

1. run systemctl stop evcc
1. run evcc charger
2. look at the third line in logfile

### Configuration details

```yaml
network:
  schema: http
  host: 192.168.30.83
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ****

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.30.40
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.30.40
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery
- type: template
  template: kostal-ksem 
  id: 71  
  host: 192.168.30.42
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid

chargers:
- type: template
  template: ellipro 
  ski: ****
  ip: 192.168.30.43
  name: wallbox

loadpoints:
- title: Garage
  charger: wallbox
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: Home
  meters:
    grid: grid
    pv: pv
    battery: battery
  residualPower: -1000
  
eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          ****
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          ****
          -----END CERTIFICATE-----

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.3011

  feedin:
    type: fixed
    price: 0.08

vehicles:
  - name: ID42024
    type: template
    template: vw
    title: ID.4 2024
    user: ****
    password: ****
    vin: ****
    capacity: 77
    timeout: 10s

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE4MDEyOTk2MDAsImlhdCI6MTcwNjY5MTYwMCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IlRyYW5xdWlsaXR5NzYiLCJzcmMiOiJnaCJ9.vgdLNlUIq0v2X0Gu78OoB0FNgjR8NaIs9UQc_mugig8
```

### Log details

```text
[main  ] INFO 2024/02/28 22:47:28 evcc 0.124.4
[main  ] INFO 2024/02/28 22:47:28 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/28 22:47:28 using sqlite database: /root/.evcc/evcc.db
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.4

andig, 2024-02-28:
Service points to

    Environment="EVCC_DATABASE_DSN=/var/lib/evcc/evcc.db"

cli default is 

	Database: dbConfig{
		Type: "sqlite",
		Dsn:  "~/.evcc/evcc.db",
	},

Both seem find for me for their respective purpose.

/cc @pauxus @goebelmeier

Tranquility76, 2024-02-28:
@andig I was assuming,  service and cli use the same database as more and more settings go from evcc.yaml into the database.

andig, 2024-02-29:
You can select the config per `-c`. Unless there are strong opinions I'd say this works as expected.

closed 2024-02-29 by andig