evcc-io/evcc #12550: Smartes Netzladen im Menü verschwunden

markustenner, 2024-02-29:
### Describe the bug

Nach dem letzten Update ist der Menüeintrag für Smartes Netzladen verschwunden, was ich sehr schade finde, weil es ein schneller Absprung war.
Außerdem konnte man bisher eigentlich auch über die Anzeige des aktuellen Preises in diese Einstellungen springen, was in der aktuellen Version ebenfalls nicht mehr möglich scheint.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Xxx
```

### Log details

```text
Xxx
```

### What type of operating system are you running?

Linux

### Version

0.124.6

andig, 2024-02-29:
Den findest Du am Ladepunkt :)

markustenner, 2024-02-29:
Hi Andi,das habe ich bereits gesehen,  vermisse den Menüpunkt und den Absprung über den aktuellen Preis trotzdem,  da es intuitiv und schnell zugänglich war.Grüße Markus

naltatis, 2024-02-29:
Mein, wir können den da nicht lassen da bei mehreren Ladepunkt nicht klar ist was geöffnet werden soll.

Einen zusätzlichen globalen Dialog einführen macht nur extra Verwirrung weil wir ja mehrere Werte haben.

Konkreter Vorschlag: explizit in die releasenotes aufnehmen. Also unter der Liste

andig, 2024-02-29:
> Hi Andi,das habe ich bereits gesehen,  vermisse den Menüpunkt und den Absprung über den aktuellen Preis trotzdem,  da es intuitiv und schnell zugänglich war.

Was spricht dagegen, das dann im Falle Anzahl LP=1 aufzunehmen?

naltatis, 2024-02-29:
Komplexität. Geht hier ja "nur" darum, dass Nutzer ein neues Verhalten lernen müssen.

Wenn wir den Menüpunkt in Abhängigkeit von der Anzahl der laddpunkten einblenden müssen wir das auch erklären.

"Ich habe einen zweiten Ladepunkt angelegt. Wo ist der Menüpunkt hin?"

Wir könnten überlegen so etwas wie "What's new" nach einem Update bei Bedarf anzuzeigen. Als Dialog oder als Tour. Wenn es etwas erklärungsbedürftiges für Nutzer gibt. Das ist sicher besser zu finden als die releasenotes.

markustenner, 2024-02-29:
Also ich verstehe eure Überlegungen bzgl Installationen mit mehreren Ladepunkten aber ich werfe mal eine These in den Raum "die meisten nutzen vermutlich nur einen Ladepunkt" und wenn es mehrere Ladepunkte sind, wüsste ich nicht warum man den dynamischen Preis je Ladepunkt differenziert einstellen muss/will.

Am Ende lebe ich mit der Änderung, auch wenn ich sie aus strategischer Sicht nicht verstehe. Bitte nehmt solche substanziellen Änderungen an der Gui aber bitte irgendwo in den Changelogs/Releasenotes etc. auf.

andig, 2024-02-29:
Sind sie: https://github.com/evcc-io/evcc/releases/tag/0.124.5. Es muss halt auch gelesen werden...

markustenner, 2024-02-29:
Kam die von mir angesprochene Änderung in der 0.124.5 oder 0.124.6? Aufgefallen ist es mir nämlich erst in der 0.124.6. 
PS: Ich habe mir jetzt die Änderungen beider Versionen durchgelesen, konnte aber ehrlicherweise keinen Hinweis auf diese Änderung finden. Ich fand die Idee mit dem "Was ist neu" direkt im UI ganz charmant.

PS: Bitte meine Kritik nicht falsch verstehen, ihr macht einen mega geilen Job. Danke für eure Freizeit die ihr hier reinstreckt, ich finde das Projekt mega, mache auch überall wo es geht Werbung dafür.