evcc-io/evcc #12562: Better line description for translations

RTTTC, 2024-02-29:
It would be very helpful if there was a better description of a string to be translated.
For example: Source EN "Current"

It could have a few meanings, such as _Up to date, current_ OR _the flow of current in amps_

And thus I see that it is translated in some languages with one meaning, and with the other meaning in other languages :)

Is there a way to improve this situation?

https://hosted.weblate.org/translate/evcc/evcc/en/?checksum=a7e2e09e8d81a62e#translations

RTTTC, 2024-03-02:
(For this particular string I have used German translation as a cross-reference, as I assumed it should be correct for sure, but now I think that it was also incorrect)

naltatis, 2024-03-04:
Yes, in weblate it's possible to add a longer explanation and/or a screenshot highlighting the context in which the text is used. I've done this for a couple of screens in the beginning, but this is quite some work.

[image]

Seems like Weblate offeres the option to grand permission for screenshot management on a per-user basis. If you (or anyone else) can imagine doing this we'd gladly accept help here.

RTTTC, 2024-03-04:
How would I know what it originally means or where would it be on the screen? Let’s use my example string from above.

naltatis, 2024-03-04:
> How would I know what it originally means or where would it be on the screen?

Looking at the Source-Code :D

You example is indeed a little special because its a) from experimental config UI and b) auto-generated based on the devices (vehicle, meter, ...) capabilities.

[image]

naltatis, 2024-03-04:
The naming/grouping of the translation keys (here `[config.***]***` > Config UI) should give a pretty good hint which area the text is used in.

RTTTC, 2024-03-05:
> Looking at the Source-Code :D

Out of my league ..

(I did fix all of the wrong translations of the spring above.)

naltatis, 2024-03-05:
Ok, then I'll close it for now.

closed 2024-03-05 by naltatis