evcc-io/evcc #12563: Ladeplanung

gereic, 2024-02-29:
Hallo,

Ich hätte einen Feature-Request für die Ladeplanung.
Folgendes Problem:
Die PV-Anlage bringt das Auto am Tag nicht voll (Auto nicht zu Hause, oder zu wenig Leistung, ....)
Zusätzlich werden die Netzkosten Leistungsabhängig abgerechnet (Max. Leistung im Durchschnitt über 15min.) d.h. wenn ich mit mehr kW lade, so zahle ich mehr Grundgebühr. 
Das Auto soll um 07:00 wieder auf XX% voll geladen sein.

Lösung:
Wochenzeitschaltuhr in Ladeplaner für die Freigabe des Planers.
Im Ladeplaner wird die Abfahrtszeit eingestellt. (07:00)
Wird das Fahrzeug angesteckt, so gilt folgende Logik:
- Ist die PV-Anlage aktiv, wird mit max. Leistung aus der PV geladen.
- Ist die PV-Anlage nicht aktiv (unter Leistung XXW), so wird die Zeit bis zur Abfahrt berechnet und durch die benötigte Lademenge gerechnet. Danach wird der Ladevorgang gestartet.

Somit, wird mit minimaler Leistung geladen und die Grundgebühr niedrig gehalten.

andig, 2024-03-01:
Ich habs nicht verstanden. Du möchtest die Planladung möglichst zeitlich strecken?

gereic, 2024-03-01:
Sorry, kannte ich noch nicht.
Aber ja, Ich habe die gleiche Anforderung.
Ich möchte, dass mein Auto um 07:00 z.B.: 90% geladen ist.
Wie bei bolisaf beschrieben, habe ich die gleiche Abrechnungsmethode.
d.H.: ich möchte so wenig wie möglich Spitzenleistung beim laden haben.

Ich bin schon am überlegen, das ganze über MQTT zu implementieren.
Wäre natürlich super, wenn ich einfach die max. Leistung ([W]) per MQTT senden könnte.
Da es das aber nicht gibt, müsste ich die Phasenanzahl und die max. Stromstärke verstellen
Den Lademodus muss ich natürlich auch setzen.

Alles in allem nicht ganz trivial.
Die Frage ist einfach ob es Bestrebungen gibt dies in EVCC zu integrieren, dann kann ich mir den "Schmutztrick" über MQTT sparen.

andig, 2024-03-01:
Genau, ist im Backlog...

closed 2024-03-01 by andig

seki09, 2024-03-03:
Ähnliche Anforderungen kommen auch bald bei uns in Österreich.
Hier erfolgt bald eine Umstellung, wonach Netzgebühren nach der Peak Leistung gerechnet werden, diese kann man sich dann entsprechend zukaufen.  

Ich möchte daher anregen ob es nicht besser wäre ein zusätzliches config property einzufügen, welches dem heutigen Max current entspricht, aber nur gilt, wenn nicht per Überschuss geladen wird. 

Beispiel: 16A Max current bei pv Modus
9A Max current wenn per Planer oder smarten laden in der Nacht geladen wird. 

Somit kann jeder User steuern, wie viel Peak shaving er betreiben will (auch je nachdem wie viel Peak Leistung er erworben hat beim Netzbetreiber) und es wird nicht automatisch auf das Minimum reduziert.