evcc-io/evcc #12567: Modbus TCP i/o timeout with 0.124.6 with heidelberg wallbox

kerstinbaskakow, 2024-02-29:
### Describe the bug

i just updated to 0.124.6 and i got a every 10s a i/o timeout of my modbus tcp connection to the Heidelberg Energy & Control wall box. evcc.yaml file was not updated. Red warning symbol in the frontend. 
After downgrad to previous version 0.124.4, error disappeared. 

### Steps to reproduce

1. Update to 0.124.6
2. i/o timeout occured again
3. Downgrad to 0.124.4
4. TO disappeared. 
...

### Configuration details

```yaml
chargers:
- type: template
  template: heidelberg 
  id: 1  
  host: xxx.xxx.x.xxx  
  port: 502  
  modbus: tcpip  
  name: wallbox4
- type: template
  template: heidelberg 
  id: 2  
  host: xxx.xxx.x.xxx  
  port: 502  
  modbus: tcpip  
  name: wallbox5
```

### Log details

```text
Feb 29 16:36:09 raspberrypi evcc[13177]: [lp-2  ] ERROR 2024/02/29 16:36:09 charge meter: read tcp xxx.xxx.x.xxx:57766->xxx.xxx.x.xxx:502: i/o timeout$
Feb 29 16:36:20 raspberrypi evcc[13177]: [lp-2  ] ERROR 2024/02/29 16:36:20 charge meter: read tcp xxx.xxx.x.xxx:57850->1xxx.xxx.x.xxx:502: i/o timeout$
Feb 29 16:36:30 raspberrypi evcc[13177]: [lp-2  ] ERROR 2024/02/29 16:36:30 charge meter: read tcp xxx.xxx.x.xxx:57940->xxx.xxx.x.xxx:502: i/o timeout/$
Feb 29 16:36:40 raspberrypi evcc[13177]: [lp-2  ] ERROR 2024/02/29 16:36:40 charge meter: read tcp xxx.xxx.x.xxx:58026->xxx.xxx.x.xxx:502: i/o timeout$
Feb 29 16:36:50 raspberrypi evcc[13177]: [lp-2  ] ERROR 2024/02/29 16:36:50 charge meter: read tcp xxx.xxx.x.xxx:58118->xxx.xxx.x.xxx:502: i/o timeout$
Feb 29 16:37:00 raspberrypi evcc[13177]: [lp-2  ] ERROR 2024/02/29 16:37:00 charge meter: read tcp xxx.xxx.x.xxx:58202->1xxx.xxx.x.xxx:502: i/o timeout$
```

### What type of operating system are you running?

Linux

### Version

0.124.6

andig, 2024-02-29:
Thats a bit confusing since nothing was changed regarding Heidelberg.

StefanSchoof, 2024-03-01:
Are you able to use `git bisect` (https://www.metaltoad.com/blog/beginners-guide-git-bisect-process-elimination) to identify the commit which starts the timeouts?

VolkerK62, 2024-03-01:
Two Heidlberg Charger, but only LP-2 with error?

MarcSpring, 2024-03-01:
Hallo,
ich hänge mich hier mal dran, denke mal habe den gleichen Fehler. Es kamen keinerlei Messwerte an und der Ladevorgang ließ sich nicht starten. Habe alle möglichen Restarts erfolglos probiert.
Ein Downgrade von 124.6 auf 124.5 hat den Fehler nicht behoben. Erst ein Downgrade auf 124.4 hat den Fehler behoben.
Beim Downgrade von 124.5 auf 124.4 kam eine Fehlermeldung, dass die Config incompatibel ist. Ich habe die Abfrage ob ich die alte Version behalten will mit nein beantwortet, danach lief es wieder.
Gab es zwischen 124.4 und 124.5 eine Änderung die eine Anpassung der evcc.yaml notwendig macht? Da steht was von dass er den "checkconfig" Befehl nicht kennt?!?

[image]
[image]
[image]

andig, 2024-03-03:
i/o timeout ist *immer* ein Netzwerk- oder Geräteproblem.

closed 2024-03-03 by andig

AKInSIDE, 2024-09-20:
Hallo, ich müsste mich hier leider auch mal melden. Ich habe auch I/O Timeouts. Ich kann mittels QModMaster auf die Heidelberg zugreifen. z.B. meine Firmwareversion auslesen (1.0.8) ich kann Register lesen und schreiben z.B. Spannung auslesen, oder auch das Stanby der Heidelberg deaktivieren. Aber evcc sagt immer Timeout. Ich habe die Settings doppelt und dreifach geprüft, Port, IP Adresse des EE11 alles korrekt. bekomme aber trotzdem diesen Fehler: read tcp 192.168.178.218:47912->192.168.178.222:8888
i/o timeout.

Von der Firmware V1.0. zur 1.0.8 sind ein paar Register dazu gekommen, aber im groben sollte es das gleiche geblieben sein. Haber auch das Amperfied Setting versucht.
Auch die alte evcc Version die oben als funktionierend dargestellt wurde läuft bei mir nicht.

andig, 2024-09-20:
> Auch die alte evcc Version die oben als funktionierend dargestellt wurde läuft bei mir nicht.

...dann hast Du ein anderes Problem das nicht hier her gehört. Gerne Discussion dazu aufmachen.

AKInSIDE, 2024-09-20:
Ich muss mich Korrigieren, ich habe in der jaml Datei bei modbus noch das rs485tcpip zu nur tcpip geändert, dann macht die alte Version doch was sie soll. Hatte nur etwas länger gedauert und auf einmal ging es. Ich werde mit gleicher config später nochmal die  aktuelle Version testen und mich sonst nochmal melden. 

Erstmal danke für die Hilfe