evcc-io/evcc #12577: Keine Restzeit vorhanden

fL4sH3r, 2024-03-01:
### Describe the bug

Wenn ein Auto geladen wird dann wird mir keine Restzeit angezeigt. Früher hatte ich diese Anzeige noch. VW hatte letztes Jahr die App umgestellt eventuell haben sich ja Variablen geändert?

Die Restlaufzeit wir auf jeden fall übermittelt.
[image]

[image]
[image]

siehe auch issue #12522

### Steps to reproduce

1. Auto wird mit der Wallbox verbunden
2. EVCC startet den Ladevorgang
3. Restzeit wird beim Ladevorgang nicht angezeigt.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

plant: *************************

interval: 15s 

sponsortoken: *******************

telemetry: true

meters:
 - name: grid
   type: template
   template: sma-home-manager
   usage: grid
   host: 192.168.178.111
 - name: pv
   type: sma
   uri: 192.168.178.112
   interface: eth0
   password: "**********"
 - name: battery
   type: sma
   uri: 192.168.178.113
   interface: eth0
   password: "********"
   capacity: 9.6

chargers:
- type: template
  template: abb
  id: 1
  host: 192.168.178.158
  port: 502
  modbus: tcpip
  name: wallbox

vehicles:
- type: template
  template: id
  title: E-Golf
  user: ****************
  password: ******************
  vin: WV***************
  capacity: 35.8
  name: ev

loadpoints:
- title: Carport
  charger: wallbox
  vehicle: ev
  mode: pv
  phases: 2
  mincurrent: 6
  maxcurrent: 16

site:
  title: Mein Zuhause
  meters:
    grid: grid
    pv:
    - pv
    battery:
    - battery
  residualPower: 100

tariffs:
  currency: EUR 
  grid:
    type: fixed
    price: 0.3359
  feedin:
    type: fixed
    price: 0.07
  co2:
    type: grünstromindex
    zip: 04****

hems:
  type: sma
  allowcontrol: false
```

### Log details

```text
root@evcc:~# evcc --log debug
[main  ] INFO 2024/03/01 08:03:06 evcc 0.124.6
[main  ] INFO 2024/03/01 08:03:06 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/01 08:03:06 starting ui and api at :7070
[db    ] INFO 2024/03/01 08:03:06 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] WARN 2024/03/01 08:03:10 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/01 08:03:10 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/01 08:03:10 deprecated: phases setting is ignored, please remove
[semp  ] WARN 2024/03/01 08:03:10 SEMP_BASE_URL unspecified, using http://192.168.178.214:7070 instead
[site  ] INFO 2024/03/01 08:03:10 site config:
[site  ] INFO 2024/03/01 08:03:10   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/01 08:03:10     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/01 08:03:10     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/01 08:03:10     battery 1: power ✓ energy ✓ currents ✓ soc ✓ capacity ✓
[site  ] INFO 2024/03/01 08:03:10   vehicles:
[site  ] INFO 2024/03/01 08:03:10     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/03/01 08:03:10 loadpoint 1:
[lp-1  ] INFO 2024/03/01 08:03:10   mode:        pv
[lp-1  ] INFO 2024/03/01 08:03:10   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/03/01 08:03:10   meters:      charge ✓
[lp-1  ] INFO 2024/03/01 08:03:10     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2024/03/01 08:03:10 phase timer inactive
[lp-1  ] DEBUG 2024/03/01 08:03:10 pv timer inactive
[lp-1  ] INFO 2024/03/01 08:03:10 vehicle updated: unknown -> E-Golf
[site  ] WARN 2024/03/01 08:03:10 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/03/01 08:03:10 ----
[lp-1  ] DEBUG 2024/03/01 08:03:10 charge power: 0W
[site  ] DEBUG 2024/03/01 08:03:10 pv power: 163W
[site  ] DEBUG 2024/03/01 08:03:10 battery soc: 30%
[site  ] DEBUG 2024/03/01 08:03:10 battery power: 137W
[site  ] DEBUG 2024/03/01 08:03:10 grid meter: -1W
[site  ] DEBUG 2024/03/01 08:03:10 grid powers: [-4 24 -22]W
[site  ] DEBUG 2024/03/01 08:03:10 grid currents: [-1.43 0.602 -0.341]A
[site  ] DEBUG 2024/03/01 08:03:10 site power: 236W
[lp-1  ] DEBUG 2024/03/01 08:03:15 vehicle odometer: 28752km
[lp-1  ] DEBUG 2024/03/01 08:03:15 charge voltages: [232 234 233]V
[lp-1  ] DEBUG 2024/03/01 08:03:15 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/01 08:03:15 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/01 08:03:16 charger status: B
[lp-1  ] INFO 2024/03/01 08:03:16 car connected
[lp-1  ] DEBUG 2024/03/01 08:03:16 pv timer elapse
[lp-1  ] DEBUG 2024/03/01 08:03:16 pv timer inactive
[lp-1  ] DEBUG 2024/03/01 08:03:16 vehicle status: B
[lp-1  ] DEBUG 2024/03/01 08:03:16 vehicle soc: 31%
[lp-1  ] DEBUG 2024/03/01 08:03:16 vehicle soc limit: 0%
[lp-1  ] DEBUG 2024/03/01 08:03:16 vehicle range: 62km
[lp-1  ] DEBUG 2024/03/01 08:03:16 pv charge current: 0A = 0A + -0.342A (236W @ 3p)
[lp-1  ] DEBUG 2024/03/01 08:03:16 pv enable timer reset
[lp-1  ] DEBUG 2024/03/01 08:03:16 pv timer inactive
[site  ] DEBUG 2024/03/01 08:03:25 ----
[lp-1  ] DEBUG 2024/03/01 08:03:25 charge power: 0W
[site  ] DEBUG 2024/03/01 08:03:25 pv power: 162W
[site  ] DEBUG 2024/03/01 08:03:25 battery soc: 30%
[site  ] DEBUG 2024/03/01 08:03:25 battery power: 141W
[site  ] DEBUG 2024/03/01 08:03:25 grid meter: -2W
[site  ] DEBUG 2024/03/01 08:03:25 grid powers: [-5 25 -22]W
[site  ] DEBUG 2024/03/01 08:03:25 grid currents: [-1.42 0.607 -0.339]A
[site  ] DEBUG 2024/03/01 08:03:25 site power: 239W
[lp-1  ] DEBUG 2024/03/01 08:03:25 vehicle odometer: 28752km
[lp-1  ] DEBUG 2024/03/01 08:03:25 charge voltages: [232 234 234]V
[lp-1  ] DEBUG 2024/03/01 08:03:25 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/01 08:03:26 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/01 08:03:26 charger status: B
[lp-1  ] DEBUG 2024/03/01 08:03:26 pv charge current: 0A = 0A + -0.347A (239W @ 3p)
[site  ] DEBUG 2024/03/01 08:03:40 ----
[lp-1  ] DEBUG 2024/03/01 08:03:40 charge power: 0W
[site  ] DEBUG 2024/03/01 08:03:40 pv power: 162W
[site  ] DEBUG 2024/03/01 08:03:40 battery soc: 30%
[site  ] DEBUG 2024/03/01 08:03:40 battery power: 161W
[site  ] DEBUG 2024/03/01 08:03:40 grid meter: 6W
[site  ] DEBUG 2024/03/01 08:03:40 grid powers: [3 24 -21]W
[site  ] DEBUG 2024/03/01 08:03:40 grid currents: [1.4 0.605 -0.342]A
[site  ] DEBUG 2024/03/01 08:03:40 site power: 267W
[lp-1  ] DEBUG 2024/03/01 08:03:40 charge voltages: [233 234 233]V
[lp-1  ] DEBUG 2024/03/01 08:03:40 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/01 08:03:41 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/01 08:03:41 charger status: B
[lp-1  ] DEBUG 2024/03/01 08:03:41 pv charge current: 0A = 0A + -0.388A (267W @ 3p)
[site  ] DEBUG 2024/03/01 08:03:55 ----
[lp-1  ] DEBUG 2024/03/01 08:03:55 charge power: 0W
[site  ] DEBUG 2024/03/01 08:03:55 pv power: 161W
[site  ] DEBUG 2024/03/01 08:03:55 battery soc: 30%
[site  ] DEBUG 2024/03/01 08:03:55 battery power: 149W
[site  ] DEBUG 2024/03/01 08:03:55 grid meter: -2W
[site  ] DEBUG 2024/03/01 08:03:55 grid powers: [-6 24 -20]W
[site  ] DEBUG 2024/03/01 08:03:55 grid currents: [-1.41 0.596 -0.339]A
[site  ] DEBUG 2024/03/01 08:03:55 site power: 247W
[lp-1  ] DEBUG 2024/03/01 08:03:55 charge voltages: [233 234 233]V
[lp-1  ] DEBUG 2024/03/01 08:03:55 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/01 08:03:56 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/01 08:03:56 charger status: B
[lp-1  ] DEBUG 2024/03/01 08:03:56 pv charge current: 0A = 0A + -0.359A (247W @ 3p)
[site  ] DEBUG 2024/03/01 08:04:10 ----
[lp-1  ] DEBUG 2024/03/01 08:04:10 charge power: 0W
[site  ] DEBUG 2024/03/01 08:04:10 pv power: 159W
[site  ] DEBUG 2024/03/01 08:04:10 battery soc: 30%
[site  ] DEBUG 2024/03/01 08:04:10 battery power: 148W
[site  ] DEBUG 2024/03/01 08:04:10 grid meter: -7W
[site  ] DEBUG 2024/03/01 08:04:10 grid powers: [-10 23 -20]W
[site  ] DEBUG 2024/03/01 08:04:10 grid currents: [-1.43 0.593 -0.354]A
[site  ] DEBUG 2024/03/01 08:04:10 site power: 241W
[lp-1  ] DEBUG 2024/03/01 08:04:10 charge voltages: [233 234 233]V
[lp-1  ] DEBUG 2024/03/01 08:04:10 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/01 08:04:11 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/01 08:04:11 charger status: B
[lp-1  ] DEBUG 2024/03/01 08:04:11 pv charge current: 0A = 0A + -0.35A (241W @ 3
```

### What type of operating system are you running?

Linux

### Version

0.124.6

fL4sH3r, 2024-03-01:
Kann es sein das VW keine SOC Limit mehr mitgibt ? Und dadurch keine Berechnungen stattfinden ? 
[image]

schenlap, 2024-03-02:
selbiges bei audi:
```
audi
----
Soc:            100%
Capacity:       8.5kWh
Charge status:  B
Range:          29km
Odometer:       112284km
Finish time:    2024-03-02 11:43:00 +0100 CET
Climate active: false
Target Soc:     0%
Features:       []
```

fL4sH3r, 2024-03-05:
Hat denn keiner eine Idee ?

fL4sH3r, 2024-03-05:
Hier ist noch ein Auszug von der api/state.

Dabei fällt auf das folgende Werte komplett fehlen  

**-chargeRemainingDuration
-chargeRemainingEnergy** 

Kann deshalb keine Berechnung der Restzeit erfolgen ? 

`{

	"result": {
		"auth": {
			"vehicles": {}
		},
		"battery": [
			{
				"power": 4897,
				"energy": 3683.541,
				"soc": 6,
				"capacity": 9.6,
				"controllable": false
			}
		],
		"batteryCapacity": 9.6,
		"batteryConfigured": true,
		"batteryDischargeControl": false,
		"batteryEnergy": 3683.541,
		"batteryMode": "unknown",
		"batteryPower": 4897,
		"batterySoc": 6,
		"bufferSoc": 100,
		"bufferStartSoc": 0,
		"currency": "EUR",
		"greenShareHome": 1,
		"greenShareLoadpoints": 0.6905282590173246,
		"gridConfigured": true,
		"gridCurrents": [
			-5.848,
			15.267,
			-0.586
		],
		"gridEnergy": 2016.5994,
		"gridPower": 2153.4,
		"gridPowers": [
			-1249,
			3519,
			-116.60000000000001
		],
		"homePower": 421.39999999999964,
		"loadpoints": [
			{
				"chargeCurrent": 16,
				"chargeCurrents": [
					15.37,
					15.49,
					0
				],
				"chargeDuration": 180000000000,
				"chargePower": 7045,
				"chargeVoltages": [
					231.1,
					229.6,
					235.3
				],
				"chargedEnergy": 317.00000000000006,
				"chargerFeatureHeating": false,
				"chargerFeatureIntegratedDevice": false,
				"chargerIcon": null,
				"chargerPhases1p3p": false,
				"chargerPhysicalPhases": null,
				"charging": true,
				"connected": true,
				"connectedDuration": 0,
				"disableThreshold": 0,
				"effectiveLimitSoc": 90,
				"effectiveMaxCurrent": 16,
				"effectiveMinCurrent": 6,
				"effectivePlanSoc": 0,
				"effectivePlanTime": "0001-01-01T00:00:00Z",
				"effectivePriority": 0,
				"enableThreshold": 0,
				"enabled": true,
				"limitEnergy": 0,
				"limitSoc": 0,
				"maxCurrent": 16,
				"minCurrent": 6,
				"mode": "minpv",
				"phaseAction": "inactive",
				"phaseRemaining": 0,
				"phasesActive": 2,
				"phasesConfigured": 3,
				"phasesEnabled": 3,
				"planEnergy": 0,
				"planOverrun": false,
				"planProjectedStart": "0001-01-01T00:00:00Z",
				"planTime": "0001-01-01T00:00:00Z",
				"priority": 0,
				"pvAction": "inactive",
				"pvRemaining": 0,
				"sessionCo2PerKWh": 37.24911128134636,
				"sessionEnergy": 317.00000000000006,
				"sessionPrice": 0.04463432421944222,
				"sessionPricePerKWh": 0.14080228460391864,
				"sessionSolarPercentage": 73.37258946825172,
				"smartCostActive": true,
				"smartCostLimit": 200,
				"title": "Carport",
				"vehicleClimaterActive": null,
				"vehicleDetectionActive": false,
				"vehicleName": "ev",
				"vehicleOdometer": 28821,
				"vehicleRange": 150,
				"vehicleSoc": 76.28407821229051,
				"vehicleTargetSoc": 0
			}
		],
		"prioritySoc": 80,
		"pv": [
			{
				"power": 416,
				"energy": 20262.059
			}
		],
		"pvConfigured": true,
		"pvEnergy": 20262.059,
		"pvPower": 416,
		"residualPower": 100,
		"siteTitle": "Mein Zuhause",
		"smartCostType": "co2",
		"sponsor": "tobias.joachim@yahoo.com",
		"sponsorTokenExpires": 0,
		"statistics": {
			"30d": {
				"avgCo2": 62.57820300376254,
				"avgPrice": 0.17220788586724914,
				"chargedKWh": 59.080999999999996,
				"solarPercentage": 61.561532204870566
			},
			"365d": {
				"avgCo2": 119.96098854183965,
				"avgPrice": 0.24409542166311773,
				"chargedKWh": 238.71530466546352,
				"solarPercentage": 34.526283199239316
			},
			"total": {
				"avgCo2": 119.96098854183965,
				"avgPrice": 0.24409542166311773,
				"chargedKWh": 238.71530466546352,
				"solarPercentage": 34.526283199239316
			}
		},
		"tariffCo2": 151,
		"tariffCo2Home": 0,
		"tariffCo2Loadpoints": 46.73023288838398,
		"tariffFeedIn": 0.07,
		"tariffGrid": 0.3359,
		"tariffPriceHome": 0.07,
		"tariffPriceLoadpoints": 0.15228853592729338,
		"vehicles": {
			"ev": {
				"title": "E-Golf",
				"icon": "car",
				"capacity": 35.8,
				"minSoc": 10,
				"limitSoc": 90
			}
		},
		"version": "0.124.6 (0d541fc)"
	}
}`

andig, 2024-03-05:
> Hat denn keiner eine Idee ?

Klar- ins VW API schauen ob es den abgefragten Wert noch gibt!

fL4sH3r, 2024-03-05:
> > Hat denn keiner eine Idee ?
> 
> Klar- ins VW API schauen ob es den abgefragten Wert noch gibt!

Den abgefragt Wert gibt es in der api aber eventuell wurde der ja von VW umbenannt und muss neu gemappt werden.

"chargingStatus":{"value":{"carCapturedTimestamp":"2024-02-28T11:43:06Z",**"remainingChargingTimeToComplete_min":280,**"chargingState":"charging","chargeMode":"","chargeType":""}},

fL4sH3r, 2024-03-05:
Kann es eventuell hiermit zusammenhängen?
Das dort noch der Zusatz RemainingChargeTime***ToComplete_min**

[image]

fL4sH3r, 2024-03-06:
> > Hat denn keiner eine Idee ?
> 
> Klar- ins VW API schauen ob es den abgefragten Wert noch gibt!

[image]
Aus Irgend einen Grund gibt es dies Variablen nicht in der /api/state und ich habe leider keine Ahnung warum. 
Werden die nicht durch den Estimator erzeugt ?

fL4sH3r, 2024-03-07:
Wenn ich den Zeitplan aktiviere wir mir die Restzeit ebenfalls nicht angezeigt.
[image]

peregrines2, 2024-03-11:
Habe das Problem mit der fehlenden Restzeit leider ebenfalls. Bei mir in Verbindung mit Kia-Niro und Go-E Charger bzw Victron NS Wallbox. (siehe https://github.com/evcc-io/evcc/discussions/12293#discussioncomment-8640689 )
Würde mich auch sehr über eine Lösung freuen!

fL4sH3r, 2024-03-13:
Wurde durch Version 0.124.10 behoben. Vielen Dank nochmal.

closed 2024-03-13 by fL4sH3r

peregrines2, 2024-03-14:
Bei mir leider nicht. Auch unter 0.124.10 keine Restzeit :(

fL4sH3r, 2024-03-14:
> Bei mir leider nicht. Auch unter 0.124.10 keine Restzeit :(

Wie sieht denn deine Config aus? Eventuell fehlt da ja noch was .