evcc-io/evcc #12599: Tronity - Auto wird nach anschließen nicht erkannt #12292

ThiloBaWue, 2024-03-02:
### Describe the bug

Habe diesen Fehler bereits in  #12292 gemeldet gehabt, auch nach Update auf 0.124.6 ist der Fehler unverändert.
Wenn das EV über die Tronity-Integration angebunden/ausgelesen wird, und nicht über den vom Hersteller angebotenen Service, dann fällt auf, dass oftmals das Fahrzeug nicht erkannt wird und als "offline" angezeigt wird.

Dieser Fehler wird in der Diskussion https://github.com/evcc-io/evcc/discussions/2554 erwähnt und mehrfach bestätigt.

Folgendes Fehlerbild zeigt sich:

    Das EV wird z.B. über die "evcc configure" Prozedur erfolgreich eingebunden. Beim Anschließen wird das Fahrzeug aber nur gelegentlich erkannt und wird bei mir in ca. 50% der Fälle als "offline" oder Gastfahrzeug angezeigt.
    Testet man auf der Konsole mit "evcc vehicle" könnte ich und andere den "Soc" des Fahrzeuges, sowie den Status sehen, obwohl das Fahrzeug als "offline" angezeigt wird.
    Ein Neustart des Services oder des ganzen Betriebssystems, behebt im Regelfall das Problem, und das Fahrzeug wird als verbunden angezeigt.

Deshalb gehe ich davon aus, dass es weder ein Tronity-Problem ist (dieses zeigt über die App immer den SoC und das verbundensein mit dem Fahrzeug an), sondern in der Tronity-Integration begründet sein muss. Es ist meine reine Vermutung, dass es ein Timout-Problem sein könnte und irgendwann gar nicht mehr der Status über Tronity abgefragt wird.

Zum Testen habe ich die aktuellste Version von EVCC 0.124.6 verwendet. Besonders schade weil ich und andere dies gerade als Workaround für die fehlende Stellantis-Anbindung nutzen. Ich habe eine Anleitung zum erstellen von Debug-Daten erhalten und versuche hier mal etwas aussagekräftiges zu erzeugen. (Danke an @StefanSchoof
StefanSchoof )
Steps to reproduce

    Integriere das EV über die Tronity-Integration (Evcc 0.124.1)
    Schließe das Fahrzeug an die Wallbox an
    Warte auf Erkennung, leider zu oft nur als "Offline" oder "Gastfahrzeug" erkannt
    Neustart behebt das Problem
    ...
Im Log wird ab Mär 02 13:23:51 das Fahrzeug angesteckt. Danach wiederholt sich alles hoffe der aufschlussreiche Teil ist dabei

### Steps to reproduce

1. Integriere das EV über die Tronity-Integration (Evcc 0.124.1)
2. Schließe das Fahrzeug an die Wallbox an
3. Warte auf Erkennung, leider zu oft nur als "Offline" oder "Gastfahrzeug" erkannt
   > Rotes Dreieck in der EVCC GUI 
4. Neustart behebt das Problem

### Configuration details

```yaml
[main  ] INFO 2024/03/02 13:41:44 evcc 0.124.6
[main  ] INFO 2024/03/02 13:41:44 using config file: /home/pi/evcc.yaml
[db    ] INFO 2024/03/02 13:41:44 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] ERROR 2024/03/02 13:41:45 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'tronity': cannot get vehicles: Get "https://api.tronity.tech/tronity/vehicles": unexpected status: 401 (Unauthorized)
[lp-1  ] WARN 2024/03/02 13:41:45 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/02 13:41:45 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/02 13:41:45 deprecated: phases setting is ignored, please remove
[lp-2  ] WARN 2024/03/02 13:41:45 deprecated: minCurrent setting is ignored, please remove
[lp-2  ] WARN 2024/03/02 13:41:45 deprecated: maxcurrent setting is ignored, please remove
[lp-2  ] WARN 2024/03/02 13:41:45 deprecated: phases setting is ignored, please remove
[lp-1  ] DEBUG 2024/03/02 13:41:45 charge total import: 2271.447kWh
[lp-2  ] DEBUG 2024/03/02 13:41:45 charge total import: 3047.366kWh
[site  ] WARN 2024/03/02 13:41:45 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower

Konfiguration (/home/pi/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

[... 385 lines cut ...]
Mär 02 13:26:51 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 13:26:51 ----
Mär 02 13:26:51 EVCC evcc[614]: [lp-1  ] DEBUG 2024/03/02 13:26:51 charge power: 0W
Mär 02 13:26:51 EVCC evcc[614]: [lp-2  ] DEBUG 2024/03/02 13:26:51 charge power: 0W
Mär 02 13:26:51 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 13:26:51 pv power: 3129W
Mär 02 13:26:51 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 13:26:51 battery soc: 55%
Mär 02 13:26:51 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 13:26:51 battery power: -2609W
Mär 02 13:26:51 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 13:26:51 grid meter: 29W
Mär 02 13:26:52 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 13:26:52 grid currents: [0.05 0.06 0.05]A
Mär 02 13:26:52 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 13:26:52 site power: -2580W
Mär 02 13:26:52 EVCC evcc[614]: [lp-1  ] DEBUG 2024/03/02 13:26:52 charge voltages: [229 236 241]V
Mär 02 13:26:52 EVCC evcc[614]: [lp-1  ] DEBUG 2024/03/02 13:26:52 detected connected phases: 3p
Mär 02 13:26:52 EVCC evcc[614]: [lp-1  ] DEBUG 2024/03/02 13:26:52 charge currents: [0 0 0]A
Mär 02 13:26:52 EVCC evcc[614]: [lp-1  ] DEBUG 2024/03/02 13:26:52 charge total import: 2270.562kWh
Mär 02 13:26:52 EVCC evcc[614]: [lp-1  ] DEBUG 2024/03/02 13:26:52 charger status: B
Mär 02 13:26:52 EVCC evcc[614]: [main  ] ERROR 2024/03/02 13:26:52 vehicle status: unexpected status: 401 (Unauthorized)
Mär 02 13:26:52 EVCC evcc[614]: [lp-1  ] DEBUG 2024/03/02 13:26:52 pv charge current: 3.74A = 0A + 3.74A (-2580W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

0.124.6

ThiloBaWue, 2024-03-02:
Ich hoffe, diesmal nimmt sich jemand des Fehlers an, Danke! Da die Stellantis- Integration ein bestehenes Problem ist, wäre es schön wenn wenigstens die Tronity-Integration funktionieren würde. Großen Respekt für Eure Arbeit an dem Projekt, aber ich habe momentan das Gefühl, dass die Kommunikation schon mal besser war. Hoffe das das an zu viel Arbeit an zu vielen Baustellen gleichzeitig liegt. Programmiertechnisch kann ich leider hier nicht helfen, das ist über meinem Horizont.

andig, 2024-03-02:
Bitte `tronity: trace` der Fahrzeugerkennung und Zugangsdaten an info@evcc.io

andig, 2024-03-02:
OT @ThiloBaWue es wäre schön wenn Stellantis wenigstens unseren Antrag bearbeiten würde ;). Wäre Klasse wenn ihr als Kunden mal an der Stelle Druck machen würdet wo das Problem liegt...

ThiloBaWue, 2024-03-02:
@andig Die Zugangsdaten habe ich geschickt. Was bitte soll ich mit "tronity: trace" machen? Auf der Kommandozeile wird das als Kommando nicht erkannt. Auch nicht in Verbindung mit "evcc vehicle". 

Auch ich habe bereits über 2 Kanäle meinen Unmut bei Stellantis geäußert. Mal schauen.

ThiloBaWue, 2024-03-02:
Okay, ich glaube ich habe es verstanden. Habe "tronity: trace" in die evcc.yaml eingebaut. Mal schauen ob ich etwas aussagekräftiges herausbekomme......

ThiloBaWue, 2024-03-03:
Mär 02 20:03:20 EVCC systemd[1]: Stopping evcc...
Mär 02 20:03:20 EVCC systemd[1]: evcc.service: Succeeded.
Mär 02 20:03:20 EVCC systemd[1]: Stopped evcc.
Mär 02 20:03:20 EVCC systemd[1]: evcc.service: Consumed 7min 14.694s CPU time.
Mär 02 20:03:37 EVCC systemd[1]: Started evcc.
Mär 02 20:03:41 EVCC evcc[614]: [main  ] INFO 2024/03/02 20:03:41 evcc 0.124.6
Mär 02 20:03:41 EVCC evcc[614]: [main  ] INFO 2024/03/02 20:03:41 using config file: /etc/evcc.yaml
Mär 02 20:03:41 EVCC evcc[614]: [main  ] INFO 2024/03/02 20:03:41 starting ui and api at :7070
Mär 02 20:03:42 EVCC evcc[614]: [db    ] INFO 2024/03/02 20:03:42 using sqlite database: /var/lib/evcc/evcc.db
Mär 02 20:03:42 EVCC evcc[614]: [mqtt  ] INFO 2024/03/02 20:03:42 connecting evcc-470832649 at tcp://iobrokerNUC.fritz.box:1884
Mär 02 20:03:42 EVCC evcc[614]: [mqtt  ] DEBUG 2024/03/02 20:03:42 tcp://iobrokerNUC.fritz.box:1884 connected
Mär 02 20:03:42 EVCC evcc[614]: [tronity] TRACE 2024/03/02 20:03:42 POST https://api.tronity.tech/authentication
Mär 02 20:03:52 EVCC evcc[614]: [tronity] TRACE 2024/03/02 20:03:52 {"client_id":"***","client_secret":"***","grant_type":"app"}
Mär 02 20:03:52 EVCC evcc[614]: [main  ] ERROR 2024/03/02 20:03:52 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'tronity': cannot get vehicles: Get "https://api.tronity.tech/tronity/vehicles": Post "https://api.tronity.tech/authentication": context deadline exceeded (Client.Timeout exceeded while awaiting headers) (Client.Timeout exceeded while awaiting headers)
Mär 02 20:03:53 EVCC evcc[614]: [lp-1  ] DEBUG 2024/03/02 20:03:53 charge total import: 2287.945kWh
Mär 02 20:03:53 EVCC evcc[614]: [lp-2  ] DEBUG 2024/03/02 20:03:53 charge total import: 3047.366kWh
Mär 02 20:03:53 EVCC evcc[614]: [site  ] WARN 2024/03/02 20:03:53 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
Mär 02 20:03:53 EVCC evcc[614]: [site  ] INFO 2024/03/02 20:03:53 site config:
Mär 02 20:03:53 EVCC evcc[614]: [site  ] INFO 2024/03/02 20:03:53   meters:      grid ✓ pv ✓ battery ✓
Mär 02 20:03:53 EVCC evcc[614]: [site  ] INFO 2024/03/02 20:03:53     grid:      power ✓ energy ✓ currents ✓
Mär 02 20:03:53 EVCC evcc[614]: [site  ] INFO 2024/03/02 20:03:53     pv 1:      power ✓ energy ✓ currents ✗
Mär 02 20:03:53 EVCC evcc[614]: [site  ] INFO 2024/03/02 20:03:53     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
Mär 02 20:03:53 EVCC evcc[614]: [site  ] INFO 2024/03/02 20:03:53   vehicles:
Mär 02 20:03:53 EVCC evcc[614]: [site  ] INFO 2024/03/02 20:03:53     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Mär 02 20:03:53 EVCC evcc[614]: [lp-1  ] INFO 2024/03/02 20:03:53 loadpoint 1:
Mär 02 20:03:53 EVCC evcc[614]: [lp-1  ] INFO 2024/03/02 20:03:53   mode:        pv
Mär 02 20:03:53 EVCC evcc[614]: [lp-1  ] INFO 2024/03/02 20:03:53   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
Mär 02 20:03:53 EVCC evcc[614]: [lp-1  ] INFO 2024/03/02 20:03:53   meters:      charge ✓
Mär 02 20:03:53 EVCC evcc[614]: [lp-1  ] INFO 2024/03/02 20:03:53     charge:    power ✓ energy ✓ currents ✓
Mär 02 20:03:53 EVCC evcc[614]: [lp-2  ] INFO 2024/03/02 20:03:53 loadpoint 2:
Mär 02 20:03:53 EVCC evcc[614]: [lp-2  ] INFO 2024/03/02 20:03:53   mode:        pv
Mär 02 20:03:53 EVCC evcc[614]: [lp-2  ] INFO 2024/03/02 20:03:53   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
Mär 02 20:03:53 EVCC evcc[614]: [lp-2  ] INFO 2024/03/02 20:03:53   meters:      charge ✓
Mär 02 20:03:53 EVCC evcc[614]: [lp-2  ] INFO 2024/03/02 20:03:53     charge:    power ✓ energy ✓ currents ✓
Mär 02 20:03:53 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 20:03:53 set buffer soc: 25
Mär 02 20:03:53 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 20:03:53 set buffer start soc: 62.5
Mär 02 20:03:53 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 20:03:53 set priority soc: 25
Mär 02 20:03:53 EVCC evcc[614]: [lp-1  ] DEBUG 2024/03/02 20:03:53 phase timer inactive
Mär 02 20:03:53 EVCC evcc[614]: [lp-1  ] DEBUG 2024/03/02 20:03:53 pv timer inactive
Mär 02 20:03:53 EVCC evcc[614]: [lp-2  ] DEBUG 2024/03/02 20:03:53 phase timer inactive
Mär 02 20:03:53 EVCC evcc[614]: [lp-2  ] DEBUG 2024/03/02 20:03:53 pv timer inactive
Mär 02 20:03:53 EVCC evcc[614]: [site  ] WARN 2024/03/02 20:03:53 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
Mär 02 20:03:53 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 20:03:53 ----
Mär 02 20:03:53 EVCC evcc[614]: [lp-1  ] DEBUG 2024/03/02 20:03:53 charge power: 0W
Mär 02 20:03:53 EVCC evcc[614]: [lp-2  ] DEBUG 2024/03/02 20:03:53 charge power: 0W
Mär 02 20:03:53 EVCC evcc[614]: [mqtt  ] ERROR 2024/03/02 20:03:53 set evcc/loadpoints/2/minCurrent/set: invalid float value: NaN
Mär 02 20:03:53 EVCC evcc[614]: [mqtt  ] ERROR 2024/03/02 20:03:53 set evcc/loadpoints/2/maxCurrent/set: invalid float value: NaN
Mär 02 20:03:53 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 20:03:53 pv power: 0W
Mär 02 20:03:53 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 20:03:53 battery soc: 27%
Mär 02 20:03:53 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 20:03:53 battery power: 89W
Mär 02 20:03:54 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 20:03:54 grid meter: 8W
Mär 02 20:03:54 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 20:03:54 grid currents: [0.04 0.05 0.05]A
[... cut ...]
Mär 02 20:05:18 EVCC evcc[614]: [site  ] DEBUG 2024/03/02 20:05:18 site power: 94W
Mär 02 20:05:18 EVCC evcc[614]: [lp-2  ] DEBUG 2024/03/02 20:05:18 charge voltages: [232 228 232]V
Mär 02 20:05:18 EVCC evcc[614]: [lp-2  ] DEBUG 2024/03/02 20:05:18 detected connected phases: 3p
Mär 02 20:05:18 EVCC evcc[614]: [lp-2  ] DEBUG 2024/03/02 20:05:18 charge currents: [0 0 0]A
Mär 02 20:05:18 EVCC evcc[614]: [lp-2  ] DEBUG 2024/03/02 20:05:18 charge total import: 3047.366kWh
Mär 02 20:05:18 EVCC evcc[614]: [lp-2  ] DEBUG 2024/03/02 20:05:18 charger status: A

ThiloBaWue, 2024-03-03:
Erste Erkenntnis:
Habe gestern nach der Aktivierung von tronity: trace den evcc Service neu gestartet. Es kommen dann zu Beginn die Trace-Log Fehler von Tronity. Dann passiert bis am Folgetag morgens, tronity-bezogen nichts mehr. (Log-Ausschnitt)

- Er versuchte gar nicht mehr den Status von Tronity abzufragen, bis morgens wurde der Ladezustand und Status nicht erkannt
- Mich wundert der ständige Wechsel von "Chargerstatus A auf B"
Ziehe jetzt das Auto ab und teste weiter....

andig, 2024-03-03:
Es fehlt auch weiter ein trace log oder Zugangsdaten für info@evcc.io

andig, 2024-03-03:
```
Mär 02 20:03:42 EVCC evcc[614]: [tronity] TRACE 2024/03/02 20:03:42 POST https://api.tronity.tech/authentication
Mär 02 20:03:52 EVCC evcc[614]: [tronity] TRACE 2024/03/02 20:03:52 {"client_id":"","client_secret":"","grant_type":"app"}
Mär 02 20:03:52 EVCC evcc[614]: [main ] ERROR 2024/03/02 20:03:52 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'tronity': cannot get vehicles: Get "https://api.tronity.tech/tronity/vehicles": Post "https://api.tronity.tech/authentication": context deadline exceeded (Client.Timeout exceeded while awaiting headers) (Client.Timeout exceeded while awaiting headers)
```

Tronity Server antwortet nicht.

closed 2024-03-03 by andig

xbaun, 2024-03-03:
Diesen Fehler kann ich auch in meinen Logs nach einem Neustart beobachten. Der gleiche Fehler tritt auch auf wenn "last_records" abgerufen wird. Im Grunde genommen wäre hier eine Retry-Strategie mit Circuit-Breaker erforderlich, wenn es zu einem Timeout kommt.

> Mar 01 06:17:32 raspberrypi evcc[8085]: [lp-1  ] INFO 2024/03/01 06:17:32 car connected
> Mar 01 06:17:40 raspberrypi evcc[8085]: [lp-1  ] INFO 2024/03/01 06:17:40 start charging ->
> Mar 01 07:16:52 raspberrypi evcc[8085]: [lp-1  ] WARN 2024/03/01 07:16:52 vehicle soc: Get "https://api.tronity.tech/tronity/vehicles/65d82a971ee4e80013000a38/last_record": context deadline exceeded (Client.Timeout exceeded while awaiting headers) (ignored by estimator)
> Mar 01 07:16:52 raspberrypi evcc[8085]: [lp-1  ] ERROR 2024/03/01 07:16:52 vehicle range: Get "https://api.tronity.tech/tronity/vehicles/65d82a971ee4e80013000a38/last_record": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
> Mar 01 07:16:52 raspberrypi evcc[8085]: [lp-1  ] WARN 2024/03/01 07:16:52 vehicle soc: Get "https://api.tronity.tech/tronity/vehicles/65d82a971ee4e80013000a38/last_record": context deadline exceeded (Client.Timeout exceeded while awaiting headers) (ignored by estimator)
> Mar 01 07:16:52 raspberrypi evcc[8085]: [lp-1  ] ERROR 2024/03/01 07:16:52 vehicle range: Get "https://api.tronity.tech/tronity/vehicles/65d82a971ee4e80013000a38/last_record": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
> Mar 01 07:32:22 raspberrypi evcc[8085]: [lp-1  ] WARN 2024/03/01 07:32:22 vehicle soc: Get "https://api.tronity.tech/tronity/vehicles/65d82a971ee4e80013000a38/last_record": context deadline exceeded (Client.Timeout exceeded while awaiting headers) (ignored by estimator)
> Mar 01 07:32:22 raspberrypi evcc[8085]: [lp-1  ] ERROR 2024/03/01 07:32:22 vehicle range: Get "https://api.tronity.tech/tronity/vehicles/65d82a971ee4e80013000a38/last_record": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
> Mar 01 07:32:23 raspberrypi evcc[8085]: [lp-1  ] WARN 2024/03/01 07:32:23 vehicle soc: Get "https://api.tronity.tech/tronity/vehicles/65d82a971ee4e80013000a38/last_record": context deadline exceeded (Client.Timeout exceeded while awaiting headers) (ignored by estimator)
> Mar 01 07:32:23 raspberrypi evcc[8085]: [lp-1  ] ERROR 2024/03/01 07:32:23 vehicle range: Get "https://api.tronity.tech/tronity/vehicles/65d82a971ee4e80013000a38/last_record": context deadline exceeded (Client.Timeout exceeded while awaiting headers)

andig, 2024-03-03:
Du siehst ja dass retried wird. Nur kommt halt nix. Bitte Ticket bei Tronity auf machen.

ThiloBaWue, 2024-03-03:
Sorry, nach meinem Verständnis kann tronity nichts dafür. Ein Neustart von evcc hilft bisher immer. Mit dem abstecken und testen meinte ich, dass ich nach trennen und wieder anstecken wieder das Log anschaue.

ThiloBaWue, 2024-03-03:
Nach längerem Trennen des Fahrzeuges habe ich das EV wieder verbunden. Evcc bleibt bei Gastfahrzeug stehen. Im Log steht nichts von Tronity, offenbar versucht evcc nicht mal den Status abzufragen (lade das Log nicht hoch, ist ja nichts zu sehen)
Wenn ich dann Evcc neustarte (reboot) bleibt er wieder auf Gastfahrzeug stehen. Wenn ich manuell auf tronity umschalte, dann liest evcc die Werte korrekt aus.
Siehe Log:

Mär 03 12:34:41 EVCC evcc[614]: [main  ] FATAL 2024/03/03 12:34:41 evcc was stopped by user. OS should restart the service. Or restart manually.
Mär 03 12:34:41 EVCC systemd[1]: evcc.service: Succeeded.
Mär 03 12:34:41 EVCC systemd[1]: evcc.service: Consumed 18min 5.010s CPU time.
Mär 03 12:34:51 EVCC systemd[1]: evcc.service: Scheduled restart job, restart counter is at 1.
Mär 03 12:34:51 EVCC systemd[1]: Stopped evcc.
Mär 03 12:34:51 EVCC systemd[1]: evcc.service: Consumed 18min 5.010s CPU time.
Mär 03 12:34:51 EVCC systemd[1]: Started evcc.
Mär 03 12:34:52 EVCC evcc[4769]: [main  ] INFO 2024/03/03 12:34:52 evcc 0.124.6
Mär 03 12:34:52 EVCC evcc[4769]: [main  ] INFO 2024/03/03 12:34:52 using config file: /etc/evcc.yaml
Mär 03 12:34:52 EVCC evcc[4769]: [main  ] INFO 2024/03/03 12:34:52 starting ui and api at :7070
Mär 03 12:34:53 EVCC evcc[4769]: [db    ] INFO 2024/03/03 12:34:53 using sqlite database: /var/lib/evcc/evcc.db
Mär 03 12:34:53 EVCC evcc[4769]: [mqtt  ] INFO 2024/03/03 12:34:53 connecting evcc-1092346854 at tcp://iobrokerNUC.fritz.box:1884
Mär 03 12:34:53 EVCC evcc[4769]: [mqtt  ] DEBUG 2024/03/03 12:34:53 tcp://iobrokerNUC.fritz.box:1884 connected
Mär 03 12:34:54 EVCC evcc[4769]: [tronity] TRACE 2024/03/03 12:34:54 POST https://api.tronity.tech/authentication
Mär 03 12:34:54 EVCC evcc[4769]: [tronity] TRACE 2024/03/03 12:34:54 {"client_id":"***","client_secret":"***","grant_type":"app"}
Mär 03 12:34:54 EVCC evcc[4769]: --
Mär 03 12:34:54 EVCC evcc[4769]: {"id":"0a41b421-d714-4a10-ba11-04ed5d12f1fd","access_token":"eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJzdWIiOiIwYTQxYjQyMS1kNzE0LTRhMTAtYmExMS0wNGVkNWQxMmYxZmQiLCJhcHBJZCI6ImQwOTAxNWVhLTAyMGItNDM3Yi05ZDU3LWQ2MWZkYzhjOWU5OSIsInZlaGljbGVJZCI6bnVsbCwic2NvcGVzIjpbXSwicm9sZSI6InVzZXIiLCJpYXQiOjE3MDk0NjU2OTQsImV4cCI6MTcwOTU1MjA5NCwiYXVkIjoiaHR0cHM6Ly9wbGF0Zm9ybS50cm9uaXR5LmlvIiwiaXNzIjoicGxhdGZvcm0udHJvbml0eS5pbyJ9.Z6QRdQ7AKjfCY2U6gucEsjcctq-UdiIuq9jHbV0zDqY","expires_in":3600,"token_type":"bearer"}
Mär 03 12:34:54 EVCC evcc[4769]: [tronity] TRACE 2024/03/03 12:34:54 GET https://api.tronity.tech/tronity/vehicles
Mär 03 12:34:54 EVCC evcc[4769]: [tronity] TRACE 2024/03/03 12:34:54 {"data":[{"id":"65d78f16425c3e001240812a","scopes":["read_odometer","read_location","read_charge","read_battery","read_vin","read_vehicle_info","tronity_charges","tronity_charging","tronity_control_charging","tronity_factsheet","tronity_location","tronity_odometer","tronity_range","tronity_soc","tronity_vehicle_data"],"createdAt":"2024-02-22T18:14:46.340Z","updatedAt":"2024-02-22T18:14:46.340Z","vin":"VXKUKZKXZNW075397","displayName":"Mokka-e","manufacture":"Opel","model":"Mokka-e","year":2022,"kWh":47.5,"liter":null,"offline":false,"valid":true}],"count":1,"total":1,"page":1,"pageCount":1}
Mär 03 12:34:54 EVCC evcc[4769]: [lp-1  ] DEBUG 2024/03/03 12:34:54 charge total import: 2288.000kWh
Mär 03 12:34:54 EVCC evcc[4769]: [lp-2  ] DEBUG 2024/03/03 12:34:54 charge total import: 3047.366kWh
Mär 03 12:34:54 EVCC evcc[4769]: [site  ] WARN 2024/03/03 12:34:54 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
Mär 03 12:34:54 EVCC evcc[4769]: [site  ] INFO 2024/03/03 12:34:54 site config:
Mär 03 12:34:54 EVCC evcc[4769]: [site  ] INFO 2024/03/03 12:34:54   meters:      grid ✓ pv ✓ battery ✓
Mär 03 12:34:54 EVCC evcc[4769]: [site  ] INFO 2024/03/03 12:34:54     grid:      power ✓ energy ✓ currents ✓
Mär 03 12:34:54 EVCC evcc[4769]: [site  ] INFO 2024/03/03 12:34:54     pv 1:      power ✓ energy ✓ currents ✗
Mär 03 12:34:54 EVCC evcc[4769]: [site  ] INFO 2024/03/03 12:34:54     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
Mär 03 12:34:54 EVCC evcc[4769]: [site  ] INFO 2024/03/03 12:34:54   vehicles:
Mär 03 12:34:54 EVCC evcc[4769]: [site  ] INFO 2024/03/03 12:34:54     vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
Mär 03 12:34:54 EVCC evcc[4769]: [lp-1  ] INFO 2024/03/03 12:34:54 loadpoint 1:
Mär 03 12:34:54 EVCC evcc[4769]: [lp-1  ] INFO 2024/03/03 12:34:54   mode:        pv
Mär 03 12:34:54 EVCC evcc[4769]: [lp-1  ] INFO 2024/03/03 12:34:54   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
Mär 03 12:34:54 EVCC evcc[4769]: [lp-1  ] INFO 2024/03/03 12:34:54   meters:      charge ✓
Mär 03 12:34:54 EVCC evcc[4769]: [lp-1  ] INFO 2024/03/03 12:34:54     charge:    power ✓ energy ✓ currents ✓
Mär 03 12:34:54 EVCC evcc[4769]: [lp-2  ] INFO 2024/03/03 12:34:54 loadpoint 2:
Mär 03 12:34:54 EVCC evcc[4769]: [lp-2  ] INFO 2024/03/03 12:34:54   mode:        pv
Mär 03 12:34:54 EVCC evcc[4769]: [lp-2  ] INFO 2024/03/03 12:34:54   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
Mär 03 12:34:54 EVCC evcc[4769]: [lp-2  ] INFO 2024/03/03 12:34:54   meters:      charge ✓
Mär 03 12:34:54 EVCC evcc[4769]: [lp-2  ] INFO 2024/03/03 12:34:54     charge:    power ✓ energy ✓ currents ✓
Mär 03 12:34:54 EVCC evcc[4769]: [site  ] DEBUG 2024/03/03 12:34:54 set buffer soc: 25
[... cut ...]
Mär 03 12:37:54 EVCC evcc[716]: [site  ] DEBUG 2024/03/03 12:37:54 site power: -2060W
Mär 03 12:37:54 EVCC evcc[716]: [lp-2  ] DEBUG 2024/03/03 12:37:54 charge voltages: [229 239 231]V
Mär 03 12:37:54 EVCC evcc[716]: [lp-2  ] DEBUG 2024/03/03 12:37:54 detected connected phases: 3p
Mär 03 12:37:54 EVCC evcc[716]: [lp-2  ] DEBUG 2024/03/03 12:37:54 charge currents: [0 0 0]A
Mär 03 12:37:54 EVCC evcc[716]: [lp-2  ] DEBUG 2024/03/03 12:37:54 charge total import: 3047.366kWh
Mär 03 12:37:54 EVCC evcc[716]: [lp-2  ] DEBUG 2024/03/03 12:37:54 charger status: A

andig, 2024-03-03:
Probier mal bitte morgen die nightly Version. Beim app flow gabs tatsächlich einen Fehler mit dem Token Refresh!

ThiloBaWue, 2024-03-05:
Leider bin ich bis Freitag auf einem Lehrgang und kann vorher nicht testen, schade

ThiloBaWue, 2024-03-08:
Hallo, leider ist für mich mit 0.124.7 alles noch schlimmer als zuvor: Beim kontaktieren / testen von tronity hängt sich evcc komplett auf. Kein HTTP-Zugriff auf die GUI mehr möglich und Befehle wie "evcc vehicle" laufen endlos.
Was ich nun versucht habe:
1. Neustart (ändert nichts, HTTP-WebGUI startet nicht, auch keine Fehlermeldung)
2. Neue Client-ID erstellt und in evcc.yaml eingetragen + Neustart (keine Änderung)
3. Versuch neue Config mit dem Assistenten zu erstellen ( Stürzt bei testen von Tronity ab, es passiert einfach nichts)
4. Pi stromlos gemacht und damit neu gestartet (keine Änderung)
5. Downgrade auf 0.124.6 gemacht !!! Funktioniert wieder mit den bekannten Problemen !!! GUI wieder bedienbar!
    aber halt keine automatische Erkennung des Fahrzeuges und erst nach "rumspielen" kommt der SoC des EV
6. Wieder upgrade auf 0.124.7: Gleicher Fehler wie zuvor: GUI startet nicht, Befehl "evcc vehicle" läuft ohne Meldung ewig.

Hat irgend jemand den Code getestet? Ich konnte ja nicht, weil ich erst heute mittag nach hause gekommen bin? 
@andig Kannst du bitte nochmal schauen was da los ist? Danke!! Habe die Zugangsdaten aber geändert. Kann diese nochmal übermitteln wenn Du Zeit hast.

andig, 2024-03-08:
Ist im nightly behoben.

ThiloBaWue, 2024-03-08:
Danke @andig! Ein erster Test eben, sieht gut aus. Das System startet mit der neuesten Nighly wieder, und erkennt auch das Auto.