evcc-io/evcc #12600: TWC3 with Model 3 Control Client.Timeout on API request

Shuflel, 2024-03-02:
### Describe the bug

Car does not respond to any API commands evcc sends but is able to retrieve information.
Even unlocking or waking up the car does not help. Manually starting/stopping the charging or setting amperage via the tesla app works normally.

I was unable to find any information about the broken pipe error but it also disappears a few minutes after restarting evcc and only the API error remains.

### Steps to reproduce

1. Create tokens from tesla.evcc.io, allow permissions and install virtual key.
2. Add access and refresh token to config and start evcc.
3. Start charging in any mode.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: homeassistant.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: XXX
plant: XXX

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true

[... 303 lines cut ...]
[lp-1  ] DEBUG 2024/03/02 14:32:17 charge power: 0W
[site  ] DEBUG 2024/03/02 14:32:17 pv power: 9295W
[site  ] DEBUG 2024/03/02 14:32:17 battery soc: 100%
[site  ] DEBUG 2024/03/02 14:32:17 battery power: 0W
[site  ] DEBUG 2024/03/02 14:32:17 grid meter: -9022W
[site  ] DEBUG 2024/03/02 14:32:17 site power: -8922W
[lp-1  ] DEBUG 2024/03/02 14:32:17 charge voltages: [3.8 0 0.5]V
[lp-1  ] DEBUG 2024/03/02 14:32:17 charge currents: [0 0.4 0]A
[lp-1  ] DEBUG 2024/03/02 14:32:17 charger status: B
[lp-1  ] DEBUG 2024/03/02 14:32:17 next soc poll remaining time: 59m0s
[lp-1  ] DEBUG 2024/03/02 14:32:17 pv charge current: 12.9A = 0A + 12.9A (-8922W @ 3p)
[lp-1  ] DEBUG 2024/03/02 14:32:17 site power -8922W <= 0W enable threshold
[lp-1  ] DEBUG 2024/03/02 14:32:17 pv enable in 0s
[lp-1  ] DEBUG 2024/03/02 14:32:17 pv enable timer elapsed
[lp-1  ] ERROR 2024/03/02 14:32:27 max charge current 5A: Post "https://tesla.evcc.io/api/1/vehicles/XXX/command/set_charging_amps": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[site  ] ERROR 2024/03/02 14:32:27 battery mode: write tcp 192.168.188.172:42520->192.168.188.150:502: write: broken pipe
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.6

andig, 2024-03-02:
@Shuflel ist das ein temporäres oder andauernes Problem? Wenn es auftritt- funktioniert parallel ein 

    evcc vehicle --log trace,db:error -i 5

Falls nein- könntest Du bitte testweise Zugangsdaten an info@evcc.io schicken? Ggf. ist das ein Überlastungsproblem mit unserem Proxyserver.

Shuflel, 2024-03-02:
> @Shuflel ist das ein temporäres oder andauernes Problem? Wenn es auftritt- funktioniert parallel ein
> 
> ```
> evcc vehicle --log trace,db:error -i 5
> ```
> 
> Falls nein- könntest Du bitte testweise Zugangsdaten an [info@evcc.io](mailto:info@evcc.io) schicken? Ggf. ist das ein Überlastungsproblem mit unserem Proxyserver.

Es ist ein andauerndes Problem. Ich bin zwar mit docker attach in den evcc Docker reingekommen, aber habe nicht wirklich Ahnung, wie ich den Befehl dort ausführen soll.
Meinst du mit Zugangsdaten die beiden Tesla Tokens?

andig, 2024-03-02:
> Meinst du mit Zugangsdaten die beiden Tesla Tokens?

Ja bitte

> Ich bin zwar mit docker attach in den evcc Docker reingekommen, aber habe nicht wirklich Ahnung, wie ich den Befehl dort ausführen soll.

Du kannst das auch auf jedem anderen Rechner machen (oder ich schau's mir an...)

Shuflel, 2024-03-02:
> > Meinst du mit Zugangsdaten die beiden Tesla Tokens?
> 
> Ja bitte
> 
> > Ich bin zwar mit docker attach in den evcc Docker reingekommen, aber habe nicht wirklich Ahnung, wie ich den Befehl dort ausführen soll.
> 
> Du kannst das auch auf jedem anderen Rechner machen (oder ich schau's mir an...)

Klar... blöd von mir. Ich habe dir mal die Tokens und den Output des Befehls an die Mail Adresse geschickt.

andig, 2024-03-03:
@Shuflel in Deinem Fall antwortet tatsächlich das Tesla API nichts. Wir können daran erstmal nichts ändern- Issue ist unter https://github.com/teslamotors/vehicle-command/issues/200 erstellt.

closed 2024-03-03 by andig

Shuflel, 2024-03-10:
@andig Du kannst das Issue bei Tesla schließen...

Der Virtual Key war zwar auf dem Fahrzeug installiert und wurde auch unter den Schlüsseln aufgelistet, aber hat dem Wagen oder der API wohl trotzdem nicht gefallen.

1. Virtual Key gelöscht
2. Tesla mit den beiden Knöpfen am Lenkrad neu gestartet
3. Tokens neu erstellt und Virtual Key nochmal installiert

Jetzt gehen die Proxy Befehle vernünftig durch.
Danke für deine Hilfe :)

andig, 2024-03-10:
Ich lasse das mal auf- das wäre ja trotzdem ein Bug bei Tesla...