evcc-io/evcc #12607: TWC Schlüsseleinrichtung (nicht wirklich ein Bug aber wichtige Info)

peterle-k, 2024-03-02:
### Describe the bug

in der Anleitung steht folgendes:

Hinweis: Sollte der Einrichtungsprozess fehlschlagen, kann es helfen, die evcc Integration in [deinem Tesla-Konto](https://accounts.tesla.com/account-settings/security?tab=tpty-apps) zu entfernen und den Tokenerstellungsprozess erneut zu durchlaufen.

Das ist allerdings richtig kontraproduktiv. Entfernt man nämlich die App aus den Drittanbieter Apps so kann man sie nicht wieder aktivieren und man bekommt nur mehr die Nachricht, dass evcc gesperrt ist. Ich habe noch keinen Weg gefunden es zu reaktivieren - ich hoffe es geht Montag mit dem Tesla support.

### Steps to reproduce

Drittanbieter APP in Tesla Account entfernen

### Configuration details

```yaml
kein Problem
```

### Log details

```text
nicht erforderlich
```

### What type of operating system are you running?

Linux

### Version

0.124.6

mucki12, 2024-03-02:
Aber man kann doch einfach einen neuen Token für evcc erstellen (oder habe ich dich da irgendwie falsch verstanden?).
Habe ich jedenfalls bestimmt schon ca. 8x gemacht :-)

andig, 2024-03-03:
@peterle-k spannend wäre ja warum Du die entfernen musst- Tokens sollten sich auch ohne diesen Schritt neu erzeugen lassen?

peterle-k, 2024-03-03:
es ist genau so ... danke für den Tip, bin leider noch nicht auf diese Idee gekommen, hab nur versucht den TWC Schlüssel zu erneuern -> macht man den Token neu, kann man auch den Tesla Schlüssel erneuern. 
Was mir aber noch nicht klar ist: wo finde ich dann den key, den ich unter vehicle/control in evcc.yaml angeben muss?

mucki12, 2024-03-03:
Mit der neusten Version sollte der Parameter control eigentlich gar nicht mehr notwendig sein. Wenn du ihn drinnen lässt, würde ich ihn auf true stellen. Dann kannst du auch gleich mal auf das neue template umstellen (template: tesla) wie in der Doku beschrieben. 
Neben den bereits von dir hinterlegten Token hast du auch die vin angegeben. Diese ist a) nur notwendig bei mehreren Teslas und b) muss wenn sie drinnen bleibt ebenfalls genau richtig sein (ansonsten geht es auch nicht, aber die hast du ja sicherlich oben nur anonymisiert).
Neben den beiden Token muss natürlich der Fahrzeugschlüssel direkt im Tesla hinterlegt sein (letzter Schritt von tesla.evcc.io mit QR-Code und der Tesla App).