evcc-io/evcc #12608: New Mercedes Integration stopped working - Token error

rothdani, 2024-03-02:
### Describe the bug

Hi,
first of all, thank you for the effort you made to integrate Mercedes Cars again. I configured it yesterday according to the instructions shown here https://github.com/evcc-io/evcc/blob/master/templates/definition/vehicle/mercedes.yaml and it worked until today. The car was shown as offline in the evcc UI this morning and since then I was not able to get it online again. I tried first to restart the service without luck. More details in "Steps to reproduce".

Remark: I "XXX" all personal Data in the config as well as the tokens. But there are the real tokens in the config file

### Steps to reproduce

1. Added car to evcc.yaml according to template
2. generated token again via ./evcc token ev7 (that is the name of the car, so I had to use ev7 instead of mercedes)
3. restarted evcc service --> car still offline
4. Started "evcc vehicle" --> I get all information from the Car
5. restarted evcc service again --> car still offline
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
  mercedes: trace

# unique installation id
plant: XXX

interval: 30s # control cycle interval

sponsortoken: XXX

telemetry: false

meters:
- type: template
  template: victron-energy
  usage: grid
  host: 10.10.10.13
  port: 502
  name: grid1
- type: template
  template: victron-energy
  usage: pv
  host: 10.10.10.13
  port: 502
  name: pv2
- type: template
  template: victron-energy
  usage: battery
  host: 10.10.10.13
  port: 502
  name: battery3

chargers:
- type: template
  template: go-e
  host: 10.10.10.14
  name: wallbox5

vehicles:
- type: template
  template: smart
  title: Smart
  mode: minpv
  mincurrent: 9
  capacity: 17
  user: XXX
  password: XXX
  vin: W1A45339XXX
  name: ev6

- name: ev7
  type: mercedes
  account: XXX
  region: EMEA
  vin: W1KEG1BB8PFXXXX
  capacity: 89
  title: EQE DR
  tokens:
    access: XXX
    refresh: XXX

loadpoints:
- title: Parkplatz
  charger: wallbox5
  mode: pv

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
```

### Log details

```text
Mär 02 18:39:35 EVCC systemd[1]: Stopping evcc...
Mär 02 18:39:35 EVCC systemd[1]: evcc.service: Succeeded.
Mär 02 18:39:35 EVCC systemd[1]: Stopped evcc.
Mär 02 18:39:35 EVCC systemd[1]: evcc.service: Consumed 1.097s CPU time.
Mär 02 18:39:35 EVCC systemd[1]: Started evcc.
Mär 02 18:39:35 EVCC evcc[532]: [main  ] INFO 2024/03/02 18:39:35 evcc 0.124.6 (db3f355a)
Mär 02 18:39:35 EVCC evcc[532]: [main  ] INFO 2024/03/02 18:39:35 using config file: /etc/evcc.yaml
Mär 02 18:39:35 EVCC evcc[532]: [main  ] INFO 2024/03/02 18:39:35 starting ui and api at :7070
Mär 02 18:39:36 EVCC evcc[532]: [db    ] INFO 2024/03/02 18:39:36 using sqlite database: /var/lib/evcc/evcc.db
Mär 02 18:39:36 EVCC evcc[532]: [mercedes] DEBUG 2024/03/02 18:39:36 identity.NewIdentity - token not valid - Add expiry
Mär 02 18:39:36 EVCC evcc[532]: [mercedes] DEBUG 2024/03/02 18:39:36 identity.NewIdentity - token not valid - database token check started
Mär 02 18:39:36 EVCC evcc[532]: [mercedes] DEBUG 2024/03/02 18:39:36 identity.NewIdentity - token not valid - refreshToken started
Mär 02 18:39:36 EVCC evcc[532]: [mercedes] TRACE 2024/03/02 18:39:36 POST https://id.mercedes-benz.com/as/token.oauth2
Mär 02 18:39:36 EVCC evcc[532]: [mercedes] TRACE 2024/03/02 18:39:36 grant_type=refresh_token&refresh_token=XXX
Mär 02 18:39:36 EVCC evcc[532]: --
Mär 02 18:39:36 EVCC evcc[532]: {"error_description":"unknown, invalid, or expired refresh token","error":"invalid_grant"}
Mär 02 18:39:36 EVCC evcc[532]: [main  ] ERROR 2024/03/02 18:39:36 creating vehicle ev7 failed: cannot create vehicle type 'mercedes': token expired
Mär 02 18:39:37 EVCC evcc[532]: [site  ] WARN 2024/03/02 18:39:37 vehicle 'EQE DR (offline)' does not support automatic detection
Mär 02 18:39:37 EVCC evcc[532]: [site  ] INFO 2024/03/02 18:39:37 site config:
Mär 02 18:39:37 EVCC evcc[532]: [site  ] INFO 2024/03/02 18:39:37   meters:      grid ✓ pv ✓ battery ✓
Mär 02 18:39:37 EVCC evcc[532]: [site  ] INFO 2024/03/02 18:39:37     grid:      power ✓ energy ✗ currents ✗
Mär 02 18:39:37 EVCC evcc[532]: [site  ] INFO 2024/03/02 18:39:37     pv 1:      power ✓ energy ✗ currents ✗
Mär 02 18:39:37 EVCC evcc[532]: [site  ] INFO 2024/03/02 18:39:37     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
Mär 02 18:39:37 EVCC evcc[532]: [site  ] INFO 2024/03/02 18:39:37   vehicles:
Mär 02 18:39:37 EVCC evcc[532]: [site  ] INFO 2024/03/02 18:39:37     vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
Mär 02 18:39:37 EVCC evcc[532]: [site  ] INFO 2024/03/02 18:39:37     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Mär 02 18:39:37 EVCC evcc[532]: [lp-1  ] INFO 2024/03/02 18:39:37 loadpoint 1:
Mär 02 18:39:37 EVCC evcc[532]: [lp-1  ] INFO 2024/03/02 18:39:37   mode:        pv
Mär 02 18:39:37 EVCC evcc[532]: [lp-1  ] INFO 2024/03/02 18:39:37   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Mär 02 18:39:37 EVCC evcc[532]: [lp-1  ] INFO 2024/03/02 18:39:37   meters:      charge ✓
Mär 02 18:39:37 EVCC evcc[532]: [lp-1  ] INFO 2024/03/02 18:39:37     charge:    power ✓ energy ✓ currents ✓
Mär 02 18:39:37 EVCC evcc[532]: [lp-1  ] INFO 2024/03/02 18:39:37 car connected
Mär 02 18:39:55 EVCC evcc[532]: [lp-1  ] INFO 2024/03/02 18:39:55 vehicle updated: unknown -> EQE DR (offline)
Mär 02 18:40:40 EVCC systemd[1]: Stopping evcc...
Mär 02 18:40:40 EVCC systemd[1]: evcc.service: Succeeded.
Mär 02 18:40:40 EVCC systemd[1]: Stopped evcc.
Mär 02 18:40:40 EVCC systemd[1]: Started evcc.
Mär 02 18:40:40 EVCC evcc[554]: [main  ] INFO 2024/03/02 18:40:40 evcc 0.124.6 (db3f355a)
Mär 02 18:40:40 EVCC evcc[554]: [main  ] INFO 2024/03/02 18:40:40 using config file: /etc/evcc.yaml
Mär 02 18:40:40 EVCC evcc[554]: [main  ] INFO 2024/03/02 18:40:40 starting ui and api at :7070
Mär 02 18:40:40 EVCC evcc[554]: [db    ] INFO 2024/03/02 18:40:40 using sqlite database: /var/lib/evcc/evcc.db
Mär 02 18:40:40 EVCC evcc[554]: [mercedes] DEBUG 2024/03/02 18:40:40 identity.NewIdentity - token not valid - Add expiry
Mär 02 18:40:40 EVCC evcc[554]: [mercedes] DEBUG 2024/03/02 18:40:40 identity.NewIdentity - token not valid - database token check started
Mär 02 18:40:40 EVCC evcc[554]: [mercedes] DEBUG 2024/03/02 18:40:40 identity.NewIdentity - token not valid - refreshToken started
Mär 02 18:40:40 EVCC evcc[554]: [mercedes] TRACE 2024/03/02 18:40:40 POST https://id.mercedes-benz.com/as/token.oauth2
Mär 02 18:40:40 EVCC evcc[554]: [mercedes] TRACE 2024/03/02 18:40:40 grant_type=refresh_token&refresh_token=XXX
Mär 02 18:40:40 EVCC evcc[554]: --
Mär 02 18:40:40 EVCC evcc[554]: {"error_description":"unknown, invalid, or expired refresh token","error":"invalid_grant"}
Mär 02 18:40:40 EVCC evcc[554]: [main  ] ERROR 2024/03/02 18:40:40 creating vehicle ev7 failed: cannot create vehicle type 'mercedes': token expired
Mär 02 18:40:41 EVCC evcc[554]: [site  ] WARN 2024/03/02 18:40:41 vehicle 'EQE DR (offline)' does not support automatic detection
Mär 02 18:40:41 EVCC evcc[554]: [site  ] INFO 2024/03/02 18:40:41 site config:
Mär 02 18:40:41 EVCC evcc[554]: [site  ] INFO 2024/03/02 18:40:41   meters:      grid ✓ pv ✓ battery ✓
Mär 02 18:40:41 EVCC evcc[554]: [site  ] INFO 2024/03/02 18:40:41     grid:      power ✓ energy ✗ currents ✗
Mär 02 18:40:41 EVCC evcc[554]: [site  ] INFO 2024/03/02 18:40:41     pv 1:      power ✓ energy ✗ currents ✗
Mär 02 18:40:41 EVCC evcc[554]: [site  ] INFO 2024/03/02 18:40:41     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
Mär 02 18:40:41 EVCC evcc[554]: [site  ] INFO 2024/03/02 18:40:41   vehicles:
Mär 02 18:40:41 EVCC evcc[554]: [site  ] INFO 2024/03/02 18:40:41     vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
Mär 02 18:40:41 EVCC evcc[554]: [site  ] INFO 2024/03/02 18:40:41     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Mär 02 18:40:41 EVCC evcc[554]: [lp-1  ] INFO 2024/03/02 18:40:41 loadpoint 1:
Mär 02 18:40:41 EVCC evcc[554]: [lp-1  ] INFO 2024/03/02 18:40:41   mode:        pv
Mär 02 18:40:41 EVCC evcc[554]: [lp-1  ] INFO 2024/03/02 18:40:41   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Mär 02 18:40:41 EVCC evcc[554]: [lp-1  ] INFO 2024/03/02 18:40:41   meters:      charge ✓
Mär 02 18:40:41 EVCC evcc[554]: [lp-1  ] INFO 2024/03/02 18:40:41     charge:    power ✓ energy ✓ currents ✓
Mär 02 18:40:41 EVCC evcc[554]: [lp-1  ] INFO 2024/03/02 18:40:41 car connected
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.6 (db3f355a)

andig, 2024-03-03:
/cc @ReneNulschDE

ReneNulschDE, 2024-03-03:
Could you please check first if the token was stored in the database?
please execute `evcc settings get` and check if an entry with the key `mercedes-[account]-[region]` is present. Please share the first 3 lines of the output too.

rothdani, 2024-03-03:
Did evcc settings get. mercedes-[account]-[region] is present. See the first 3 lines of the output:
`[main  ] INFO 2024/03/03 12:24:07 evcc 0.124.6 (db3f355a)
[main  ] INFO 2024/03/03 12:24:07 using config file: /etc/evcc.yaml
[db    ] INFO 2024/03/03 12:24:07 using sqlite database: /home/rothdani/.evcc/evcc.db`

I also cross-checked the tokens which were shown when executing 'evcc settings get' with the latest ones in the evcc.yaml.
The Access Token is the same, the Refresh token is not matching the one in evcc.yaml

m0ndkind, 2024-03-03:
I have the same problem here on the current nightly build: access-token is correct, refresh-token is wrong. If I do "evcc vehicle" it shows an error "token expired". Doing "sudo evcc vehicle" with the same config-file shows the correct SoC and everything without any errors.

andig, 2024-03-03:
Please share trace log. 

> refresh-token is wrong

Why wrong? When the tokens get updated, both AT and RT can change.

m0ndkind, 2024-03-03:
Like rothdani said: the AT in the database and the one in my config-file are the same, RT in the database and in my config-file are different. Can send a trace log later.

ReneNulschDE, 2024-03-03:
The tokens have to be different. After the first start a new token is requested and stored in the database. The component is using always the DB-tokens. I need to check why different paths to the db are used when "evcc" and "evcc settings get" are executed.

m0ndkind, 2024-03-03:
trace logs: 1st as normal user showing the error and no car info....2nd as su as proof of concept...tokens are working! 

1. 

hoobs@hoobs:~$ evcc vehicle Mercedes --log trace
[main  ] INFO 2024/03/03 16:29:41 evcc 0.124.6 (b2921078)
[main  ] INFO 2024/03/03 16:29:41 using config file: /home/hoobs/evcc.yaml
[db    ] INFO 2024/03/03 16:29:41 using sqlite database: /home/hoobs/.evcc/evcc.db
[db    ] TRACE 2024/03/03 16:29:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT * FROM `settings` 1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT       fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT       fk_devices_details      %") -1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%      device_id       %") -1 <nil>
[db    ] TRACE 2024/03/03 16:29:41 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[mercedes] DEBUG 2024/03/03 16:29:41 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/03/03 16:29:41 identity.NewIdentity - token not valid - database token check started
[mercedes] DEBUG 2024/03/03 16:29:41 identity.NewIdentity - token not valid - refreshToken started
[mercedes] TRACE 2024/03/03 16:29:41 POST https://id.mercedes-benz.com/as/token.oauth2
[mercedes] TRACE 2024/03/03 16:29:42 grant_type=refresh_token&refresh_token=53CkahqRng2jtLOQjbipW9NS5RzqKsztAFT.HTlsEL
--
{"error_description":"unknown, invalid, or expired refresh token","error":"invalid_grant"}
[main  ] ERROR 2024/03/03 16:29:42 creating vehicle Mercedes failed: cannot create vehicle type 'mercedes': token expired
.Soc:     vehicle not available: cannot create vehicle type 'mercedes': token expired
Capacity: 13.5kWh
Features: [Offline]

2.

[main  ] INFO 2024/03/03 16:27:10 evcc 0.124.6 (b2921078)
[main  ] INFO 2024/03/03 16:27:10 using config file: /home/hoobs/evcc.yaml
[db    ] INFO 2024/03/03 16:27:10 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/03/03 16:27:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/03/03 16:27:10 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/03 16:27:10 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/03 16:27:10 SELECT * FROM `settings` 7 <nil>
[db    ] TRACE 2024/03/03 16:27:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/03/03 16:27:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/03/03 16:27:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/03/03 16:27:10 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/03 16:27:10 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/03 16:27:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[... cut ...]
departureTimeMod����P�����1 Z,
precondAtDepartureDisabl����P�����1(Z

precondNoч��P�У��1(Z%
flipWindowStatu��̑P��ջ�/@Z2
distanceZEStar��P����1Z41.3�9fffff�D@Z&
drivenTimeZERese��P����1 ǤZ)
windowstatusfrontrigh����P�����1 Z 
precondActiv����P�����1(Z
maxSo����P�����1�@Z
chargin����P��Ӧ�1(Z"
od�P����1Z24282� ڽZ%
sunroofEventActiv����P�����1(
Soc:           95%
Capacity:      13.5kWh
Charge status: B
Range:         31km
Odometer:      24282km
Finish time:   2024-03-04 00:00:00 +0100 CET
Position:      0,0
Features:      []

ReneNulschDE, 2024-03-03:
This looks like 2 different databases are used. Trace one is using /home/hoobs/.evcc/evcc.db and trace two /root/.evcc/evcc.db

The token set that is generated with evcc token is only valid for one user instance.

m0ndkind, 2024-03-03:
Argh...generated the tokens like the user did in
 #https://github.com/evcc-io/evcc/discussions/12581#discussioncomment-8641615
worked for him but not for me...

ReneNulschDE, 2024-03-03:
The user used "sudo"... this is only valid if you run evcc always with sudo.

In your case: just generate a new token set and copy it in your evcc.yaml. then start evcc without sudo.

m0ndkind, 2024-03-03:
Did so, but:
hoobs@hoobs:~$ evcc vehicle Mercedes --log trace
[main  ] INFO 2024/03/03 16:57:47 evcc 0.124.6 (b2921078)
[main  ] INFO 2024/03/03 16:57:47 using config file: /home/hoobs/evcc.yaml
[db    ] INFO 2024/03/03 16:57:48 using sqlite database: /home/hoobs/.evcc/evcc.db
[db    ] TRACE 2024/03/03 16:57:48 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT * FROM `settings` 1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT       fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT       fk_devices_details      %") -1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%      device_id       %") -1 <nil>
[db    ] TRACE 2024/03/03 16:57:48 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[mercedes] DEBUG 2024/03/03 16:57:48 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/03/03 16:57:48 identity.NewIdentity - token not valid - database token check started
[mercedes] DEBUG 2024/03/03 16:57:48 identity.NewIdentity - token not valid - refreshToken started
[mercedes] TRACE 2024/03/03 16:57:48 POST https://id.mercedes-benz.com/as/token.oauth2
[mercedes] TRACE 2024/03/03 16:57:48 grant_type=refresh_token&refresh_token=53CkahqRng2jtLOQjbipW9NS5RzqKsztAFT.HTlsEL
--
{"error_description":"unknown, invalid, or expired refresh token","error":"invalid_grant"}
[main  ] ERROR 2024/03/03 16:57:48 creating vehicle Mercedes failed: cannot create vehicle type 'mercedes': token expired
.Soc:     vehicle not available: cannot create vehicle type 'mercedes': token expired
Capacity: 13.5kWh
Features: [Offline]

ReneNulschDE, 2024-03-03:
Can you execute evcc settings get (without sudo) and check if a mercedes-.... entry is existing

m0ndkind, 2024-03-03:
Yes it exists with the right email for mercedes me.
As I said I just generated a new refresh token but in /home/hoobs/.evcc/evcc.db (or evcc settings get) it is not refreshed and has an "expiry":"2024-02-29T23:55:18.818864482+01:00"

ReneNulschDE, 2024-03-03:
We have to delete this setting. Give me a few minutes, I have to check if we can remove this with evcc settings set. (In case you are a Pro you can just delete the record from the db.)

m0ndkind, 2024-03-03:
As the beautiful formatting of my GitHub-comments clearly show I'm not a Pro and will wait for your reply :-)

ReneNulschDE, 2024-03-03:
You could remove the setting with
`evcc settings set mercedes.[account]-[region] ""`
replace [account] and [region] with your values.
example: `evcc settings set mercedes.testaccount@something-EMEA ""`

restart evcc

m0ndkind, 2024-03-03:
I deleted the wrong settings in the database. Now "evcc vehicle" works fine. So far so good but to my surprise the system service evcc uses /var/lib/evcc/evcc.db as the source...that makes debugging a bit complicated. I never changed the default!

ReneNulschDE, 2024-03-03:
Yeah, this is not optimal. You can change your evcc.yaml and set the path to /var/lib/evcc/evcc.db

ReneNulschDE, 2024-03-03:
@rothdani : Could you please read the discussion and do the same?

andig, 2024-03-03:
@ReneNulschDE is this an actual issue or more a "how to discussion" then I'd move it to discussions?

ReneNulschDE, 2024-03-03:
@andig : Discussion...