evcc-io/evcc #12611: evcc freezes when one inverter does not react

JulianOhm, 2024-03-02:
### Describe the bug

after sunset my deye 600w BKW is not reachable anylonger. in earlier versions that was no problem. my huawei inverter is still online and works with the luna battery

but now (0.124.6) evcc stops working when the inverter is not reachable. After sunrise, when enough sun is on the deye inverter to start, then evcc will work as it should. in earlier versions it just gave a warning

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 6b9621d271971d6ca2326352835c10d327851afc74103be75bfd3cc3a2bc76b3

interval: 30s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.178.140
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.178.140
  port: 502
  usage: pv
  modbus: tcpip
  name: pv1
- type: template
  template: huawei-dongle-powersensor
  id: 2
  host: 192.168.178.140
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.178.140
  port: 502
  usage: battery
  modbus: tcpip
  name: Luna2000
- name: bkw
  type: template
  template: solarman
  usage: pv
  host: 192.168.178.29
  user: admin
  password: admin

vehicles:
- type: template
  template: jaguar-landrover
  title: Jaguar IPace
  user: XXXXXXX@XXXXXXX.de
  password: XXXXXX
  capacity: 90
  name: ipace

site:
  title: EVCC - Seeblick
  meters:
    grid: grid1
    pv:
    - pv1
    - pv2
    - bkw
    battery: Luna2000
  residualPower: 100

chargers:
   - name: juiceme
     type: template
     template: bender
     host: 192.168.178.35

loadpoints:
  - title: Parkplatz
    charger: juiceme
    mode: pv
    enable:
      threshold: 0
      delay: 2m
    disable:
      threshold: 200
      delay: 5m
    guardduration: 15m
    priority: 10

sponsortoken: <censored>
telemetry: true

tariffs:
  grid:
    type: fixed
    price: 0.4
  feedin:
    type: fixed
    price: 0.0786
```

### Log details

```text
[main  ] INFO 2024/03/02 20:28:54 evcc 0.124.6
[main  ] INFO 2024/03/02 20:28:54 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/02 20:28:54 starting ui and api at :7070
[db    ] INFO 2024/03/02 20:28:55 using sqlite database: /home/encc/.evcc/evcc.db
[jlr   ] WARN 2024/03/02 20:28:55 new device id generated, add `deviceid` to config: d34ce911-1b6c-461d-b430-58725bf2bc16
[lp-1  ] DEBUG 2024/03/02 20:28:57 charge total import: 123.100kWh
[site  ] INFO 2024/03/02 20:28:57 site config:
[site  ] INFO 2024/03/02 20:28:57   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/02 20:28:57     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/02 20:28:57     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/02 20:28:57     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/02 20:28:57     pv 3:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/02 20:28:57     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/03/02 20:28:57   vehicles:
[site  ] INFO 2024/03/02 20:28:57     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/03/02 20:28:57 loadpoint 1:
[lp-1  ] INFO 2024/03/02 20:28:57   mode:        pv
[lp-1  ] INFO 2024/03/02 20:28:57   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/03/02 20:28:57   meters:      charge ✓
[lp-1  ] INFO 2024/03/02 20:28:57     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/03/02 20:28:57 phase timer inactive
[lp-1  ] DEBUG 2024/03/02 20:28:57 pv timer inactive
[site  ] DEBUG 2024/03/02 20:28:57 ----
[lp-1  ] DEBUG 2024/03/02 20:28:57 charge power: 0W
[site  ] ERROR 2024/03/02 20:29:06 pv 3 power: Get "http://192.168.178.29/status.html": dial tcp 192.168.178.29:80: connect: no route to host
[site  ] DEBUG 2024/03/02 20:29:27 ----
[lp-1  ] DEBUG 2024/03/02 20:29:27 charge power: 0W
[site  ] ERROR 2024/03/02 20:29:35 pv 3 power: Get "http://192.168.178.29/status.html": dial tcp 192.168.178.29:80: connect: no route to host
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.6

RTTTC, 2024-03-02:
https://github.com/evcc-io/evcc/issues/12557
The fix will be in the next release version, for now you can use a temporrary workaround as described in the link.

andig, 2024-03-02:
Fixed in master/nightly

closed 2024-03-02 by andig