evcc-io/evcc #12617: evcc publishes api.Offline feature on MQTT with retained flag, but does not clear MQTT topic when it is not present anymore

MarkusGH, 2024-03-03:
### Describe the bug

See description.

Topics affected:

evcc/loadpoints/+/vehicleFeatureOffline true
evcc/vehicles/+/features/+ Offline

### Steps to reproduce

1. "online" vehicle goes to offline by misconfiguration or server error
2. "online" vehicle goes online again
3. Affected MQTT topics are still present

### Configuration details

```yaml
Not relevant
```

### Log details

```text
Not available yet
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.6 (b2921078)

andig, 2024-03-03:
That's a general problem- deleting mqtt topics is horribly complicated and we are not doing this properly at the moment.

MarkusGH, 2024-03-03:
OK - understood.
But for vehicleFeatureOffline there is an additional problem: It seems vehicleFeatureOffline = true is posted, while vehicleFeatureOffline = false is never posted.

andig, 2024-03-03:
It's the same underlying problem. We only post features, not non-features. The right thing to do would be cleaning the entire Mqtt tree once on startup.

andig, 2024-03-03:
@MarkusGH what is confusing though is that you've mentioned retained. I don't think we're posting any retained messages?

MarkusGH, 2024-03-03:
Looks a lot like everything is retained:

```
> service evcc stop
> mosquitto_sub -t "evcc/#" -v --retained-only

evcc/status offline
evcc/site/gridPower 82
evcc/site/pvPower 4508
evcc/site/homePower 1379
evcc/site/siteTitle H73
evcc/site/gridConfigured true
evcc/site/pvConfigured true
evcc/site/batteryConfigured true
evcc/site/currency EUR
evcc/site/tariffGrid 0.2842
evcc/site/tariffFeedIn 0.0973
evcc/site/residualPower 1
evcc/site/version 0.124.6 (b2921078)
evcc/site/prioritySoc 10
evcc/site/bufferSoc 90
evcc/site/pv 1
evcc/site/pv/1/power 4508
evcc/site/pv/1/energy 0
evcc/site/bufferStartSoc 95
evcc/site/smartCostType pricestatic
evcc/site/pvEnergy 0
evcc/site/greenShareHome 1
evcc/site/greenShareLoadpoints 0.96908
evcc/site/tariffPriceHome 0.0973
evcc/site/tariffPriceLoadpoints 0.10308
evcc/site/statistics/30d/avgPrice 0.20099
evcc/site/statistics/30d/avgCo2 0
evcc/site/statistics/30d/solarPercentage 44.521
evcc/site/statistics/30d/chargedKWh 653.5
evcc/site/statistics/365d/solarPercentage 40.153
evcc/site/statistics/365d/chargedKWh 4362.7
evcc/site/statistics/365d/avgPrice 0.24013
evcc/site/statistics/365d/avgCo2 0
evcc/site/statistics/total/chargedKWh 4362.7
evcc/site/statistics/total/avgPrice 0.24013
evcc/site/statistics/total/avgCo2 0
evcc/site/statistics/total/solarPercentage 40.153
evcc/site/batteryDischargeControl false
evcc/site/batteryMode unknown
evcc/site/battery 1
evcc/site/battery/1/power -559
evcc/site/battery/1/energy 0
evcc/site/battery/1/soc 17
evcc/site/battery/1/capacity 0
evcc/site/battery/1/controllable true
evcc/site/sponsor markus@woessner.online
evcc/site/sponsorTokenExpires 0
evcc/site/batteryCapacity 0
evcc/site/batterySoc 17
evcc/site/batteryPower -559
evcc/site/batteryEnergy 0
evcc/loadpoints 3
evcc/loadpoints/1/mode pv
evcc/loadpoints/1/vehicleRange 30
evcc/loadpoints/1/vehicleOdometer 35122
evcc/loadpoints/1/chargeRemainingDuration 0
evcc/loadpoints/1/chargeCurrent 6
evcc/loadpoints/1/chargedEnergy 1008
evcc/loadpoints/1/chargeDuration 2610
evcc/loadpoints/1/chargePower 1352
evcc/loadpoints/1/chargeCurrents 9.7
evcc/loadpoints/1/chargeCurrents/l1 9.7
evcc/loadpoints/1/chargeCurrents/l2 0
evcc/loadpoints/1/chargeCurrents/l3 0
evcc/loadpoints/1/connected true
evcc/loadpoints/1/charging false
evcc/loadpoints/1/enabled true
evcc/loadpoints/1/chargeRemainingEnergy 1764.9
evcc/loadpoints/1/pvAction disable
evcc/loadpoints/1/pvRemaining 179
evcc/loadpoints/1/minCurrent 6
evcc/loadpoints/1/connectedDuration 0
evcc/loadpoints/1/title Carport
evcc/loadpoints/1/maxCurrent 32
evcc/loadpoints/1/phaseAction inactive
evcc/loadpoints/1/phaseRemaining 0
evcc/loadpoints/1/phasesConfigured 0
evcc/loadpoints/1/phasesEnabled 0
evcc/loadpoints/1/phasesActive 1
evcc/loadpoints/1/vehicleDetectionActive false
evcc/loadpoints/1/chargeTotalImport 4194.9
evcc/loadpoints/1/vehicleSoc 91.126
evcc/loadpoints/1/vehicleTargetSoc 0
evcc/loadpoints/1/chargerFeatureIntegratedDevice false
evcc/loadpoints/1/sessionEnergy 1008
evcc/loadpoints/1/sessionSolarPercentage 96.716
evcc/loadpoints/1/priority 0
evcc/loadpoints/1/chargerFeatureHeating false
evcc/loadpoints/1/enableThreshold -1380
evcc/loadpoints/1/disableThreshold 0
evcc/loadpoints/1/limitSoc 0
evcc/loadpoints/1/planEnergy 0
evcc/loadpoints/1/limitEnergy 0
...
```

closed 2024-03-03 by andig