evcc-io/evcc #12640: Version 0.124.6 or 0.124.5 is crashing with panicUnaligned() error in github.com/mabunixda/wattpilot@v1.6.5/wattpilot.go

kad49007, 2024-03-03:
### Describe the bug

Version 0.124.6 or 0.124.5 is crashing with panicUnaligned() error in github.com/mabunixda/wattpilot@v1.6.5/wattpilot.go on Raspberry Pi 4 with Kernel Linux 6.1.21-v8+

Most likely because of new version of mabunixda/wattpilot@v1.6.5 is linked into Version 0.124.6 and 0.124.5.
0.124.4 is running without problems with mabunixda/wattpilot@v1.6.3 which. 

### Steps to reproduce

1. Upgrade from 0.124.4 to 0.124.5 or 0.124.6
2. Start evcc
3. After a few second evcc is crashing
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: debug
  mini: debug
  site: trace
  lp-1: trace
  lp-2: trace
  eMini: trace
  e-Tron: trace
  charger: trace
  aux: trace

# unique installation id
plant: c3e77ab967105b1ccae8945ecccde2e3987556b17683360aa09957f8cdd84b27

interval: 30s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.4.67  
  name: grid1
- type: template
  template: fronius-solarapi-v1
  usage: pv
  host: 192.168.4.15
  name: pv
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.4.67  
  name: pv2
- type: template
  template: fronius-solarapi-v1 
  usage: battery  
  host: 192.168.4.67  
  name: battery3
  user: customer
  password: XXX
  capacity: 19.3

- type: custom
  name: Heizstab
  power: 
    source: http
    
[... 187 lines cut ...]
Mar 03 17:48:32 openhabian evcc[9865]: panic: unaligned 64-bit atomic operation
Mar 03 17:48:32 openhabian evcc[9865]: goroutine 1547 [running]:
Mar 03 17:48:32 openhabian evcc[9865]: runtime/internal/atomic.panicUnaligned()
Mar 03 17:48:32 openhabian evcc[9865]: 	runtime/internal/atomic/unaligned.go:8 +0x24
Mar 03 17:48:32 openhabian evcc[9865]: runtime/internal/atomic.Xadd64(0x68224dc, 0x1)
Mar 03 17:48:32 openhabian evcc[9865]: 	runtime/internal/atomic/atomic_arm.s:263 +0x14
Mar 03 17:48:32 openhabian evcc[9865]: github.com/mabunixda/wattpilot.(*Wattpilot).getRequestId(...)
Mar 03 17:48:32 openhabian evcc[9865]: 	github.com/mabunixda/wattpilot@v1.6.5/wattpilot.go:174
Mar 03 17:48:32 openhabian evcc[9865]: github.com/mabunixda/wattpilot.(*Wattpilot).RequestStatusUpdate(0x68224d8)
Mar 03 17:48:32 openhabian evcc[9865]: 	github.com/mabunixda/wattpilot@v1.6.5/wattpilot.go:683 +0x90
Mar 03 17:48:32 openhabian evcc[9865]: github.com/mabunixda/wattpilot.(*Wattpilot).processLoop.func1()
Mar 03 17:48:32 openhabian evcc[9865]: 	github.com/mabunixda/wattpilot@v1.6.5/wattpilot.go:429 +0x3c
Mar 03 17:48:32 openhabian evcc[9865]: created by github.com/mabunixda/wattpilot.(*Wattpilot).processLoop in goroutine 65
Mar 03 17:48:32 openhabian evcc[9865]: 	github.com/mabunixda/wattpilot@v1.6.5/wattpilot.go:427 +0x4a0
Mar 03 17:48:32 openhabian systemd[1]: evcc.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Mar 03 17:48:32 openhabian systemd[1]: evcc.service: Failed with result 'exit-code'.
```

### What type of operating system are you running?

Linux

### Version

0.124.6

andig, 2024-03-03:
@mabunixda

mabunixda, 2024-03-04:
Nein `go vet` liefert nichts - das lauft bei mir automatisch im [pre-commit hook](https://github.com/mabunixda/wattpilot/blob/main/.pre-commit-config.yaml#L12) 

PR ist erstellt #12648 basieren auf [wattpilot#35](https://github.com/mabunixda/wattpilot/pull/35)

closed 2024-03-04 by andig

kad49007, 2024-03-07:
@mabunixda 
Der beschriebene Fehler ist gefixed. Mit der Zeit wird aber der Ladestroms /Status des Wattpilots nicht mehr ausgelesen. Bin daher erstmal zurück auf 0.124.4, um später in Ruhe diesen neuen möglichen Fehler zu analysieren.