evcc-io/evcc #12646: Mennekes Wallbox springt immer auf 3P

TobiasNeuschulz, 2024-03-03:
### Describe the bug

Nun - der 2. Versuch heute den Fehler zu beschreiben.
Mennekes Wallbox ist fest auf 1 Phase begrenzt. (Sicherung für Phase 2 & 3 aus; Mennekes Konfiguration auf "Einphasen System" eingestellt. Auf der Mennekes Konfigurationsseite wird unter Ladestrom [A] nur ein Wert angezeigt; sonst logischerweise 3 Werte bei 3 Phasen.
Wenn ich in EVCC den Ladestrom auf 1-phasig einstelle springt er immer wieder auf 

Max. Ladestrom
~ 11,0 kW
Min. Ladestrom
~ 4,1 kW

Vermutlich mit jeder automatischen Phasenerkennung innerhalb von EVCC siehe: 
[lp-1  ] DEBUG 2024/03/03 21:28:05 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/03 21:28:05 charge currents: [0.004 0 0]A

### Steps to reproduce

1. Mennekes Wallbox auf "Einphasen System eingestellen"; Sicherung für Phase 2&3 aus
2. Ladepunkt auf 1-phasig stellen
3. EVCC erkennt Wallbox als 3P
4. Überschuss laden beginnt erst ab 4,1kw

### Configuration details

```yaml
[...]
chargers:
- type: template
  template: bender
  host: <ip>
  port: 502
  name: wallbox4

- name: trocknungsgeraet
  type: template
  template: shelly
  host: <ip> # IP-Adresse oder Hostname
  channel: 0
  integrateddevice: true
  icon: cooler

loadpoints:
- title: Garage
  charger: wallbox4
  priority: 0
  mode: pv
  enable:
   delay: 1m
  disable:
   delay: 10m

- title: Keller Trocknungsgerät
  charger: trocknungsgeraet
  priority: 1
  mode: off
  guardduration: 15m
  enable:
   delay: 1m
  disable:
   delay: 10m

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.4711 # EUR/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0815 # EUR/kWh
site:
  title: SiteName
  meters:
    grid: grid1
    pv:
    - pv1
    - pv2
    battery:
    - battery3
  maxGridSupplyWhileBatteryCharging: 50
  residualPower: 100
```

### Log details

```text
[main  ] INFO 2024/03/03 21:28:00 evcc 0.124.6
[main  ] INFO 2024/03/03 21:28:00 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/03 21:28:00 starting ui and api at :7070
[db    ] INFO 2024/03/03 21:28:00 using sqlite database: /home/admin/.evcc/evcc.db
[lp-1  ] DEBUG 2024/03/03 21:28:00 charge total import: 985.522kWh
[lp-2  ] DEBUG 2024/03/03 21:28:00 charge total import: 3.815kWh
[site  ] INFO 2024/03/03 21:28:00 site config:
[site  ] INFO 2024/03/03 21:28:00   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/03 21:28:00     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/03 21:28:00     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/03 21:28:00     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/03 21:28:00     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[lp-1  ] INFO 2024/03/03 21:28:00 loadpoint 1:
[lp-1  ] INFO 2024/03/03 21:28:00   mode:        pv
[lp-1  ] INFO 2024/03/03 21:28:00   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/03/03 21:28:00   meters:      charge ✓
[lp-1  ] INFO 2024/03/03 21:28:00     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2024/03/03 21:28:00 loadpoint 2:
[lp-2  ] INFO 2024/03/03 21:28:00   mode:        off
[lp-2  ] INFO 2024/03/03 21:28:00   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2024/03/03 21:28:00   meters:      charge ✓
[lp-2  ] INFO 2024/03/03 21:28:00     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] DEBUG 2024/03/03 21:28:00 phase timer inactive
[lp-1  ] DEBUG 2024/03/03 21:28:00 pv timer inactive
[lp-2  ] DEBUG 2024/03/03 21:28:00 phase timer inactive
[lp-2  ] DEBUG 2024/03/03 21:28:00 pv timer inactive
[site  ] WARN 2024/03/03 21:28:00 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/03/03 21:28:00 ----
[lp-1  ] DEBUG 2024/03/03 21:28:00 charge power: 0W
[lp-2  ] DEBUG 2024/03/03 21:28:01 charge power: 0W
[site  ] DEBUG 2024/03/03 21:28:03 pv power: 0W
[site  ] DEBUG 2024/03/03 21:28:04 battery soc: 55%
[site  ] DEBUG 2024/03/03 21:28:04 battery power: 262W
[site  ] DEBUG 2024/03/03 21:28:05 grid meter: 13W
[site  ] DEBUG 2024/03/03 21:28:05 site power: 375W
[lp-1  ] DEBUG 2024/03/03 21:28:05 charge voltages: [236 4.29e+09 4.29e+09]V
[lp-1  ] DEBUG 2024/03/03 21:28:05 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/03 21:28:05 charge currents: [0.004 0 0]A
[lp-1  ] DEBUG 2024/03/03 21:28:05 charge total import: 985.522kWh
[lp-1  ] DEBUG 2024/03/03 21:28:05 charger status: B
[lp-1  ] INFO 2024/03/03 21:28:05 car connected
[lp-1  ] DEBUG 2024/03/03 21:28:05 pv timer elapse
[lp-1  ] DEBUG 2024/03/03 21:28:05 pv timer inactive
[lp-1  ] DEBUG 2024/03/03 21:28:05 charge total import: 985.522kWh
[lp-1  ] DEBUG 2024/03/03 21:28:05 pv charge current: 0A = 0A + -0.544A (375W @ 3p)
[lp-1  ] DEBUG 2024/03/03 21:28:05 pv enable timer reset
[lp-1  ] DEBUG 2024/03/03 21:28:05 pv timer inactive
```

### What type of operating system are you running?

Linux

### Version

 0.124.6

VolkerK62, 2024-03-04:
Deine Wallbox meldet auf den Phasen 2 und 3 utopisch hohe Werte.
`[lp-1  ] DEBUG 2024/03/03 21:28:05 charge voltages: [236 4.29e+09 4.29e+09]V`
Weil die > 207V sind, wird 3p erkannt.

premultiply, 2024-03-04:
Davon mal bitte `evcc charger --log trace`.

TobiasNeuschulz, 2024-03-04:
@evcc:~ $ evcc charger --log trace

[main  ] INFO 2024/03/04 09:49:19 evcc 0.124.6
[main  ] INFO 2024/03/04 09:49:19 using config file: /etc/evcc.yaml
[db    ] INFO 2024/03/04 09:49:19 using sqlite database: /home/admin/.evcc/evcc.                                                                                                                                                 db
[db    ] TRACE 2024/03/04 09:49:19 SELECT count(*) FROM sqlite_master WHERE type                                                                                                                                                 ='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT sql FROM sqlite_master WHERE type IN (                                                                                                                                                 "table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "                                                                                                                                                 table" desc 1 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT * FROM `settings` 8 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT count(*) FROM sqlite_master WHERE type                                                                                                                                                 ='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT count(*) FROM sqlite_master WHERE type                                                                                                                                                 ='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT count(*) FROM sqlite_master WHERE type                                                                                                                                                 ='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT sql FROM sqlite_master WHERE type IN (                                                                                                                                                 "table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "t                                                                                                                                                 able" desc 1 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT count(*) FROM sqlite_master WHERE type                                                                                                                                                 ='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT sql FROM sqlite_master WHERE type IN (                                                                                                                                                 "table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by ty                                                                                                                                                 pe = "table" desc 2 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT * FROM `config_details` LIMIT 1 -1 <ni                                                                                                                                                 l>
[db    ] TRACE 2024/03/04 09:49:19 SELECT count(*) FROM sqlite_master WHERE type                                                                                                                                                  = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_confi                                                                                                                                                 gs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CON                                                                                                                                                 STRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" O                                                                                                                                                 R sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT count(*) FROM sqlite_master WHERE type                                                                                                                                                  = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/03/04 09:49:19 SELECT count(*) FROM sqlite_master WHERE type                                                                                                                                                  = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devic                                                                                                                                                 es_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CON                                                                                                                                     [... cut ...]
[bender] TRACE 2024/03/04 09:49:19 modbus: recv 00 0f 00 00 00 17 ff 03 14 20 20                                                                                                                                                  20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20
Power:          0W
Energy:         985.5kWh
Current L1..L3: 0.003A 0A 0A
Voltage L1..L3: 234V 4.29e+09V 4.29e+09V
Charge status:  B
Enabled:        false
Charged:        0.0kWh
Duration:       1h42m42s
Identifier:     <none>

TobiasNeuschulz, 2024-03-04:
Die rest API der Wallbox liefert über <IP>/rest/full_state
```
conn_state:vehicle_connected_type2
auth_state:authorized_for_charging
auth_uid:
time_since_charging_start:465
meter_wh:985522
power_w:0
transaction_wh:0
cp_id:+49*839*00000000001
ocpp_state:occupied
type2_state:b
type2_proximity:32
sig_current:0
schuko_state:idle
backend_conn_state:pending
free_charging:on
slave_state:
ocpp_meter_cfg:modbus_meter_nzr
ocpp_meter_serial:00150976
current_a:0.00
energy_man_current:0
ambient_temp:+4.00
firmware_ver:5.22.5-13500
cc_serial_n:2101537951/b94060010me2
con_cycles_schuko:0
con_cycles_type2:293
max_current:16
rcmb_state:okay
rcmb_max_values: 0.0, 0.1
rcmb_current_values: 0.0, 0.1
cable_attached:on
schuko_cfg:disable
rcd_state:disable
mcb_type2_state:disable
mcb_schuko_state:disable
cp_vendor:MENNEKES
errors:no_errors
cp_model:CC612_2S0R_CC
display_text:
```

Schon mal herzlichen Dank, dass ihr euch dies anschaut!

TobiasNeuschulz, 2024-03-04:
https://github.com/evcc-io/evcc/discussions/7704
Hier hat ein anderer User vermutlich die Mennekes Wallbox auf 1P konfiguriert. Dort taucht auch: "4.29e+09V" auf...

andig, 2024-03-05:
Anscheinend kommen da falsche Werte. Gibts ein Ticket bei Menneckes dafür?

premultiply, 2024-03-05:
Ja, vermutlich.
Ich wollt es einfach per Condition rausfiltern und auf 0 setzen.

andig, 2024-03-05:
Das ist doch der hier?

> ff ff ff ff ff ff ff ff

Wäre da nicht einer den `nan` Werte schon hilfreich?

premultiply, 2024-03-05:
Wie könnte ich die hier verwenden?

closed 2024-03-08 by andig

TobiasNeuschulz, 2024-03-09:
Guten Morgen, vielen lieben Dank für den Fix - ihr seid so spitze. Ich habe das Upgrade schon durchgeführt; kann es erst später die Tage testen; das Auto ist voll.

Ich könnte versuchen über einen indirekten Draht ein Ticket eröffnen zu lassen und das Hersteller-seitig durch Mennekes fixen zu lassen. Ich kann aber nicht abschätzen, ob ich Erfolg damit habe. Aber der Hardcoding-Ansatz innerhalb von evcc innerhalb von 5 Tagen war definitiv schneller.

premultiply, 2024-03-09:
Mennekes kann da ziemlich wahrscheinlich selbst nichts tun.
Ist ein zugekaufter Bender-Controller.