evcc-io/evcc #12653: tesla mit gen3 wallconnector

Nadel234, 2024-03-04:
### Describe the bug

Hallo! 
Ich habe EVCC komplett neu installiert. Im PV Modus, nach Ablauf von drei Minuten, wird die Ladung nicht abgebrochen und lädt weiter und zeigt dazu noch Fehlermeldungen.  

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 490c093885a6d8eea6344256db5e315eff961011d97fe913cb55a09

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoiTmFkZWwyMzQiLCJleHAiOjE4MDQwODI0MDAsI

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.59  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.59  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery2

chargers:
- type: template
  template: twc3 
  host: 192.168.178.71  
  name: wallbox4

vehicles:
- type: template
  template: tesla 
  title: Tesla Model Y  
  accessToken: eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCIsImtpZCI6InE0dHg3Q1UyYzI2V1BiemwxZjZjanM3QnhzayJ9.eyJpc3MiOiJodHRwczovL2F1dGgudGVzbGEuY29tL29hdXRoMi92My9udHMiLCJhenAiOiIwOGZmODI3ZmEzMWItNGVhNi1iZWQxLTgxN2JlMGFiNDBjOCIsInN1YiI6IjhjMDNmNzZlLThhMjAtNDYzYi05ZTE3LTQ2MGZiMTcwY2M4YiIsImF1ZCI6WyJodHRwczovL2ZsZWV0LWFwaS5wcmQubmEudm4uY2xvdWQudGVzbGEuY29tIiwiaHR0cHM6Ly9mbGVldC1hcGkucHJkLmV1LnZuLmNsb3VkLnRlc2xhLmNvbSIsImh0dHBzOi8vYXV0aC50ZXNsYS5jb20vb2F1dGgyL3YzL3VzZXJpbmZvIl0sInNjcCI6WyJvcGVuaWQiLCJvZmZsaW5lX2FjY2VzcyIsInVzZXJfZGF0YSIsInZlaGljbGVfZGV2aWNlX2RhdGEiLCJ2ZWhpY2xlX2NtZHMiLCJ2ZWhpY2xlX2NoYXJnaW5nX2NtZHMiXSwiYW1yIjpbInB3ZCIsIm1mYSIsIm90cCJdLCJ  
  refreshToken: EU_d9cc823d84989a5f3b7a3d0fecd3cc31111111111111111111 
  vin: LRWYGC111111111  
  capacity: 77  
  name: ev3

[... 537 lines cut ...]
Mär 04 11:28:130 evcc[5254]: [lp-1  ] DEBUG 2024/03/04 11:28:13 charge voltages: [227 228 227]V
Mär 04 11:28:130 evcc[5254]: [lp-1  ] DEBUG 2024/03/04 11:28:13 detected connected phases: 3p
Mär 04 11:28:130 evcc[5254]: [lp-1  ] DEBUG 2024/03/04 11:28:13 charge currents: [5.1 5.1 5.1]A
Mär 04 11:28:130 evcc[5254]: [lp-1  ] DEBUG 2024/03/04 11:28:13 detected active phases: 3p
Mär 04 11:28:130 evcc[5254]: [lp-1  ] DEBUG 2024/03/04 11:28:13 charger status: C
Mär 04 11:28:130 evcc[5254]: [lp-1  ] WARN 2024/03/04 11:28:13 vehicle soc: asleep (ignored by estimator)
Mär 04 11:28:130 evcc[5254]: [lp-1  ] DEBUG 2024/03/04 11:28:13 soc estimated: 81.24% (vehicle: 81.00%)
Mär 04 11:28:130 evcc[5254]: [lp-1  ] DEBUG 2024/03/04 11:28:13 vehicle soc: 81%
Mär 04 11:28:130 evcc[5254]: [lp-1  ] ERROR 2024/03/04 11:28:13 vehicle soc limit: asleep
Mär 04 11:28:130 evcc[5254]: [lp-1  ] ERROR 2024/03/04 11:28:13 vehicle range: asleep
Mär 04 11:28:130 evcc[5254]: [lp-1  ] DEBUG 2024/03/04 11:28:13 pv charge current: 0A = 0A + -3.16A (2179W @ 3p)
Mär 04 11:28:130 evcc[5254]: [lp-1  ] DEBUG 2024/03/04 11:28:13 projected site power 2179W &gt;= 0W disable threshold
Mär 04 11:28:130 evcc[5254]: [lp-1  ] DEBUG 2024/03/04 11:28:13 pv disable in 0s
Mär 04 11:28:130 evcc[5254]: [lp-1  ] DEBUG 2024/03/04 11:28:13 pv disable timer elapsed
Mär 04 11:28:13 evcc[5254]: [lp-1  ] ERROR 2024/03/04 11:28:13 charger disable: 500 Internal Server Error: vehicle rejected request: your public key has not been paired with the vehicle
^C
```

### What type of operating system are you running?

Linux

### Version

0.124.5

simonhard, 2024-03-04:
Du solltest in jedem Fall mal deine Secrets aus der Config nehmen, wenn du sie hier pastest

mucki12, 2024-03-04:
>ERROR 2024/03/04 11:28:13 charger disable: 500 Internal Server Error: vehicle rejected request: your public key has not been paired with the vehicle

Hört sich für mich so an, als hättest du den Fahrzeugschlüssel nicht im Tesla hinterlegt (letzter Schritt bei tesla.evcc.io mit QR Code und Tesla App).

andig, 2024-03-04:
> vehicle rejected request: your public key has not been paired with the vehicle

Virtual key ist nicht auf dem Fahrzeug installiert.

closed 2024-03-04 by andig

Nadel234, 2024-03-08:
> > > Fahrzeug abgelehnt Antrag: Ihr öffentlicher Schlüssel wurde nicht mit dem Fahrzeug gepaart
> > 
> > 
> > 
> > Virtueller Schlüssel ist auf dem Fahrzeug.
> 
> Ich habe das Problem. Und bekomme ich die Virtual Key ins Fahrzeug ?

Bei mir war Multi-Faktor-Authentifizierung (MFA) tesla konto schuldig, habe ausgeschaltet und  letzter Schritt bei tesla.evcc.io mit QR Code und Tesla App wiederholt. Nach paar Stunden hat es sich verbunden und zeigt keinen Fehler mehr.

Richy0007, 2024-03-08:
Nun hat es auch bei mir mit den Virtuellen Schlüssel geklappt.
Vielen Dank...

ich hatte nicht gesehen dass ich noch nach ganz unten scrollen muss.

phoet, 2024-04-17:
Es hat bei mir etwas gedauert, weil ich nicht verstanden habe, dass man diese Seite mit dem Smartphone aufrufen muss, auf dem die Tesla App läuft. Steht auch nirgendwo ein hinweis.

naltatis, 2024-04-17:
@phoet Hast du nen Vorschlag für bessere Formulierung bzw. nen Hinweis?

phoet, 2024-04-17:
> Für die Steuerung des Tesla Wall Connectors oder der Powerwall ist eine signierte Kommunikation mit Tesla notwendig. Steuerbefehle deiner lokalen evcc Instanz werden dabei über tesla.evcc.io geleitet. Dafür ist eine zusätzliche Berechtigung erforderlich, die du über deine Tesla-App erteilen musst: 

> Für die Steuerung des Tesla Wall Connectors oder der Powerwall ist eine signierte Kommunikation mit Tesla notwendig. Steuerbefehle deiner lokalen evcc Instanz werden dabei über tesla.evcc.io geleitet. Dafür ist eine zusätzliche Berechtigung erforderlich, die du über deine Tesla-App erteilen musst. Öffne dazu den Link mit dem Smartphone, auf dem die Tesla-App installiert ist:

und dann noch einen hinweis

> Wenn es zu Problemen beim Öffnen des Links kommt, dann beende zunächst die Tesla-App.

mucki12, 2024-04-17:
> Für die Steuerung des Tesla Wall Connectors

Das müsste noch einmal geändert werden, da evcc den Tesla Wall Connector nicht steuert.

phoet, 2024-04-17:
Jain, das ist quasi äquivalent, weil der QR-Code auch nur ein Link auf die gleiche Seite ist. Das ist da von Testla (wie immer) suboptimal gelöst...