evcc-io/evcc #12661: openWB - Änderung der Lademodi funktioniert nicht

gibber666, 2024-03-04:
### Describe the bug

Meine Wallbox "openWB series 2" reagiert nicht auf die Änderungen (Stop/PV/Min+PV/Schnelladen).
Die Kommunikation mit der Wallbox scheint grundsätzlich zu funktionieren, da ich die aktuelle Ladeleistung im ivcc Interface sehen kann, wenn ein Auto lädt.
Wenn ich in ivcc auf "Stop" stelle, wie das Auto als verbunden angezeigt.
Sobald ich auf einen anderen Lademodus wechsle, geht die Anzeige auf "Ladebereit. Warte auf Fahrzeug … " und nichts passiert, weil die Wallbox anscheinend nicht angesteuert wird.

Ich hoffe jemand kann helfen.

### Steps to reproduce

1. In evcc umschalten von z.B. Stop auf Schnelladen
2. Wallbox bzw. Ladevorgang startet nicht.
3. Verbundendes Fahrzeug wechselt auf "Ladebereit. Warte auf Fahrzeug.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.xxx.xxx.xxx
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: 

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors
[... 282 lines cut ...]
[lp-1  ] DEBUG 2024/03/04 16:13:01 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/04 16:13:01 charge total import: 12696.977kWh
[lp-1  ] DEBUG 2024/03/04 16:13:01 charger status: B
[site  ] DEBUG 2024/03/04 16:13:11 ----
[lp-1  ] DEBUG 2024/03/04 16:13:11 charge power: 0W
[lp-2  ] DEBUG 2024/03/04 16:13:11 charge power: 0W
[site  ] DEBUG 2024/03/04 16:13:11 pv power: 872W
[site  ] DEBUG 2024/03/04 16:13:11 battery soc: 66%
[site  ] DEBUG 2024/03/04 16:13:11 battery power: -99W
[site  ] DEBUG 2024/03/04 16:13:11 grid meter: 20W
[site  ] DEBUG 2024/03/04 16:13:11 grid powers: [42 9 -32]W
[site  ] DEBUG 2024/03/04 16:13:11 grid currents: [3.09 1.97 -2.16]A
[site  ] DEBUG 2024/03/04 16:13:11 site power: -79W
[lp-2  ] DEBUG 2024/03/04 16:13:11 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/03/04 16:13:11 charge total import: 5488.300kWh
[lp-2  ] DEBUG 2024/03/04 16:13:11 charger status: B
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.6

andig, 2024-03-04:
> In evcc umschalten von z.B. Stop auf Schnelladen

Sehe ich das irgendwo? Wie passt das Logfile zu Deiner Fehlerbeschreibung?

andig, 2024-03-04:
Und Du bist sicher, dass Deine OpenWB auf "nur Ladepunkt" steht?

gibber666, 2024-03-05:
Sehr peinlich, aber das war tatsächlich der Grund.
Jetzt funktioniert alles.

Danke dir.

andig, 2024-03-05:
Immer erstmal nach den offensichtlichen Fehlern schauen…

closed 2024-03-05 by andig