evcc-io/evcc #12667: meters acthor -  error in logfile strconv.ParseFloat

deejott23, 2024-03-04:
### Describe the bug

Hello,

I have installed the my-pv Acthor integration and would like to take it into account for the surplus calculation.
https://docs.evcc.io/docs/devices/meters#acthor

Unfortunately, I always get the following error message in the log:

[site  ] ERROR 2024/03/04 17:03:15 aux meter 1: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] DEBUG 2024/03/04 17:03:15 aux power: 0W

this is a snap of the evcc.yaml
  - name: my_aux
    type: template
    template: thor
    usage: aux
    host: 123.123.123.123 # < IP-Adresse ist hier nat. nur Dummy

If I comment out the code, I no longer get the error message. 

Does anyone have any idea what could be causing this, or what I could test?

Thanks
Simon

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.178.75
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: 123abc

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
[... 227 lines cut ...]
[lp-1  ] DEBUG 2024/03/04 17:26:39 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/04 17:26:39 charge total import: 1738.472kWh
[lp-1  ] DEBUG 2024/03/04 17:26:39 charger status: A
[site  ] DEBUG 2024/03/04 17:27:09 ----
[lp-1  ] DEBUG 2024/03/04 17:27:09 charge power: 0W
[site  ] WARN 2024/03/04 17:27:12 pv 1 power: -2061W is negative - check configuration if sign is correct
[site  ] DEBUG 2024/03/04 17:27:12 pv power: 0W
[site  ] DEBUG 2024/03/04 17:27:12 grid meter: 2073W
[site  ] DEBUG 2024/03/04 17:27:12 grid powers: [677 741 655]W
[site  ] DEBUG 2024/03/04 17:27:12 grid currents: [3.1 3.45 3]A
[site  ] ERROR 2024/03/04 17:27:12 aux meter 1: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] DEBUG 2024/03/04 17:27:12 aux power: 0W
[site  ] DEBUG 2024/03/04 17:27:12 site power: 2073W
[lp-1  ] DEBUG 2024/03/04 17:27:12 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/04 17:27:12 charge total import: 1738.472kWh
[lp-1  ] DEBUG 2024/03/04 17:27:12 charger status: A
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.6

andig, 2024-03-04:
Mach mal bitte ein trace log damit wir sehen was da vom API kommt: `http: trace`

deejott23, 2024-03-04:
Hier voilà... die IPs sind nat. Dummy Werte:

***

"boostactive":0,
"legboostnext":"null",
"date":"04.03.24",
"loctime":"17:54:15",
"unixtime":1709571255,
"wp_flag":0,
"wp_time1_ctr":0,
"wp_time2_ctr":0,
"wp_time3_ctr":0,
"schicht_flag":0,
"act_night_flag":0,
"ctrlstate":"Conn. to my-PV Meter. P=2201",
"blockactive":0,
"error_state":0,
"meter1_id":2312240060,
"meter1_ip":"192.168.111.111",
"meter2_id":null,
"meter2_ip":"null",
"meter3_id":null,
"meter3_ip":"null",
"meter4_id":null,
"meter4_ip":"null",
"meter5_id":null,
"meter5_ip":"null",
"meter6_id":null,
"meter6_ip":"null",
"meter_ss":35,
"meter_ssid":"Polizei",
"surplus":-2201,
"m0sum":-2201,
"m0l1":null,
"m0l2":null,
"m0l3":null,
"m0bat":null,
"m1sum":null,
"m1l1":null,
"m1l2":null,
"m1l3":null,
"m1devstate":null,
"m2sum":null,
"m2l1":null,
"m2l2":null,
"m2l3":null,
"m2soc":null,
"m2state":null,
"m2devstate":null,
"m3sum":null,
"m3l1":null,
"m3l2":null,
"m3l3":null,
"m3soc":null,
"m3devstate":null,
"m4sum":null,
"m4l1":null,
"m4l2":null,
"m4l3":null,
"m4devstate":null,
"ecarstate":"null",
"ecarboostctr":null,
"mss2":"null",
"mss3":"null",
"mss4":"null",
"mss5":"null",
"mss6":"null",
"mss7":"null",
"mss8":"null",
"mss9":"null",
"mss10":"null",
"mss11":"null",
"volt_mains":232,
"curr_mains":0,
"volt_L2":1,
"curr_L2":0,
"volt_L3":1,
"curr_L3":0,
"volt_out":0,
"freq":49970,
"temp_ps":245,
"fan_speed":0,
"ps_state":7,
"cur_ip":"192.111.99.1",
"cur_sn":"255.255.255.0",
"cur_gw":"192.111.99.1",
"cur_dns":"192.111.99.59",
"fwversionlatest":"a0021400",
"psversionlatest":110,
"p9sversionlatest":102,
"upd_state":1,
"upd_files_left":0,
"ps_upd_state":0,
"p9s_upd_state":0,
"cloudstate":4,
"debug_ip":"0.0.0.0"
}
[site  ] ERROR 2024/03/04 17:54:15 aux meter 1: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] DEBUG 2024/03/04 17:54:15 aux power: 0W
[site  ] DEBUG 2024/03/04 17:54:15 site power: 2202W
[lp-1  ] DEBUG 2024/03/04 17:54:15 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/04 17:54:15 charge total import: 1738.472kWh
[lp-1  ] DEBUG 2024/03/04 17:54:15 charger status: A

andig, 2024-03-04:
Bitte vollständiges JSON

deejott23, 2024-03-04:
Gute Frage - wir komme ich da dran? Läuft als Addon im HomeAssistent... ich google mal was ich dazu finde...

deejott23, 2024-03-04:
Also, nach meinen Recherchen gibt es bei der Nutzung als HA Addon keine Möglichkeit eines ausführlichen Traces... falls hier jemand einen Tipp freu ich mich.

Ansonsten muss das evcc mal noch auf einem anderen Weg installieren und schauen ob ich den Fehler reproduzieren kann :/

andig, 2024-03-05:
Gerne wieder melden 👍🏻

pvssch, 2024-03-05:
Hallo,

selbes (ähnliches) Problem: aux meter 1: strconv.ParseFloat: parsing \"<nil>\": invalid syntax

- name: my_aux
  type: template
  template: thor
  usage: aux
  host: 192.168.111.30

site:
  title: xxxx
  
  meters:
    grid: grid1
    pv:
      - pv_10ktl
      - pv_30ktl
      - pv_Fronius
      
    battery:
      - battery_2
      - battery_1
      
    aux: 
      - my_aux

{
    "device": "ACTHOR",
    "acthor9s": 2,
    "fwversion": "a0021400",
    "psversion": 110,
    "p9sversion": 102,
    "fsetup": 0,
    "screen_mode_flag": 0,
    "power_system": null,
    "power_act": null,
    "power_solar_act": null,
    "power_grid_act": null,
    "power_ac9": 0,
    "power_solar_ac9": 0,
    "power_grid_ac9": 0,
    "power1_solar": 0,
    "power1_grid": 0,
    "power2_solar": 0,
    "power2_grid": 0,
    "power3_solar": 0,
    "power3_grid": 0,
    "load_state": " 1:1 2:1 3:1",
    "load_nom": 0,
    "rel1_out": "0001",
    "pump_pwm": 0,
    "temp1": 351,
    "temp2": null,
    "temp3": null,
    "temp4": null,
    "boostactive": 0,
    "legboostnext": "null",
    "date": "05.03.24",
    "loctime": "23:22:52",
    "unixtime": 1709677372,
    "uptime": 24,
    "wp_flag": 0,
    "wp_time1_ctr": 0,
    "wp_time2_ctr": 0,
    "wp_time3_ctr": 0,
    "schicht_flag": 0,
    "act_night_flag": 0,
    "ctrlstate": "Conn. to Adj.Modbus P=7",
    "blockactive": 0,
    "error_state": 0,
    "meter1_id": null,
    "meter1_ip": "null",
    "meter2_id": null,
    "meter2_ip": "null",
    "meter3_id": null,
    "meter3_ip": "null",
    "meter4_id": null,
    "meter4_ip": "null",
    "meter5_id": null,
    "meter5_ip": "null",
    "meter6_id": null,
    "meter6_ip": "null",
    "meter_ss": null,
    "meter_ssid": "null",
    "surplus": -7,
    "m0sum": -7,
    "m0l1": null,
    "m0l2": null,
    "m0l3": null,
    "m0bat": null,
    "m1sum": null,
    "m1l1": null,
    "m1l2": null,
    "m1l3": null,
    "m1devstate": null,
    "m2sum": null,
    "m2l1": null,
    "m2l2": null,
    "m2l3": null,
    "m2soc": null,
    "m2state": null,
    "m2devstate": null,
    "m3sum": null,
    "m3l1": null,
    "m3l2": null,
    "m3l3": null,
    "m3soc": null,
    "m3devstate": null,
    "m4sum": null,
    "m4l1": null,
    "m4l2": null,
    "m4l3": null,
    "m4devstate": null,
    "ecarstate": "null",
    "ecarboostctr": null,
    "mss2": "null",
    "mss3": "null",
    "mss4": "null",
    "mss5": "null",
    "mss6": "null",
    "mss7": "null",
    "mss8": "null",
    "mss9": "null",
    "mss10": "null",
    "mss11": "null",
    "volt_mains": 229,
    "curr_mains": 0,
    "volt_L2": 1,
    "curr_L2": 0,
    "volt_L3": 1,
    "curr_L3": 0,
    "volt_out": 0,
    "freq": 50020,
    "temp_ps": 227,
    "fan_speed": 0,
    "ps_state": 7,
    "9s_state": 1,
    "cur_ip": "192.168.111.30",
    "cur_sn": "255.255.255.0",
    "cur_gw": "192.168.111.50",
    "cur_dns": "192.168.111.50",
    "fwversionlatest": "a0021400",
    "psversionlatest": 110,
    "p9sversionlatest": 102,
    "upd_state": 0,
    "upd_files_left": 0,
    "ps_upd_state": 0,
    "p9s_upd_state": 0,
    "cloudstate": 4,
    "debug_ip": "0.0.0.0"
}

andig, 2024-03-05:
Da steht augenscheinlich `null` drin. Ergibt das Sinn? Nachts?

pvssch, 2024-03-05:
Ja, es ist Nacht

kalyke79, 2024-03-06:
Hallo,

pvssch (sowie ich auch) hat vermutlich einen acthor_9s. Also 3 Phasen 9kW. Da steht die Leistung dann im Feld: power_ac9.
Ich wollte gerade darum bitten den acthor9s zu ergänzen und bin auf das Ticket gestoßen.
Im moment nutze ich folgende custom Einbindung:
`- type: custom
  power:
    source: http
    uri: http://10.1.1.24/data.jsn
    jq: .power_ac9
 name: acthor9s`

Vielen Dank und Viele Grüße

deejott23, 2024-03-06:
Hallo,
ah das ist ein guter Hinweis, ich nutze auch den AC•THOR 9s. Ich dachte die IMplementierung https://docs.evcc.io/docs/devices/meters#acthor deckt diesen auch ab...

pvssch, 2024-03-06:
Hallo, 
Super danke. Funktioniert.

Eine Anmerkung als Anfänger hätte ich noch. Bitte die Formatierung beibehalten:

- name: my_aux
  type: custom 
  power: 
    source: http 
    uri: http://192.168.111.30/data.jsn 
    jq: .power_ac9

deejott23, 2024-03-06:
Danke mit der custom integration scheint es beim mir auch zu funktionieren...

closed 2024-03-06 by andig