evcc-io/evcc #12672: Odometer abfrage

Geri-G93, 2024-03-04:
### Describe the bug

schönen guten Abend
folgende Bugs sind mir aufgefallen
zum einen FlobZ SchnittStelle:
es wird immer die Cache angefragt, das sind alte Werte zeitweise, es müsste zumindest beim anstöpseln mal ohne Cache abgefragt werden, dauert aber paar sekunden
deswegen mach ich das anders jetzt
ich frag im iobroker bei Evcc-Connect bei flobZ frische Werte ab und geb die für evcc per mqtt preis
funktioniert soweit gut
bis auf die Tatsache das der Soc erst beim nächsten Poll da ist, allerdings mein ärgerniss: Odometer wird dann nicht mehr aktualisiert und steht auf dem vom Vortag, auf climater springt evcc sofort los :) und die Standheizung steuere ich ebenfalls mit ioBroker 

wäre evt gut wenn evcc bei vehicle config per mqtt die Daten nimmt wenns neue gibt, den Soc aber weiterhin berechnet solang es keinen neueren gibt :) 

### Steps to reproduce

--

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 80

log: debug
levels:
  cache: error
  #mqtt: trace
interval: 30s # control cycle interval
# unique installation id
plant: **
sponsortoken: **

# mqtt message broker
mqtt:
  broker: 127.0.0.1:1889
  topic: evcc # root topic for publishing, set empty to disable publishing
  clientid: EVCC
  user: EVCC
  password: evcc2023
modbusproxy:
  - port: 503
    uri: 192.168.155.9:502    #Huawei sun2000_1
  #- port: 504
   # uri: 192.168.154.11:502   #Orno Zähler Wallbox
  
meters:
  - name: ChargerMeter
    type: template
    template: orno
    usage: charge
    
    # Modbus TCP Waveshare_1 ORNO
    # RS485 via adapter (Modbus RTU)
    modbus: rs485serial
    id: 1
    device: COM3 # USB-RS485 Adapter Adresse
    baudrate: 9600 # Prüfe die Geräteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8E1" # Kommunikationsparameter für den Adapter  
    
  - name: GridMeter
    type: template
    template: huawei-dongle-powersensor
    usage: grid
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.155.9 # Hostname
    port: 502 # Port
    timeout: 15s # optional 
    
    
  - name: Sun2000_10KTL
    type: template
    template: huawei-dongle-powersensor
    usage: pv
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.155.9 # Hostname
    port: 502 # Port
    timeout: 15s # optional 
    
  - name: Luna2000
    type: template
    template: huawei-dongle-powersensor
    usage: battery
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.155.9 # Hostname
    port: 502 # Port
    timeout: 15s # optional 
    capacity: 15
    
    
  - name: PoolWPMeter
    type: template
    template: tasmota
    usage: charge
    host: 192.168.154.71 # IP-Adresse oder Hostname
    user: admin # Standard-User ist admin (optional)
    password: admin # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (optional)
    channel: 1 # Meterkanal (1-8) 
    
chargers:

    
    
  - name: ABL_EMH01_Isabell
    type: template
    template: abl
    # RS485 via TCP/IP (Modbus RTU)
    modbus: rs485serial
    id: 3
    device: COM2 # USB-RS485 Adapter Adresse
    baudrate: 38400 # Prüfe die Geräteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8E1" # Kommunikationsparameter für den Adapter
    

loadpoints:

  - title: CarPort # display name for UI
    charger: ABL_EMH01_Isabell 
    vehicle: OpelCorsaElektric # Referenz auf Standardfahrzeug
    mode: pv # charge mode (off, now, minpv, pv) 

    meter: ChargerMeter
    priority: 2
    soc:
      poll:
        mode: connected
        interval: 30m
      estimate: true
vehicles:

  - name: OpelCorsaElektric
    type: custom
    title: Opel Corsa Elektric 
    capacity: 50
    #vin: VXKUHZKXZP4261182
    soc: # battery soc (%)
      source: mqtt
      #clientid: EVCC
      user: EVCC
      password: evcc2023
      topic: evcc/corsaSOCFlobZFix/soc
      timeout: 24h # don't accept values older than timeout
      #scale: 0.001 # floating point factor applied to result, e.g. for Wh to kWh conversion
      # ...
      
    status: # optional charge status (A..F)
      source: mqtt
      #clientid: EVCC
      user: EVCC
      password: evcc2023
      topic: evcc/corsaSOCFlobZFix/status
      timeout: 24h # don't accept values older than timeout
      
    range: # optional electric range (km)
      source: mqtt
      #clientid: EVCC
      user: EVCC
      password: evcc2023
      topic: evcc/corsaSOCFlobZFix/range
      timeout: 24h # don't accept values older than timeout
      
    odometer: # optional electric range (km)
      source: mqtt
      #clientid: EVCC
      user: EVCC
      password: evcc2023
      topic: evcc/corsaSOCFlobZFix/odometer
      timeout: 24h # don't accept values older than timeout
      
    climater: # optional electric range (km)
      source: mqtt
      #clientid: EVCC
      user: EVCC
      password: evcc2023
      topic: evcc/corsaSOCFlobZFix/climater
      timeout: 24h # don't accept values older than timeout

    
site:
  - title: Zuhause # display name for UI
    meters:
      grid: GridMeter # grid meter reference
      pv: Sun2000_10KTL
      battery: Luna2000
    #vehicle: corsa # Referenz auf Standardfahrzeug  
    residualPower: 100
    #bufferSoc: 80
    #prioritySoc: 66
   # smartCostLimit
    
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price with price zones)
    type: fixed
    price: 0.32 # EUR/kWh
    #zones:
    #  - days: Mo-Fr
    #    hours: 20-0
    #    price: 0.2 # EUR/kWh
    #  - days: Sa,So
    #    price: 0.15 # EUR/kWh
        
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh
```

### Log details

```text
--
```

### What type of operating system are you running?

Windows

andig, 2024-03-04:
Logfile fehlt

closed 2024-03-04 by andig

Geri-G93, 2024-03-04:
> Logfile fehlt -> hab ich gerade nicht mehr
kann aus meinen tests jetzt soviel sagen
mit trace mqtt sieht man das er die neuen Werte bekommt, odometer ignoriert er wenn poll ausgeführt wird, nur der Soc wird aktualisiert
habs auch gerade im code nachgeschlagen, wenn ich das recht verstanden habe holt ihr odometer nur beim connecten/identifizieren ab, geht wenn die api schnell genug ist, psa ist wirklich träge und hängt 

und flobZ kann man im browser sehen, solang die Cache nur abgerufen wird kommen da keine neuen Daten, das muss aktiv derzeit angestosen werden