evcc-io/evcc #12676: Smart grid pricing - change of functionality not documented

realist625, 2024-03-04:
### Describe the bug

The app no longer initiates charging with smart tariff (chap tariff). 
Running the command ‘evict tariff -l debug show that the price is retrieved and the GUI show the ‘day time’ tariff (the higher price).   

### Steps to reproduce

1. Start evcc.
2. Check price in gui
3. Await ‘cheap tariff’
4. Charging is not started
5.
...

### Configuration details

```yaml
etwork:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxx

interval: 5s # control cycle interval

sponsortoken: 

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: eastron-sdm220_230
  id: 1
  device: /dev/serial0
  baudrate: 9600
  comset: 8N1
  usage: grid
  modbus: rs485serial
  name: grid1
- type: template
  template: solaredge-inverter
  id: 1
  host: xx.xx.xx.xx
  port: 1502
  usage: pv
  modbus: tcpip
  timeout: 10s
  name: pv1

- type: template
  template: victron-energy
  usage: battery
  host: xx.xx.xx.xx
  port: 502
  capacity: 9
  name: battery1

chargers:
- type: template
  template: keba
  host: xx.xx.xx.xx
  rfid: xxxxx  
[... 305 lines cut ...]
[site  ] DEBUG 2024/03/04 20:29:55 site power: 749W
[lp-1  ] DEBUG 2024/03/04 20:29:55 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/04 20:29:55 charge total import: 13503.912kWh
[lp-1  ] DEBUG 2024/03/04 20:29:55 charger status: B
[lp-1  ] DEBUG 2024/03/04 20:29:55 pv charge current: 0A = 0A + -3.26A (749W @ 1p)
[site  ] DEBUG 2024/03/04 20:30:00 ----
[lp-1  ] DEBUG 2024/03/04 20:30:00 charge power: 0W
[site  ] DEBUG 2024/03/04 20:30:00 pv power: 0W
[site  ] DEBUG 2024/03/04 20:30:00 battery soc: 28%
[site  ] DEBUG 2024/03/04 20:30:00 battery power: 1068W
[site  ] DEBUG 2024/03/04 20:30:00 grid meter: 48W
[site  ] DEBUG 2024/03/04 20:30:00 site power: 766W
[lp-1  ] DEBUG 2024/03/04 20:30:00 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/04 20:30:00 charge total import: 13503.912kWh
[lp-1  ] DEBUG 2024/03/04 20:30:00 charger status: B
[lp-1  ] DEBUG 2024/03/04 20:30:00 pv charge current: 0A = 0A + -3.33A (766W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

0.124.6

naltatis, 2024-03-05:
In the last release we moved the smart cost limits from a global dialog to the individual loadpoints. This is a change in the UI. You dont have to change you tariffs. A breaking change is, that you now need to set your limit(s) once again, since there is now "global" limit any more. You can do that with the settings icon at each loadpoint.

closed 2024-03-05 by naltatis

realist625, 2024-03-06:
Ok, I have seen the changes, but there still is an issue with the functionality, the charging did not start at the ‘cheap window start point’, I looked when I was about 25 min in to the low tariff period, and charging had not started, even though it was configured (see screenshots). 

Looking at the price of the session, it looks like thge system eventually figured out that the tariff was lower than the standard price, as the total is a mix of the two tariffs. 

Could be time issue, or do I need to do further changes to my config?

[image]
[image]
[image]

andig, 2024-03-06:
> even though it was configured (see screenshots)

Please open new issue with log file