evcc-io/evcc #12688: Heater tEmperature resets to 100 °C on every reboot

buliwyf42, 2024-03-05:
### Describe the bug

I use a Heater with a shelly Plug to heat water.
Unfortunately if i set a Temperature the Temp is reseted after every reboot. 
Previously this could be set with min and max soc. 
The Option for "PLAN" like on my go-e charger is not shown for the heater. 
Perhaps i just f.. some config but setting a Limit should survive a restart. 

### Steps to reproduce

1. Restart EVCC 
2. Set Temp
3. Restart
4. Set Temp again. 

### Configuration details

```yaml
- name: Warmwasser
    type: template
    usage: charge
    template: shelly-1pm
    host: 192.168.211.46
  - name: Warmwasser
    type: custom
    title: Warmwasser
    icon: waterheater
    phases: 1
    onIdentify:
      mode: off
    soc:
      source: mqtt
      topic: esphome/1wire/sensor/warmwasser/state
      timeout: 60m
  - title: WW-Steckdose
    charger: Warmwasser-Steckdose
    meter: Warmwasser
      #    mode: now
    vehicle: Warmwasser
#    phases: 1
#    mincurrent: 16
#    maxcurrent: 16
    enable:
      threshold: 0
      delay: 1m
    disable:
      threshold: 0
      delay: 1m
    soc:
      poll:
        mode: always
        interval: 30s
```

### Log details

```text
Nothing relevant.
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.6 (0d541fcb)

VolkerK62, 2024-03-05:
No idea, why "PLAN" isn´t shown within the loadpoint. Normally it is only unvisible if `heating: true` is configured.
You can set vehicle `limitSoc` also via API  https://docs.evcc.io/docs/reference/api#vehicles

VolkerK62, 2024-03-05:
>The Charger is configured heating: true. Therefore, "plan" should not appear?

Yes.

>Is there any other Way to configure if heating: true?

You can set it via REST API in 1% steps. (see link above)

andig, 2024-03-05:
@naltatis why should planner be hidden for heating devices? Seems hot water in the morning could be nice?

VolkerK62, 2024-03-05:
Problem seems to be, to define the needed energy.

naltatis, 2024-03-05:
Yes, energy does not correlate linearly to temperature increase and we don't have charging sessions. Means our resetting mechanism don't work here.

andig, 2024-03-05:
> Yes, energy does not correlate linearly to temperature increase

OT physically it should. But we sre missing a battery capacity here. 

Seems this is more a discussion now?