evcc-io/evcc #12689: pv 1 power: outdated aber nur wenn evcc als Service läuft

bofi99, 2024-03-05:
### Describe the bug

Hallo,
seit einiger Zeit bekomme ich immer die Fehlermeldung "pv 1 power: outdated" wenn evcc als Service läuft (evcc also mit sudo systemctl start evcc) und auch die korrekten PV-Werte werden nicht angezeigt (die eine PV ist immer 0 obwohl sie werte schickt.) Ich dachte erst das es wieder einmal eine Änderung der Konfig nötig ist und habe mich ewig gepielt (inc. evcc neu aufsetzen) aber keine Änderung erreicht.
Wenn ich evcc einfach in der Konsole mit evcc starte werden die korreten Werte angezeigt.

Anzeige mit evcc als Service
[image]

und in der Konsole gestartet:
[image]

irgendwie werden die Daten von pv1 (pv_dach) nicht angenommen obwohl sie im --meters log angezeigt werden.

### Steps to reproduce

1. start evcc als service : sudo systemctl start evcc
2. Fehlermeldung: pv 1 power: outdated, und keine Daten von pv1 (pv_dach)
3.
...

### Configuration details

```yaml
# uri: 0.0.0.0:7070 # uri for ui
network:
  schema: http
  host: 192.168.178.86
  port: 7070

interval: 15s # control cycle interval

sponsortoken: ------------------------------------------------

log: debug

meters:
- name: grid1
  type: custom
  power:
    source: http
    uri: http://192.168.178.57/pages/getinformation.php?export&meterindex=1
    jq: .Wirkleistung_Bezug.Leistung170

- name: pv_dach
  type: custom
  power:
    source: mqtt
    topic: envertech-pv/power
    timeout: 60s

- name: pv_bluetti_dc
  type: custom
  power:
    source: mqtt
    topic: bluetti/state/AC200M-2204000401665/dc_input_power
    timeout: 60s
    
- name: pv_bluetti_ac
  type: custom
  power:
    source: mqtt
    topic: bluetti/state/AC200M-2204000401665/ac_input_power
    timeout: 60s
    
 # name: bluetti
 # type: custom
 # power:
 #   source: mqtt
 #   topic: bluetti/state/AC200M-2204000401665/ac_output_power
 # soc:
 #   source: mqtt
 #   topic: bluetti/state/AC200M-2204000401665/total_battery_percent
 
- name: bluetti
  type: custom
  power:
    source: calc
    add:
    - source: mqtt
      topic: bluetti/state/AC200M-2204000401665/ac_output_power
      timeout: 30s
    - source: mqtt
      topic: bluetti/state/AC200M-2204000401665/dc_input_power
      timeout: 60s
      scale: -1
    - source: mqtt
      topic: bluetti/state/AC200M-2204000401665/ac_input_power
      timeout: 60s
      scale: -1
  soc:
    source: mqtt
    topic: bluetti/state/AC200M-2204000401665/total_battery_percent
    timeout: 60s

chargers:
- type: template
  template: go-e 
  host: 192.168.178.32  
  name: wallbox3

vehicles:
  name: ev2
  type: template
  template: ovms 
  title: Soul  
  user: ------  
  password: --------- 
  vehicleid: --------
  capacity: 30  
  phases: 1  
#  mode: now  
  
loadpoints:
- title: Go-Echarger
  charger: wallbox3
  vehicle: ev2
  mode: now

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv_bluetti_dc
    - pv_dach

    battery:
    - bluetti
  residualPower: 1

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: ------
  password: -----------------
```

### Log details

```text
nur mit evcc in der Konsole gestartet:
^Cpi@raspberrypi:~ $ evcc
[main  ] INFO 2024/03/05 13:34:05 evcc 0.124.6 (0d541fcb)
[main  ] INFO 2024/03/05 13:34:05 using config file: /home/pi/evcc.yaml
[main  ] INFO 2024/03/05 13:34:05 starting ui and api at :7070
[db    ] INFO 2024/03/05 13:34:05 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2024/03/05 13:34:05 connecting evcc-1565626898 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/03/05 13:34:05 tcp://localhost:1883 connected
[lp-1  ] DEBUG 2024/03/05 13:34:05 charge total import: 13996.000kWh
[site  ] INFO 2024/03/05 13:34:06 site config:
[site  ] INFO 2024/03/05 13:34:06   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/05 13:34:06     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/05 13:34:06     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/05 13:34:06     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/05 13:34:06     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/03/05 13:34:06   vehicles:
[site  ] INFO 2024/03/05 13:34:06     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/03/05 13:34:06 loadpoint 1:
[lp-1  ] INFO 2024/03/05 13:34:06   mode:        now
[lp-1  ] INFO 2024/03/05 13:34:06   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/03/05 13:34:06   meters:      charge ✓
[lp-1  ] INFO 2024/03/05 13:34:06     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/03/05 13:34:06 phase timer inactive
[lp-1  ] DEBUG 2024/03/05 13:34:06 pv timer inactive
[lp-1  ] INFO 2024/03/05 13:34:06 vehicle updated: unknown -> Soul
[site  ] WARN 2024/03/05 13:34:06 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/03/05 13:34:06 ----
[lp-1  ] DEBUG 2024/03/05 13:34:06 charge power: 0W
[site  ] DEBUG 2024/03/05 13:34:06 pv power: 264W
[site  ] DEBUG 2024/03/05 13:34:06 battery soc: 49%
[site  ] DEBUG 2024/03/05 13:34:06 battery power: -99W
[site  ] DEBUG 2024/03/05 13:34:06 grid meter: -16W
[site  ] DEBUG 2024/03/05 13:34:06 site power: -114W
[lp-1  ] DEBUG 2024/03/05 13:34:07 vehicle odometer: 87179km
[lp-1  ] DEBUG 2024/03/05 13:34:07 charge voltages: [228 227 226]V
[lp-1  ] DEBUG 2024/03/05 13:34:07 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/05 13:34:07 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/05 13:34:07 charge total import: 13996.000kWh
[lp-1  ] DEBUG 2024/03/05 13:34:07 charger status: A
[lp-1  ] INFO 2024/03/05 13:34:07 car disconnected
[lp-1  ] DEBUG 2024/03/05 13:34:07 set session soc limit: 0
[lp-1  ] DEBUG 2024/03/05 13:34:07 set session energy limit: 0

evcc mit sudo systemctl start evcc:
pi@raspberrypi:~ $ sudo systemctl start evcc
pi@raspberrypi:~ $ evcc -l debug meter
[main  ] INFO 2024/03/05 13:35:38 evcc 0.124.6 (0d541fcb)
[main  ] INFO 2024/03/05 13:35:38 using config file: /home/pi/evcc.yaml
[db    ] INFO 2024/03/05 13:35:38 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2024/03/05 13:35:38 connecting evcc-324250749 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/03/05 13:35:38 tcp://localhost:1883 connected
grid1
-----
Power: 8W

pv_dach
-------
Power: 164W

pv_bluetti_dc
-------------
Power: 100W

pv_bluetti_ac
-------------
Power: 0W

bluetti
-------
Power: -100W
Soc:   49%

und das weitere log:
pi@raspberrypi:~ sudo systemctl start evcc
pi@raspberrypi:~ $ sudo journalctl -fau evcc
-- Journal begins at Sun 2024-02-11 14:27:45 CET. --
Mär 05 13:38:10 raspberrypi evcc[25718]: [site  ] DEBUG 2024/03/05 13:38:10 battery soc: 49%
Mär 05 13:38:10 raspberrypi evcc[25718]: [site  ] DEBUG 2024/03/05 13:38:10 battery power: -101W
Mär 05 13:38:10 raspberrypi evcc[25718]: [site  ] DEBUG 2024/03/05 13:38:10 grid meter: 87W
Mär 05 13:38:10 raspberrypi evcc[25718]: [site  ] DEBUG 2024/03/05 13:38:10 site power: 36W
Mär 05 13:38:10 raspberrypi evcc[25718]: [lp-1  ] DEBUG 2024/03/05 13:38:10 vehicle odometer: 87179km
Mär 05 13:38:10 raspberrypi evcc[25718]: [lp-1  ] DEBUG 2024/03/05 13:38:10 charge voltages: [229 228 227]V
Mär 05 13:38:10 raspberrypi evcc[25718]: [lp-1  ] DEBUG 2024/03/05 13:38:10 detected connected phases: 3p
Mär 05 13:38:10 raspberrypi evcc[25718]: [lp-1  ] DEBUG 2024/03/05 13:38:10 charge currents: [0 0 0]A
Mär 05 13:38:10 raspberrypi evcc[25718]: [lp-1  ] DEBUG 2024/03/05 13:38:10 charge total import: 13996.000kWh
Mär 05 13:38:10 raspberrypi evcc[25718]: [lp-1  ] DEBUG 2024/03/05 13:38:10 charger status: A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.6 (0d541fcb)

andig, 2024-03-05:
Wo finde ich das Problem im Logfile?

bofi99, 2024-03-05:
Ja eben nicht. Das ist ja das Problem. Ich kann den Fehler eben nicht nachvollziehen. Im Log werden Daten angezeigt, im Web-Frontend dann die Fehlermeldung. Die Bilder in der Beschreibung zeigen das Problem.

andig, 2024-03-05:
Ins UI können rein technisch nur Fehler kommen die es im Log gibt. Es braucht slso ein aussagekräftiges Logfile.

bofi99, 2024-03-05:
Ja. das ist ja das . Natürlich steht im Logfile des Dienstes pv 1 power: outdated:
Mär 05 17:11:41 raspberrypi evcc[25784]: [lp-1  ] DEBUG 2024/03/05 17:11:41 charge power: 0W
Mär 05 17:11:41 raspberrypi evcc[25784]: [site  ] ERROR 2024/03/05 17:11:41 pv 1 power: outdated
Mär 05 17:11:41 raspberrypi evcc[25784]: [site  ] DEBUG 2024/03/05 17:11:41 pv power: 0W
Mär 05 17:11:41 raspberrypi evcc[25784]: [site  ] DEBUG 2024/03/05 17:11:41 battery soc: 100%
Mär 05 17:11:41 raspberrypi evcc[25784]: [site  ] DEBUG 2024/03/05 17:11:41 battery power: 0W
Mär 05 17:11:41 raspberrypi evcc[25784]: [site  ] DEBUG 2024/03/05 17:11:41 grid meter: 346W
Mär 05 17:11:41 raspberrypi evcc[25784]: [site  ] DEBUG 2024/03/05 17:11:41 site power: 396W
Mär 05 17:11:41 raspberrypi evcc[25784]: [lp-1  ] DEBUG 2024/03/05 17:11:41 charge voltages: [227 227 225]V
Mär 05 17:11:41 raspberrypi evcc[25784]: [lp-1  ] DEBUG 2024/03/05 17:11:41 detected connected phases: 3p
Mär 05 17:11:41 raspberrypi evcc[25784]: [lp-1  ] DEBUG 2024/03/05 17:11:41 charge currents: [0 0 0]A
Mär 05 17:11:41 raspberrypi evcc[25784]: [lp-1  ] DEBUG 2024/03/05 17:11:41 charge total import: 13996.000kWh
Mär 05 17:11:41 raspberrypi evcc[25784]: [lp-1  ] DEBUG 2024/03/05 17:11:41 charger status: A
Mär 05 17:11:50 raspberrypi evcc[25784]: [site  ] DEBUG 2024/03/05 17:11:50 ----

Wenn ich aber evcc über die Konsole starte funktioniert es ja:

[site  ] DEBUG 2024/03/05 21:25:38 ----
[lp-1  ] DEBUG 2024/03/05 21:25:38 charge power: 0W
[site  ] DEBUG 2024/03/05 21:25:38 pv power: 0W
[site  ] DEBUG 2024/03/05 21:25:38 battery soc: 97%
[site  ] DEBUG 2024/03/05 21:25:38 battery power: 0W ---Aktuell ist ja Nacht, pass ja so.
[site  ] DEBUG 2024/03/05 21:25:38 grid meter: 272W
[site  ] DEBUG 2024/03/05 21:25:38 site power: 273W
[lp-1  ] DEBUG 2024/03/05 21:25:38 vehicle odometer: 87179km
[lp-1  ] DEBUG 2024/03/05 21:25:38 charge voltages: [227 226 223]V
[lp-1  ] DEBUG 2024/03/05 21:25:38 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/05 21:25:38 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/05 21:25:38 charge total import: 13996.000kWh
[lp-1  ] DEBUG 2024/03/05 21:25:38 charger status: A

irgendie werde ich da nicht schlau.

bofi99, 2024-03-05:
Ach Timeout auf 10h hilft nichts:
meters:
......

- name: pv_dach
  type: custom
  power:
    source: mqtt
    topic: envertech-pv/power
    timeout: 10h

andig, 2024-03-05:
> Ja. das ist ja das . Natürlich steht im Logfile des Dienstes pv 1 power: outdated:

In Deinem Issue ist davon nichts zu sehen. Ist das jetzt ein anderes Log? Bitte mal vollständig ab Start.

andig, 2024-03-05:
> Wenn ich loglevel: trace einstelle ändert sich das journalctl log nicht

Ich denke du hast zwei Configs und probierst mit der falschen rum

andig, 2024-03-06:
Bitte mit aktuellem Release nochmal testen. Bei Bedarf mit vollstädnigem Log aktualisieren.

closed 2024-03-06 by andig