evcc-io/evcc #12691: Lädt im Min+PV obwohl nicht genügend Sonne vorhanden

Kenterfie, 2024-03-05:
### Describe the bug

Seit dem Wechsel von v0.124.4 auf v0.124.6 lädt evcc trotz zu wenig PV Leistung weiter. Wenn ich auf die PV Stellung gehe, kommt wie zu erwarten, der Timer und das Laden wird beendet. Gab es eine Änderung im Verhalten die ich übersehen habe?

### Steps to reproduce

Auto wurde angeschlossen als genug PV Überschuss vorhanden war und wechselte direkt in den Lademodus. Nach einiger Zeit verringerte sich die PV Leistung stark, aber der Ladevorgang wurde weiterhin mit voller Leistung fortgeführt.
[image]

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: ha.fritz.box
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  ocpp: trace

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
[... 192 lines cut ...]
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.124.6

VolkerK62, 2024-03-05:
Voll normal. `Min+PV` lädt immer.
https://docs.evcc.io/docs/guides/charging#was-bedeuten-die-verschiedenen-lademodi

Kenterfie, 2024-03-05:
Ok, danke für die Richtigstellung. Nutze evcc noch nicht all zu lange und hatte eher angenommen, dass er bei "min+PV" nur bis zum Minimalwert den normalen Hausstrom nutzt. Das hätte den Vorteil das der Wagen dann mindestens bis zum Minimalwert geladen ist, aber wenn genug PV Leistung vorhanden ist er dann auch das Auto weiter lädt, aber wenn das nicht der Fall ist schade.

VolkerK62, 2024-03-05:
Das was du meinst, nennt sich `minSoc`.
Lässt sich wie folgt einstellen:
In der Benutzeroberfläche (UI) im Loadpoint Mitte unten auf "Plan" klicken.
Dann den Tab "Ankunft" auswählen.
Dort kannst du dann das Minimal- und das Standartziel einstellen.

Kenterfie, 2024-03-05:
Vielen Dank für die schnelle und kompetente Hilfe. Dann ist das Issue hiermit geschlossen.

closed 2024-03-05 by Kenterfie