evcc-io/evcc #12697: EVCC Modul in OpenWB liefert keine SoC Werte

waxipedia, 2024-03-05:
### Describe the bug

Ich nutze das EVCC Modul der OpenWB

Im Log steht seit knapp 24 Stunden folgende Meldung:

2024-03-05 17:44:34: PID: 20931: Lp1: Error from EVCC: 2024/03/05 17:44:34 rpc error: code = Unavailable desc = connection error: desc = "transport: authentication handshake failed: read tcp 192.168.0.83:49132->66.241.124.2:8080: read: connection reset by peer" (LV0)
2024-03-05 17:44:34: PID: 20931: Lp1: Requesting SoC (LV0)

Vielleicht kann ja mal jemand einen Blick drauf werfen was schief läuft.

### Steps to reproduce

SoC Abfragen über OpenWB.

### Configuration details

```yaml
?
```

### Log details

```text
Ich nutze das EVCC Modul der OpenWB

Im Log steht seit knapp 24 Stunden folgende Meldung:

2024-03-05 17:44:34: PID: 20931: Lp1: Error from EVCC: 2024/03/05 17:44:34 rpc error: code = Unavailable desc = connection error: desc = "transport: authentication handshake failed: read tcp 192.168.0.83:49132->66.241.124.2:8080: read: connection reset by peer" (LV0)
2024-03-05 17:44:34: PID: 20931: Lp1: Requesting SoC (LV0)

Vielleicht kann ja mal jemand einen Blick drauf werfen was schief läuft.
```

### What type of operating system are you running?

Linux

andig, 2024-03-05:
Bitte nochmal probieren: https://status.flyio.net

Bucky2k, 2024-03-05:
Still present:

2024-03-05 22:41:38: PID: 29474: Lp1: Error from EVCC: 2024/03/05 22:41:38 rpc error: code = Unavailable desc = connection error: desc = "transport: authentication handshake failed: read tcp 192.168.1.80:51134->66.241.124.2:8080: read: connection reset by peer" (LV0)

andig, 2024-03-05:
Mhhm. Es ist die gleiche Serverinstanz wie vor 3 Tagen und lokal klappt der Zugriff. Ich bin ratlos. Welchen Timestamp hat das evcc soc Binary auf der OpenWB? Das müsste schon älter sein?

andig, 2024-03-05:
Geht lokal:

```
gor main.go vehicle -c cfg/audi.yaml
[main  ] INFO 2024/03/05 23:13:26 evcc 0.0.0
[main  ] INFO 2024/03/05 23:13:26 using config file: cfg/audi.yaml
[db    ] INFO 2024/03/05 23:13:26 using sqlite database: /Users/andig/.evcc/evcc.db
Soc:      97%
```

Irgendwas ist also am Aufruf von der OpenWB aus anders. Ich könnte probehalber mal ein neues Binary zur Verfügung stellen, sehe aber auch nicht was sich geändert haben sollte.

Bucky2k, 2024-03-05:
Sie Soc binary hast Du zuletzt im März 2022 hochgeladen:
[image]

andig, 2024-03-05:
Magst Du mal den aktuellen Stand probieren? Bitte Mail an info@evcc.io. Unnötig zu sagen, dass man keine Binaries aus dem Internet runter laden soll...

Bucky2k, 2024-03-06:
Alles klar, neue binary läuft. Soll ich einen PR bei openWB zum Austausch der Binary einstellen?

andig, 2024-03-06:
Wenn du da gute Chancen hast gerne, sonst kann ich das auch machen. Was sich da geändert hat bleibt ein Rätsel...

Bucky2k, 2024-03-06:
> Wenn du da gute Chancen hast gerne, sonst kann ich das auch machen. Was sich da geändert hat bleibt ein Rätsel...

Top, da warst Du schneller

closed 2024-03-08 by andig

Bucky2k, 2025-02-16:
Hi Andi,

ich habe meinen Token verschlampt, bzw. evtl. ist er auch ausgelaufen, da
ich testweise unseren eGolf anderweitig in die OpenWB eingebunden habe. Nun
wollte ich einen neuen Cupra Born anbinden, finde aber den Token nicht
wieder. Bin zwischenzeitlich von 1,9 zu 2.0 auf der OpenWB umgestiegen.

Gruß

Bucky

Bucky2k, 2025-02-16:
Hab ihn doch noch gefunden und scheint auch noch zu gehen.

Gruß

Bucky