evcc-io/evcc #12698: new dynamic tariffs available from Switzerland (Groupe E provider)

michelfrederic, 2024-03-05:
My electricity provider (Groupe E from Switzerland) have a dynamic tariffs. It could be interresting to implement it into the evcc solution.

The web API is based on a HTTP GET request and is very easy.

I added the file description to this post (sorry the document is in french).

[tarif_vario_WEB_api.pdf](https://github.com/evcc-io/evcc/files/14500726/tarif_vario_WEB_api.pdf)

Is it possible to implement this new dynamic tariff to the evcc ?

Thank you in advance.

Frederic Michel

andig, 2024-03-05:
Are these three different tariffs?

michelfrederic, 2024-03-06:
It seem's that there are only 2 tariffs. 
- VARIO_PLUS
- DT_PLUS

The value of the VARIO_GRID is only the price of the grid, not including the transport and others taxes.

But to be sure, I created this morning a ticket to the provider to have the confirmation.

michelfrederic, 2024-03-08:
OK, yes the DT_PLUS is static. 
So, yes, the VARIO_PLUS would be the only actual gap.

closed 2024-03-08 by andig

michelfrederic, 2024-03-14:
Ok, yes, I will ask them to add evcc.io as a compatible energy management solution

michelfrederic, 2024-03-14:
OK, it's done. evcc is mentionned as a compatible energy management.
[https://www.groupe-e.ch/de/energie/elektrizitaet/privatkunden/vario]

spasdev, 2025-09-29:
@michelfrederic 
I would to make propose two changes to Groupe E. They are as well my local energy provider :-) 

1) complete the line by saying the application is multinlingual, only the documentation is in DE/EN. 
Currently, I think it gives the (wrong) impression the application is not at all in French.
2) make a reference to a [local service](https://captawatt.ch/) I'm providing to help people installing evcc. 

Are you ok with the changes? I reached out to them and they asked if you are ok with the changes as you made the initial announcement.

andig, 2025-09-29:
Please feel free to open PR. Local support does not belong into the tariff decsription!

roto11r, 2025-12-25:
If it can help someone here are the tariffs for " double tarif " for Groupe E with  ( with pricing zones before and after the first january 2026 )

hope it works :)

Merry Christmas to all

`        zones:
          - days: Mon-Sun
            hours: 7-21
            price: 0.2917   # HT
          - days: Mon-Sun
            hours: 21-24
            price: 0.1881   # BT
          - days: Mon-Sun
            hours: 0-7
            price: 0.1881   # BT

    # -----------------------------
    # Dès le 1er janvier 2026
    # -----------------------------
    - from: 2026-01-01
      tariff:
        type: fixed
        zones:
          - days: Mon-Sun
            hours: 7-12
            price: 0.2917   # HT
          - days: Mon-Sun
            hours: 12-17
            price: 0.1881   # BT
          - days: Mon-Sun
            hours: 17-23
            price: 0.2917   # HT
          - days: Mon-Sun
            hours: 23-24
            price: 0.1881   # BT
          - days: Mon-Sun
            hours: 0-7
            price: 0.1881   # BT

feedin:
  type: fixed
  zones:
    - days: Mon-Sun
      hours: 7-21
      price: 0.10`

just edit the prices when they change