evcc-io/evcc #12703: Ladeplanung: "Zielzeit nicht erreichbar" (Anzeige- oder Berechnungs

An-Wi, 2024-03-06:
### Describe the bug

Sporadisch auftretender Warnhinweis, dass Zielzeit nicht erreichbar ist, obwohl evcc eigentlich (fast) pünktlich startete (nach meinen Beobachtungen vielleicht 1/2 bis 1 Minute zu spät).
Im UI steht dann die Hinweismeldung, in der aber voraussichtliche und geplante Zielzeit "optisch" identisch sind - sicherlich einem Rundungsproblem/Nichtanzeigen der Sekunden geschuldet.
[image]

Idee: Entweder Ladeplanung mit zusätzlichem Puffer von 1 Minute vorher starten...
oder Alternative: (finde ich besser) Zielzeiten ohne Sekunden - also so wie im UI angezeigt - miteinander vergleichen für Entscheidung, ob Warnhinweis notwendig ist oder nicht

### Steps to reproduce

Nicht direkt reproduzierbar. 
Sehr sporadisch aufgetreten -> betrifft ca. 1 von 10 ausgeführten Ladeplanungen.

### Configuration details

```yaml
[...]

interval: 20s   

[...]
```

### Log details

```text
[site  ] DEBUG 2024/03/05 22:53:55 ----
[lp-1  ] DEBUG 2024/03/05 22:53:55 charge power: 0W
[site  ] DEBUG 2024/03/05 22:53:55 pv power: 0W
[site  ] DEBUG 2024/03/05 22:53:55 grid power: 167W
[site  ] DEBUG 2024/03/05 22:53:55 grid currents: [0.392 0.526 0.536]A
[site  ] DEBUG 2024/03/05 22:53:55 site power: -33W
[lp-1  ] DEBUG 2024/03/05 22:53:55 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/05 22:53:55 charge total import: 3342.068kWh
[lp-1  ] DEBUG 2024/03/05 22:53:55 charger status: B
[lp-1  ] DEBUG 2024/03/05 22:53:55 next soc poll remaining time: 42m40s
[lp-1  ] DEBUG 2024/03/05 22:53:55 plan: charge 3h51m10s between 2024-03-05 22:53:56 +0100 CET until 2024-03-06 02:45:00 +0100 CET (overruns by 6s, power: 11040W, avg cost: 0.280)
[lp-1  ] DEBUG 2024/03/05 22:53:55 switched phases: 3p
[lp-1  ] DEBUG 2024/03/05 22:53:55 max charge current: 16A
[lp-1  ] DEBUG 2024/03/05 22:53:55 charger enable
[lp-1  ] DEBUG 2024/03/05 22:53:55 wake-up timer: start
[lp-1  ] DEBUG 2024/03/05 22:53:55 pv timer elapse
[lp-1  ] DEBUG 2024/03/05 22:53:55 pv timer inactive
[site  ] DEBUG 2024/03/05 22:54:15 ----
[lp-1  ] DEBUG 2024/03/05 22:54:15 charge power: 10714W
[site  ] DEBUG 2024/03/05 22:54:15 pv power: 0W
[site  ] DEBUG 2024/03/05 22:54:15 grid power: 10985W
[site  ] DEBUG 2024/03/05 22:54:15 grid currents: [15.4 15.4 15.4]A
[site  ] DEBUG 2024/03/05 22:54:15 site power: 10785W
[lp-1  ] DEBUG 2024/03/05 22:54:15 charge currents: [15.1 15.1 15.1]A
[lp-1  ] DEBUG 2024/03/05 22:54:15 charge total import: 3342.101kWh
[lp-1  ] DEBUG 2024/03/05 22:54:15 charger status: C
[lp-1  ] INFO 2024/03/05 22:54:15 start charging ->
[lp-1  ] DEBUG 2024/03/05 22:54:15 wake-up timer: stop
[db    ] TRACE 2024/03/05 22:54:16 INSERT INTO `sessions` (`created`,`finished`,`loadpoint`,`identifier`,`vehicle`,`odometer`,`meter_start_kwh`,`meter_end_kwh`,`charged_kwh`,`charge_duration`,`solar_percentage`,`price`,`price_per_kwh`,`co2_per_kwh`) VALUES ("2024-03-05 22:54:15.968","0000-00-00 00:00:00","Einfahrt","","Ariya",NULL,3342.068,NULL,0,NULL,NULL,NULL,NULL,NULL) RETURNING `id` 1 <nil>
[nissan] TRACE 2024/03/05 22:54:16 GET https://alliance-platform-caradapter-prod.apps.eu2.kamereon.io/car-adapter/v1/cars/***/battery-status
[nissan] TRACE 2024/03/05 22:54:16 {"id":"***","batteryLevel":31,"plugStatus":1,"chargeStatus":1}
[lp-1  ] DEBUG 2024/03/05 22:54:16 soc estimated: 31.04% (vehicle: 31.00%)
[lp-1  ] DEBUG 2024/03/05 22:54:16 vehicle soc: 31%
[lp-1  ] ERROR 2024/03/05 22:54:16 vehicle range: not available
[lp-1  ] DEBUG 2024/03/05 22:54:16 plan: charge 3h50m58s between 2024-03-05 22:54:17 +0100 CET until 2024-03-06 02:45:00 +0100 CET (overruns by 15s, power: 11040W, avg cost: 0.280)
[site  ] DEBUG 2024/03/05 22:54:35 ----
[lp-1  ] DEBUG 2024/03/05 22:54:35 charge power: 10720W
[site  ] DEBUG 2024/03/05 22:54:35 pv power: 0W
[site  ] DEBUG 2024/03/05 22:54:36 grid power: 10995W
[site  ] DEBUG 2024/03/05 22:54:36 grid currents: [15.4 15.4 15.4]A
[site  ] DEBUG 2024/03/05 22:54:36 site power: 10795W
[lp-1  ] DEBUG 2024/03/05 22:54:36 charge currents: [15.1 15.1 15.1]A
[lp-1  ] DEBUG 2024/03/05 22:54:36 charge total import: 3342.161kWh
[... 152 lines cut ...]
[lp-1  ] INFO 2024/03/06 02:33:15 stop charging <-
[lp-1  ] DEBUG 2024/03/06 02:33:15 charge total import: 3381.076kWh
[db    ] TRACE 2024/03/06 02:33:15 UPDATE `sessions` SET `created`="2024-03-05 22:54:15.968",`finished`="2024-03-06 02:33:15.816",`loadpoint`="Einfahrt",`identifier`="",`vehicle`="Ariya",`odometer`=NULL,`meter_start_kwh`=3342.068,`meter_end_kwh`=3381.076,`charged_kwh`=39.007,`charge_duration`="3h39m0s",`solar_percentage`=0,`price`=10.910257900000053,`price_per_kwh`=0.2797000000000014,`co2_per_kwh`=196.26592662855296 WHERE `id` = 144 1 <nil>
[nissan] TRACE 2024/03/06 02:33:15 GET https://alliance-platform-caradapter-prod.apps.eu2.kamereon.io/car-adapter/v1/cars/***/battery-status
[nissan] TRACE 2024/03/06 02:33:16 {"id":"***","plugStatus":0,"chargeStatus":0,"batteryLevel":75}
[lp-1  ] DEBUG 2024/03/06 02:33:16 soc estimated: 75.00% (vehicle: 75.00%)
[lp-1  ] DEBUG 2024/03/06 02:33:16 vehicle soc: 75%
[lp-1  ] ERROR 2024/03/06 02:33:16 vehicle range: not available
[lp-1  ] DEBUG 2024/03/06 02:33:16 available power -75W < 4140W min 3p threshold
[lp-1  ] DEBUG 2024/03/06 02:33:16 phase scale1p in 0s
[lp-1  ] DEBUG 2024/03/06 02:33:16 switched phases: 1p
[lp-1  ] DEBUG 2024/03/06 02:33:16 phase timer inactive
[lp-1  ] DEBUG 2024/03/06 02:33:16 pv charge current: 0A = 0A + -0.324A (75W @ 1p)
[lp-1  ] DEBUG 2024/03/06 02:33:16 pv enable timer reset
[lp-1  ] DEBUG 2024/03/06 02:33:16 pv timer inactive
[site  ] DEBUG 2024/03/06 02:33:35 ----
```

### What type of operating system are you running?

Windows

### Version

0.124.4

An-Wi, 2024-03-22:
Nach Update der Version 0.124.10:

Der Hinweistext ist nun verschwunden, wenn es nur eine geringe Zeitüberschreitung gibt, aber leider sind die orangen Einfärbungen sowohl in der Ladeplanung als auch im Haupt-UI noch vorhanden. Das verwirrt ein wenig...
[image]
[image]

naltatis, 2024-03-23:
@andig magst du da mal reinschauen. An dieser Stelle wird rein das `planOverrun` bool ausgewertet das der Planer zum aktuell laufenden Plan published. Das ist noch mal anders als die Detailansicht/Preview im Modal.

Hier ist der Code: https://github.com/evcc-io/evcc/blob/master/core/loadpoint_plan.go#L125-L129

andig, 2024-03-24:
@naltatis wie hast Du das im Modal gelöst? Über zeitliche Flexibilität? Dann könnten wir hier die overrun duration publizieren?

@An-Wi was wäre das erwünschte Verhalten? Ab wann ist "zu lange" ok so dass es keine Warnung erzeugen sollte?

naltatis, 2024-03-24:
@andig Ja, ich hab bei der projizierten Zeit die Sekunden entfernt (auf Min abgerundet) und beim gewünschten Ziel aufgerundet. Das war in JS für mich der einfachste Weg. Das ist äquivalent zu einer Toleranz von max 1min. In Go kann man das sicher eleganter abfragen. Da ist Zeit ordentlich modelliert :D

andig, 2024-03-24:
Ich würde die Entscheidung trotzdem dem UI überlassen oder sollte das Backend schon entscheiden, ob/das 60s "ok" sind? UI wäre hier konsistent?

naltatis, 2024-03-24:
Dann macht der gepublishte `planOverrun` aber keinen Sinn mehr und wir müssten stattdessen ein `planProjectedEnd` (analog zu `planProjectedStart`) publishen.

naltatis, 2024-03-24:
Ich find auch gut, wenn wir die Logik auf einer Seite halten. Ob in UI oder in Backend wäre mir relativ egal.

Wenn wir sagen das soll ins Backend müssten `/plan/preview` und `/plan` in ihrer Antwort ein Bool enthalten das mitteilt ob die gewünschte Planung `reachable` ist.

Wenn wirs in der UI machen brauchen wir den geschätzten Endwert im globalen Publishing (nicht nur als HTTP Call `/plan/[preview]`).

andig, 2024-03-24:
Wir machen es doch an anderer Stelle schon in der UI? Ich würde einfach aus `planOverrun` eine `time.Duration` machen, die müsste im UI mit 60s verglichen werden. Der von Dir verlinkte vergleich testet übrigens nicht auf 60s sondern prüft nur, ob beide in der gleichen Minute liegen ;)

naltatis, 2024-03-24:
> Der von Dir verlinkte vergleich testet übrigens nicht auf 60s sondern prüft nur, ob beide in der gleichen Minute liegen ;)

Ich hab ja auch `max 1min` geschrieben :D

andig, 2024-03-24:
Es wird halt wieder asynchron wenn wir planoverrun of 1min prüfen ;) JS kann keine Dates subtrahieren? Nichtmal als UNIX?

naltatis, 2024-03-24:
> JS kann keine Dates subtrahieren? Nichtmal als UNIX?

Klar geht das, ist aber aktuell nicht so, weil das für den Anwendungsfall ausreichend war.

An-Wi, 2024-03-24:
> @An-Wi was wäre das erwünschte Verhalten? Ab wann ist "zu lange" ok so dass es keine Warnung erzeugen sollte?

Ich würde 1-5min "drüber laufen" für ok halten, aber wenn es nur 60s sind und dann steht bspw. "Ladeziel nicht erreichbar. Voraussichtliches Ende 02:01" bei eingestellter Zielzeit von "02:00", dann ist es zumindest inhaltlich nachvollziehbar.