evcc-io/evcc #12708: Unplausible Werte für Einspeisung und Batterieladung bei SolarEdge Hybridwechselrichter

MarkusGH, 2024-03-06:
### Describe the bug

In Situationen, wo die Batterie meines SolarEdge Hybridwechselrichters zwischen Laden und Entladen wechselt, zeigt evcc unplausible Werte für Netzeinspeisung/Netzbezug.

[image]

Die Spikes haben jeweils die Breite des Regelintervalls, verkürzt man das Regelintervall so werden auch die Spikes schmäler:

[image]

Links 10 sec, rechts 30 sec Intervall.

Das Verhalten ist insofern unverständlich, als es in meiner Installation einen Bezugszähler gibt, der Netzeinspeisung/Netzbezug direkt misst. Parallele Messung mit mbmd / modbus Bridge im 3 - Sekunden Intervall zeigt, dass es die Netzeinspeisung in der Höhe wie sie evcc anzeigt nicht gibt.

- 

### Steps to reproduce

1. evcc mit SE10K-RWS, LG Batterie
2. Situation, wo die Batterie zwischen Laden/Entladen wechselt

### Configuration details

```yaml
network:
  schema: http
  host: localhost
  port: 7070

interval: 30s

...

meters:
- name: grid
  type: template
  template: solaredge-hybrid
  id: 1
  host: solaredge.xxx.de
  port: 502
  usage: grid
  modbus: tcpip
  timeout: 20s
- name: pv1
  type: template
  template: solaredge-hybrid
  id: 1
  host: solaredge.xxx.de
  port: 502
  usage: pv
  modbus: tcpip
  timeout: 20s
- name: battery1
  type: template
  template: solaredge-hybrid
  id: 1
  host: solaredge.xxx.de
  port: 502
  usage: battery
  modbus: tcpip
  timeout: 20s

...

site:
  title: H73
  meters:
    grid: grid
    pv: [pv1]
    battery: [battery1]
[... 115 lines cut ...]
Mar  6 11:48:47 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:48:47 battery soc: 11%
Mar  6 11:48:47 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:48:47 battery power: 1321W
Mar  6 11:48:47 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:48:47 grid meter: 34W
Mar  6 11:48:47 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:48:47 site power: 1356W
Mar  6 11:49:17 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:49:17 ----
Mar  6 11:49:17 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:49:17 pv power: 2333W
Mar  6 11:49:17 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:49:17 battery soc: 11%
Mar  6 11:49:17 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:49:17 battery power: 1151W
Mar  6 11:49:17 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:49:17 grid meter: 24W
Mar  6 11:49:17 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:49:17 site power: 1176W
Mar  6 11:49:47 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:49:47 ----
Mar  6 11:49:47 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:49:47 pv power: 2315W
Mar  6 11:49:47 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:49:47 battery soc: 11%
Mar  6 11:49:47 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:49:47 battery power: -666W
Mar  6 11:49:47 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:49:47 grid meter: 27W
Mar  6 11:49:47 homebridge evcc[2443353]: [site  ] DEBUG 2024/03/06 11:49:47 site power: -638W
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.7

andig, 2024-03-06:
> Parallele Abfrage mit mbmd via modbus Bridge im 3 - Sekunden Intervall zeigt, dass es die Netzeinspeisung in der Höhe wie sie evcc anzeigt real nicht gibt.

evcc kann ja nur zeigen, was es vom WR bekommt. Dann wäre mal zu prüfen, ob du andere Register verwendest als im Tempalte hinterlegt. Schaust Du mal rein?

MarkusGH, 2024-03-06:
Ich habe nichts geändert - fragt MBMD denn andere Register ab als evcc?

```
Hier die mbmd config:

adapters:
- device: evcc.xxx.de:502
  rtu: false

devices:
- name: SE10K
  type: solaredge
  id: 1
  subdevice: 0
- name: GRID
  type: solaredge
  id: 1
  subdevice: 1
```

andig, 2024-03-08:
Das müsste geprüft werden. Am besten Modbus Log abgleichen.

MarkusGH, 2024-03-09:
Protokolliert evcc im modbus Trace auch die Daten, die über den Proxy gehen?

premultiply, 2024-03-09:
residualPower: 1 ist vielleicht auch etwas sehr wenig.

Ansonsten sieht das für mich soweit erstmal "normal" aus. Wie wenn irgendwo ein größerer taktender Verbraucher (z.B. Herdplatte) läuft der dann von der Batterie ausgeregelt wird.

MarkusGH, 2024-03-09:
> residualPower: 1 ist vielleicht auch etwas sehr wenig.

residualpower 1 ist eigentlich 0 - um Warnungen zu vermeiden. Ich halte nichts davon die Hausbatterie mit ihrer bescheidenen Effizient unnötig zu laden, wenn man die Leistung stattdessen in ein angeschlossenes EV stecken kann.

> Ansonsten sieht das für mich soweit erstmal "normal" aus. Wie wenn irgendwo ein größerer taktender Verbraucher (z.B. Herdplatte) läuft der dann von der Batterie ausgeregelt wird.

War hier aber eher nicht der Fall. 

Wie ich geschrieben habe: Ich polle per MBMD/modbus Proxy parallel und erfasse laufend MIN/MAX Werte. Dort sieht man diese Spikes nicht. Außerdem zeigt mit EVCC bisweilen negative PV Leistungen im Bereich > 1kW an.

Alles sehr seltsam.

premultiply, 2024-03-09:
Dann hast du diese Einstellung nicht verstanden.
Die Regelung kann ohne diesen Wert nicht stabil arbeiten und führt u.a. zu solchen Effekten.

MarkusGH, 2024-03-09:
> Dann hast du diese Einstellung nicht verstanden. 

Möglicherweise.
Gibt es irgendwo eine Erklärung oder ein Diskussion zu dem Thema?

> Die Regelung kann ohne diesen Wert nicht stabil arbeiten und führt u.a. zu solchen Effekten.

Ich gehe nicht davon aus, dass das hier ursächlich ist. Stelle es aber jetzt dennoch auf 100 und beobachte weiter.

andig, 2024-03-11:
Ohne Modbus trace beider Varianten würde ich dann mal zu machen?

MarkusGH, 2024-03-11:
@andig: Bitte nicht.

Protokolliert evcc im modbus Trace auch die Daten, die über den modbus Proxy gehen?

MarkusGH, 2024-03-11:
Denn ich schaue mir das genau in diesem Moment an und es ist mal wieder wild. 
Aktuell hat evcc auch nichts zu regeln, denn es ist kein Fahrzeug mit freier Kapazität verfügbar.
Insofern kann die Regelung durch evcc nicht schuld sein.

andig, 2024-03-11:
Mir ist nicht klar, was wir hier beitragen können. Erstmal Hinweis von @premultiply befolgen, dann mal 

    evcc meter

mit trace log für richtige + falsche Version machen. Dann können wir wieder rein schauen.

MarkusGH, 2024-03-11:
@andig: Bitte lies meine Posts:

1) Ich habe den Hinweis von premultiply schon lange befolgt. Keine Änderung.

2) Es gibt keine richtige und falsche Version - mbmd scheint andere Daten zu sehen als evcc. Deswegen die Frage: _Protokolliert evcc im modbus Trace auch die Daten, die über den modbus Proxy gehen?_

MarkusGH, 2024-03-11:
Hier noch die Situation gerade eben:

[image]
[image]

Umgebungsbedingungen:

- Keine größeren taktenden Verbraucher
- Keine durch evcc regelbaren Verbraucher verfügbar
- Hausverbrauch ungefähr gleich Erzeugung (Batterie wechselt zwischen Laden/Lntladen)

mbmd (Saldiert Batterie + PV) sieht um 18:30 das:

[image]

evcc meldet:

- Gesamtverbrauch min -2426 W, max 4133 W
- Hausverbrauch max 5282 W
- Erzeugung min -1627, max 3111
- Netzbezug min -1807, max 1066
- Batterie min -682, max 4848 W

premultiply, 2024-03-11:
@andig Glaskugel: SunSpec SF ?

andig, 2024-03-11:
Was soll ich da ohne Modbus Log zu sagen???

MarkusGH, 2024-03-11:
@andig: Nachdem ich jetzt drei mal vergeblich gefragt habe, ob/wie die Requests, die über den Proxy gehen protokolliert werden, habe ich es selber aus den Quelltexten rausgefummelt. Danke für die Hilfe! Log kommt sobald es wieder auftritt und ich am Rechner sitze.

andig, 2024-03-11:
Eigentlich musst du das nur ausprobieren?

MarkusGH, 2024-03-11:
Meinst Du das:

```
[proxy-502] TRACE 2024/03/11 21:51:05 read holdings: id 1 addr 40071 qty 50
[proxy-502] TRACE 2024/03/11 21:51:05 read holding: 01e400a000a400a0fffe0fb80fc50fbb090e0918091bffff2b02ffff138afffe2c1effff9d90fffed9f1fffe01ef44640000394bfffc1dc4ffff2ba9ffff800013d280008000fffe00040000ffffffffffffffff00000000ffffffffffffffff00000000
[proxy-502] TRACE 2024/03/11 21:51:05 read holdings: id 1 addr 40190 qty 105
[proxy-502] TRACE 2024/03/11 21:51:05 read holding: 0035000f000d0018ffff5a855a855af85aff0000000000000000fffe1388fffe0005ffe50109ff18000002f10128011d01a80000fd0ffed9ff99fe9e0000fae6012bded710adfffe00f7cef40064139e006a0b84005657b6016fe348007bd80f0068a73800b80be600000000000000000000000000000000000000000000000000000000000000000000800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000800000000000

```
```
[modbus] TRACE 2024/03/11 21:50:54 modbus: send 01 3c 00 00 00 06 01 03 9c fe 00 69
[modbus] TRACE 2024/03/11 21:50:54 modbus: recv 01 3c 00 00 00 d5 01 03 d2 00 33 00 0f 00 0d 00 15 ff ff 5a 78 5a 78 5a f6 5b 05 00 00 00 00 00 00 00 00 ff fe 13 88 ff fe 00 0e ff dc 01 0e ff 25 00 00 02 e2 01 26 01 21 01 97 00 00 fd 1e fe dc ff 99 fe a9 00 00 fc 01 03 d3 de c2 11 6c ff fe 00 f7 ce f4 00 64 13 9e 00 6a 0b 83 00 56 57 b6 01 6f e3 48 00 7b d8 0f 00 68 a7 38 00 b8 0b e5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00
[modbus] TRACE 2024/03/11 21:50:55 modbus: send 01 3d 00 00 00 06 01 03 9c 87 00 32
[modbus] TRACE 2024/03/11 21:50:55 modbus: recv 01 3d 00 00 00 67 01 03 64 01 f1 00 a5 00 a7 00 a4 ff fe 0f b7 0f c5 0f b8 09 0f 09 19 09 1a ff ff 2c 42 ff ff 13 89 ff fe 2d 4c ff ff 9f 84 ff fe d9 da ff fe 01 ef 44 60 00 00 3a ec ff fc 1d c9 ff ff 2c ee ff ff 80 00 13 d6 80 00 80 00 ff fe 00 04 00 00 ff ff ff ff ff ff ff ff 00 00 00 00 ff ff ff ff ff ff ff ff 00 00 00 00

```

MarkusGH, 2024-03-11:
@andig: Habe die Logs ergänzt

andig, 2024-03-12:
> dann mal evcc meter mit trace log für richtige + falsche Version machen. 

Wo???

MarkusGH, 2024-03-12:
> > dann mal evcc meter mit trace log für richtige + falsche Version machen.
> 
> Wo???

Ich verstehe nicht ganz was Du meinst, aber versuche mal das:

```
> evcc meter
[main  ] INFO 2024/03/12 11:53:42 evcc 0.124.9 (090eddde)
[main  ] INFO 2024/03/12 11:53:42 using config file: /etc/evcc.yaml
[db    ] INFO 2024/03/12 11:53:42 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/03/12 11:53:42 connecting evcc-1293785003 at tcp://mqtt.xxx.de:1883
[mqtt  ] DEBUG 2024/03/12 11:53:42 tcp://mqtt.xxx.de:1883 connected
grid
----
Power: -19W

pv1
---
Power: 3517W

battery1
--------
Power:    -140W
Soc:      24%
Capacity: 11.7kWh

heatpump
--------
Power: 0W
```

andig, 2024-03-13:
Ich weiss nicht wie ich es noch schreiben soll. 

> dann mal evcc meter mit trace log für richtige + falsche Version machen.

Nichts davon sehe ich :(

closed 2024-03-13 by andig

MarkusGH, 2024-03-13:
Hatte vergessen modbus trace zu aktivieren - sorry.

```
[main  ] INFO 2024/03/13 11:48:42 evcc 0.124.9 (090eddde)
[main  ] INFO 2024/03/13 11:48:42 using config file: /etc/evcc.yaml
[db    ] INFO 2024/03/13 11:48:42 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/03/13 11:48:42 connecting evcc-1216477408 at tcp://mqtt.xxx.de:1883
[mqtt  ] DEBUG 2024/03/13 11:48:42 tcp://mqtt.xxx.de:1883 connected
grid
----
[modbus] TRACE 2024/03/13 11:48:43 modbus: send 00 2b 00 00 00 06 01 03 9c fe 00 69
[modbus] TRACE 2024/03/13 11:48:43 modbus: recv 00 2b 00 00 00 d5 01 03 d2 00 68 00 2a 00 26 00 17 ff ff 59 c1 59 c1 59 79 59 e3 00 00 00 00 00 00 00 00 ff fe 13 85 ff fe ff ec fc ca 03 71 ff b2 00 00 03 8a 03 be 03 77 01 3f 00 00 fc 76 fe 14 ff 98 fe cb 00 00 ff e6 20 30 d9 86 05 f9 ff fe 00 f7 d0 2f 00 64 1c 81 00 6a 24 d2 00 56 5b 18 01 70 90 04 00 7c 24 83 00 68 cd 7c 00 b8 6a 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00
Power: 20W

pv1
---
[modbus] TRACE 2024/03/13 11:48:43 modbus: send 00 2c 00 00 00 06 01 03 9c 87 00 32
[modbus] TRACE 2024/03/13 11:48:43 modbus: recv 00 2c 00 00 00 67 01 03 64 04 a3 01 8d 01 89 01 8c ff fe 0f 8e 0f 9b 0f 90 09 00 08 fc 08 fe ff ff 6a 9a ff ff 13 87 ff fe 6a fd ff ff a4 fc ff fe d9 15 ff fe 01 ef b2 38 00 00 0e 33 ff fd 1d c5 ff ff 6c 39 ff ff 80 00 13 3c 80 00 80 00 ff fe 00 04 00 00 ff ff ff ff ff ff ff ff 00 00 00 00 ff ff ff ff ff ff ff ff 00 00 00 00
[modbus] TRACE 2024/03/13 11:48:43 modbus: send 00 2d 00 00 00 06 01 03 f5 74 00 02
[modbus] TRACE 2024/03/13 11:48:43 modbus: recv 00 2d 00 00 00 07 01 03 04 00 00 42 40
Power: 2818W

battery1
--------
[modbus] TRACE 2024/03/13 11:48:43 modbus: send 00 2e 00 00 00 06 01 03 e1 74 00 02
[modbus] TRACE 2024/03/13 11:48:43 modbus: recv 00 2e 00 00 00 07 01 03 04 00 00 42 40
[modbus] TRACE 2024/03/13 11:48:43 modbus: send 00 2f 00 00 00 06 01 03 e1 84 00 02
[modbus] TRACE 2024/03/13 11:48:43 modbus: recv 00 2f 00 00 00 07 01 03 04 00 00 41 70
Power:    -48W
Soc:      15%
Capacity: 11.7kWh

heatpump
--------
Power: 0W
```

MarkusGH, 2024-03-13:
Da steht aber nichts anderes als in dem Log, das ich schon gesendet hatte.
Und was Du mit "richtige + falsche Version" meinst verstehe ich nicht.

andig, 2024-03-13:
Ich meine funktionierende und kaputte evcc Version zum Vergleich. Und es braucht auch die Info, welcher der Messwerte denn nun falsch ist. Die anderen kannst Du auch gerne raus löschen, dann ists eindeutig.

MarkusGH, 2024-03-13:
Andi - lies doch bitte mal was ich oben schreibe.

Es gibt keine "funktionierende " evcc Version, das vVerhalten ist in allen Versionen, die ich kenne gleich.

Es scheint schlicht, dass [mbmd ](https://github.com/volkszaehler/mbmd) (solltest Du kennen, Du hast daran mitgearbeitet) in bestimmten Betriebszuständen andere Werte als evcc sieht. mbmd greift über den Modbux Proxy zu, also sieht man im trace log auch, was mbmd abfragt und enpfängt.

andig, 2024-03-13:
> lies doch bitte mal was ich oben schreibe.

Das wäre wahrscheinlich einfacher, wenn ich nicht ständig angefragt Informationen suchen würde ;)

> zeigt evcc unplausible Werte für Netzeinspeisung/Netzbezug.

Also, es geht um `grid`? Kurzer Blick ins Template zeigt:

```
  {{- if eq .usage "grid" }}
  power:
    source: sunspec
    {{- include "modbus" . | indent 2 }}
    timeout: {{ .timeout }}
    subdevice: 1 # Metering device
    value: 203:W
    scale: -1
  {{- end }}
```

Einfacher gehts ja nicht. Sunspec Model 203 ist

    // Block203 - wye-connect three phase (abcn) meter -

Jetzt müsste mir jemand verraten, was daran falsch sein kann?

andig, 2024-03-13:
Mach mal bitte noch ein:

    evcc sunspec -id 1 solaredge.xxx.de:502

Und: ich bräuchte irgendwo mal ein trace log von einem *falschen* Wert den evcc anzeigt um der These auf den Grund zu gehen, dass es vielleicht an den Scale Factors liegen könnte.

MarkusGH, 2024-03-13:
-id nimmt er nicht:

```
evcc sunspec -id 1 solaredge.xxx.de:502
Error: invalid argument "d" for "-i, --id" flag: strconv.ParseInt: parsing "d": invalid syntax
Usage:
  evcc sunspec <connection> [flags]

Flags:
  -i, --id int   Slave id (default 1)

Global Flags:
  -c, --config string   Config file (default "~/evcc.yaml" or "/etc/evcc.yaml")
  -h, --help            Help
  -l, --log string      Log level (fatal, error, warn, info, debug, trace) (default "info")
      --log-headers     Log headers

invalid argument "d" for "-i, --id" flag: strconv.ParseInt: parsing "d": invalid syntax

```

-i geht:

```
# evcc sunspec -i 1 solaredge.xxx.de:502
[main  ] INFO 2024/03/13 12:24:05 evcc 0.124.9 (090eddde)
--------- Model 1 common ---------
Mn         SolarEdge     string
Md    SE10K-RWS48BNN4    string
Opt                      string
Vr           0004.001    string
SN           7E04494A    string
DA               1.00    uint16
Pad            0x0067    pad
--------- Model 103 inverter ---------
A                    9.75    uint16
AphA                 3.24    uint16
AphB                 3.25    uint16
AphC                 3.25    uint16
A_SF                   -2    sunssf
PPVphAB            398.50    uint16
PPVphBC            399.90    uint16
PPVphCA            400.40    uint16
PhVphA             230.00    uint16
PhVphB             229.00    uint16
PhVphC             232.90    uint16
V_SF                   -1    sunssf
W                 2244.00    int16
W_SF                   -1    sunssf
Hz                  50.01    uint16
Hz_SF                  -2    sunssf
VA                2254.90    int16
VA_SF                  -1    sunssf
VAr               -222.00    int16
VAr_SF                 -2    sunssf
PF                 -99.50    int16
PF_SF                  -2    sunssf
WH            32487436.00    acc32
WH_SF                   0    sunssf
DCA                  2.99    uint16
DCA_SF                 -4    sunssf
DCV                762.00    uint16
DCV_SF                 -1    sunssf
DCW               2278.10    int16
DCW_SF                 -1    sunssf
TmpCab                n/a    int16
TmpSnk              49.70    int16
TmpTrns               n/a    int16
TmpOt                 n/a    int16
Tmp_SF                 -2    sunssf
[... 69 lines cut ...]
TotVAhImpPhC                  NaN    acc32
TotVAh_SF                     n/a    sunssf
TotVArhImpQ1                  NaN    acc32
TotVArhImpQ1PhA               NaN    acc32
TotVArhImpQ1PhB               NaN    acc32
TotVArhImpQ1PhC               NaN    acc32
TotVArhImpQ2                  NaN    acc32
TotVArhImpQ2PhA               NaN    acc32
TotVArhImpQ2PhB               NaN    acc32
TotVArhImpQ2PhC               NaN    acc32
TotVArhExpQ3                  NaN    acc32
TotVArhExpQ3PhA               NaN    acc32
TotVArhExpQ3PhB               NaN    acc32
TotVArhExpQ3PhC               NaN    acc32
TotVArhExpQ4                  NaN    acc32
TotVArhExpQ4PhA               NaN    acc32
TotVArhExpQ4PhB               NaN    acc32
TotVArhExpQ4PhC               NaN    acc32
TotVArh_SF                    n/a    sunssf
Evt                    0x00000000    bitfield32

```

Trace mit falschem Wert kommt gelegentlich.
Wenn es an den Scale factors läge würden die Spikes 10x die höhe des tatsächlichen Werts haben?

andig, 2024-03-13:
Eigentlich schon. Mir fehlt hier leider jede Idee, v.a. da mbmd auch nichts anderes ausliest. Da hilft tatsächlich nur warten auf Logfile des echten Fehlers, sorry...

MarkusGH, 2024-03-13:
Alles gut für mich - ich hatte da keine schnelle Lösung erwartet, wollte das Verhalten aber dokumentieren.
Magst Du das issue bitte wieder aufmachen?
Ich sehe zu dass ich Diagnosedaten liefere sobald es geht, das Verhalten ist aber nur in bestimmten Betriebszuständen da und für den Abgleich mit mbmd muss ich am Computer sein.

andig, 2024-03-13:
Ich mach wieder auf wenns nachvollziehbar ist. Da die Issues unser Backlog sind möchte ich da nur offene Themen haben bei denen es eine Todo bei uns gibt.

MarkusGH, 2024-04-02:
```
Apr  2 18:57:29 homebridge evcc[12571]: [modbus] TRACE 2024/04/02 18:57:29 modbus: recv 17 00 00 00 00 67 01 03 64 09 42 03 0d 03 1d 03 17 ff fe 0f ae 0f be 0f b7 09 0b 09 0a 09 21 ff ff 15 90 00 00 13 84 ff fe 15 98 00 00 f3 b2 ff ff d9 01 ff fe 01 fc
6b 62 00 00 1c be ff fd 1d c0 ff ff 15 e4 00 00 80 00 14 4d 80 00 80 00 ff fe 00 04 00 00 ff ff ff ff ff ff ff ff 00 00 00 00 ff ff ff ff ff ff ff ff 00 00 00 00
Apr  2 18:57:29 homebridge evcc[12571]: [proxy-502] TRACE 2024/04/02 18:57:29 read holding: 0942030d031d0317fffe0fae0fbe0fb7090b090a0921ffff159000001384fffe15980000f3b2ffffd901fffe01fc6b6200001cbefffd1dc0ffff15e400008000144d80008000fffe00040000fffffffff
fffffff00000000ffffffffffffffff00000000
Apr  2 18:57:29 homebridge evcc[12571]: [proxy-502] TRACE 2024/04/02 18:57:29 read holdings: id 1 addr 40071 qty 50
Apr  2 18:57:29 homebridge evcc[12571]: [modbus] TRACE 2024/04/02 18:57:29 modbus: send 17 01 00 00 00 06 01 03 9c 87 00 32
Apr  2 18:57:29 homebridge evcc[12571]: [modbus] TRACE 2024/04/02 18:57:29 modbus: recv 17 01 00 00 00 67 01 03 64 09 42 03 0d 03 1d 03 17 ff fe 0f ae 0f be 0f b7 09 0b 09 0a 09 21 ff ff 15 90 00 00 13 84 ff fe 15 98 00 00 f3 b2 ff ff d9 01 ff fe 01 fc
6b 62 00 00 1c be ff fd 1d c0 ff ff 15 e4 00 00 80 00 14 4d 80 00 80 00 ff fe 00 04 00 00 ff ff ff ff ff ff ff ff 00 00 00 00 ff ff ff ff ff ff ff ff 00 00 00 00
Apr  2 18:57:29 homebridge evcc[12571]: [proxy-502] TRACE 2024/04/02 18:57:29 read holding: 0942030d031d0317fffe0fae0fbe0fb7090b090a0921ffff159000001384fffe15980000f3b2ffffd901fffe01fc6b6200001cbefffd1dc0ffff15e400008000144d80008000fffe00040000fffffffff
fffffff00000000ffffffffffffffff00000000
Apr  2 18:57:29 homebridge evcc[12571]: [proxy-502] TRACE 2024/04/02 18:57:29 read holdings: id 1 addr 40190 qty 105
Apr  2 18:57:29 homebridge evcc[12571]: [modbus] TRACE 2024/04/02 18:57:29 modbus: send 17 02 00 00 00 06 01 03 9c fe 00 69
Apr  2 18:57:29 homebridge evcc[12571]: [modbus] TRACE 2024/04/02 18:57:29 modbus: recv 17 02 00 00 00 d5 01 03 d2 00 2b 00 0f 00 0a 00 11 ff ff 5a 1e 5a 1e 5a 26 5a d4 00 00 00 00 00 00 00 00 ff fe 13 83 ff fe f9 37 fd a2 fe 7d fd 19 00 00 07 6f 02 ac
01 97 03 2e 00 00 fc f9 fe c3 ff 83 fe b4 00 00 fb 9a 02 bc e2 f0 0d 1f ff fe 00 fa 78 93 00 66 24 c7 00 6d 03 3c 00 57 46 96 01 73 4f 56 00 7d ca 9c 00 69 9e dd 00 bb db e5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0
0 00 00 00 00 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00
Apr  2 18:57:29 homebridge evcc[12571]: [proxy-502] TRACE 2024/04/02 18:57:29 read holding: 002b000f000a0011ffff5a1e5a1e5a265ad40000000000000000fffe1383fffef937fda2fe7dfd190000076f02ac0197032e0000fcf9fec3ff83feb40000fb9a02bce2f00d1ffffe00fa7893006624c70
06d033c0057469601734f56007dca9c00699edd00bbdbe500000000000000000000000000000000000000000000000000000000000000000000800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000800000
000000
Apr  2 18:57:29 homebridge evcc[12571]: [proxy-502] TRACE 2024/04/02 18:57:29 read holdings: id 1 addr 40190 qty 105
Apr  2 18:57:29 homebridge evcc[12571]: [modbus] TRACE 2024/04/02 18:57:29 modbus: send 17 03 00 00 00 06 01 03 9c fe 00 69
Apr  2 18:57:29 homebridge evcc[12571]: [modbus] TRACE 2024/04/02 18:57:29 modbus: recv 17 03 00 00 00 d5 01 03 d2 00 2b 00 0f 00 0a 00 11 ff ff 5a 1e 5a 1e 5a 26 5a d4 00 00 00 00 00 00 00 00 ff fe 13 83 ff fe f9 37 fd a2 fe 7d fd 19 00 00 07 6f 02 ac 01 97 03 2e 00 00 fc f9 fe c3 ff 83 fe b4 00 00 fb 9a 02 bc e2 f0 0d 1f ff fe 00 fa 78 93 00 66 24 c7 00 6d 03 3c 00 57 46 96 01 73 4f 56 00 7d ca 9c 00 69 9e dd 00 bb db e5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00
Apr  2 18:57:29 homebridge evcc[12571]: [proxy-502] TRACE 2024/04/02 18:57:29 read holding: 002b000f000a0011ffff5a1e5a1e5a265ad40000000000000000fffe1383fffef937fda2fe7dfd190000076f02ac0197032e0000fcf9fec3ff83feb40000fb9a02bce2f00d1ffffe00fa7893006624c7006d033c0057469601734f56007dca9c00699edd00bbdbe500000000000000000000000000000000000000000000000000000000000000000000800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000800000000000
Apr  2 18:57:29 homebridge evcc[12571]: [proxy-502] TRACE 2024/04/02 18:57:29 read holdings: id 1 addr 40190 qty 105
Apr  2 18:57:29 homebridge evcc[12571]: [modbus] TRACE 2024/04/02 18:57:29 modbus: send 17 04 00 00 00 06 01 03 9c fe 00 69
Apr  2 18:57:29 homebridge evcc[12571]: [modbus] TRACE 2024/04/02 18:57:29 modbus: recv 17 04 00 00 00 d5 01 03 d2 00 2b 00 0f 00 0a 00 11 ff ff 5a 1e 5a 1e 5a 26 5a d4 00 00 00 00 00 00 00 00 ff fe 13 83 ff fe f9 37 fd a2 fe 7d fd 19 00 00 07 6f 02 ac 01 97 03 2e 00 00 fc f9 fe c3 ff 83 fe b4 00 00 fb 9a 02 bc e2 f0 0d 1f ff fe 00 fa 78 93 00 66 24 c7 00 6d 03 3c 00 57 46 96 01 73 4f 56 00 7d ca 9c 00 69 9e dd 00 bb db e5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0 [... cut ...] 00 00 00 00 ff ff ff ff ff ff ff ff 00 00 00 00
Apr  2 18:57:30 homebridge evcc[12571]: [modbus] TRACE 2024/04/02 18:57:30 modbus: send 17 0c 00 00 00 06 01 03 f5 74 00 02
Apr  2 18:57:30 homebridge evcc[12571]: [modbus] TRACE 2024/04/02 18:57:30 modbus: recv 17 0c 00 00 00 07 01 03 04 a0 00 c5 80
Apr  2 18:57:30 homebridge evcc[12571]: [site  ] WARN 2024/04/02 18:57:30 pv 1 power: -3390W is negative - check configuration if sign is correct
```
@andig: Hier habe ich ein Trace eines "Ereignisses"