evcc-io/evcc #12710: Idee: Ladeplanung nach Kosten

umrath, 2024-03-06:
Bisher wird die Ladeplanung vor allem nach Zeit, Energiequelle, Verfügbarkeit und Bedarf gesteuert.

Eine möglicherweise interessante Alternative wäre eine (rein?) preisgesteuerte Lösung.

Kurze Erläuterung:
Jeder Strom hat einen Preis. Auch der vom Dach, denn in aller Regel bekomme ich Einspeisevergütung für den (oder eben nicht, wenn ich ihn nutze).
Auch der Strom aus dem Akku hat einen Preis, der über dem vom Dach liegt, denn er ist trägt Verluste mit sich, lässt den Akku altern etc.
Der Strom aus dem Netz kann ggfs. dynamisch sein.

Also: Alle Stromquellen haben einen individuellen Preis.

Ich könnte also festlegen, dass ich mein Auto bis 20 % zu jedem Preis laden will, da mir sonst durchs rumstehen ggfs. der Akku kaputt geht und ich auch die immer benötigte Mindestreichweite nicht gegeben ist, ich also nicht zu Arbeit komme oder im Notfall zum Arzt, etc.
Von 20-40 % möchte ich gern laden, aber nicht zu jedem Preis, sonder nur für maximal 15 ct/kWh.
Damit würde PV-Strom, ggfs. Batteriestrom - oder extrem günstiger Netzstrom in den Mix kommen.
von 40-80% möchte ich dann aber nur noch maximal 9 ct/kWh ausgeben, so dass ausschließlich mit PV-Strom geladen wird, da Batterie- und Netzstrom wohl niemals so billig werden, dass sie unter diese Grenze fallen.

Alle Werte sind eigentlich schon vorhanden. Lediglich für den Batteriestrom müsste man noch eine neue Kategorie erstellen.

Die Idee lässt sich sicher noch sehr viel weiter spinnen. Aber der Ansatz sollte klar sein.

andig, 2024-03-06:
> Bisher wird die Ladeplanung vor allem nach Zeit, Energiequelle, Verfügbarkeit und Bedarf gesteuert.

?

> Eine möglicherweise interessante Alternative wäre eine (rein?) preisgesteuerte Lösung.

Das ist schon umgesetzt- über die Tarife aber auch Smart Cost je Ladepunkt.

closed 2024-03-06 by andig

andig, 2024-03-06:
> Meines Wissens nach lässt sich dieses Szenario derzeit so nicht abbilden.

Das kannst Du über die APIs extern steuern. Für den Normalanwender nicht zu verstehen da viel zu kompliziert.