evcc-io/evcc #12713: Probleme mit 0.124.7

MMA150475, 2024-03-06:
### Describe the bug

Ich habe eben auf 0.124.7 upgedated, und jetzt hängt evcc. Nach dem downgrade zurück auf 0.124.6 gehts wieder.  Im Anhang mein Log.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: xxx 
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-ksem-inverter 
  id: 71  
  host:xxx
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid3

chargers:
- type: template
  template: go-e-v3 
  host: xxxx
  name: wallbox5

vehicles:
  - name: Mokka-e
    type: template
    template: tronity
    title: Papa's Flitzer Tronity
    clientid: xxxx
    clientsecret: xxxx
    capacity: 47.5 
#  - name: ev4
#    type: template
#    template: opel 
#    title: Papa's Flitzer  
#    user: xxxxxxxxxxx
#    password: xxxxxxxxxxx
#    capacity: 50  
#    icon: tractor

loadpoints:
- title: Schuppen
  vehicle: Mokka-e
  charger: wallbox5
  mode: pv
  phases: 0                 # 0 = Phasenautomatik
  enable:
    threshold: -300         # schaltet ein wenn 1 Minute lang stabil mind. 300W Export war
    delay: 1m
  disable:
    threshold: 1100          # schaltet aus wenn 10 Minuten lang mehr als 1100W aus dem Netz bezogen wurde
    delay: 10m
  guardduration: 15m        # wartet mindestens 15 Minuten zwischen zwei Ladevorgängen

site:
[... 122 lines cut ...]
Mar  6 14:40:04 raspberrypi systemd[1]: Stopping evcc...
Mar  6 14:40:04 raspberrypi evcc[10754]: [lp-1  ] DEBUG 2024/03/06 14:40:04 charge total import: 1849.886kWh
Mar  6 14:40:04 raspberrypi systemd[1]: evcc.service: Succeeded.
Mar  6 14:40:04 raspberrypi systemd[1]: Stopped evcc.
Mar  6 14:40:06 raspberrypi dhcpcd[452]: wlan0: Router Advertisement from fe80::1eed:6fff:fe85:8ded
Mar  6 14:40:14 raspberrypi systemd[1]: Reloading.
Mar  6 14:40:15 raspberrypi systemd[1]: /lib/systemd/system/xrdp.service:8: PIDFile= references path below legacy directory /var/run/, updating /var/run/xrdp/xrdp.pid → /run/xrdp/xrdp.pid; please update the unit file accordingly.
Mar  6 14:40:15 raspberrypi systemd[1]: /lib/systemd/system/xrdp-sesman.service:9: PIDFile= references path below legacy directory /var/run/, updating /var/run/xrdp/xrdp-sesman.pid → /run/xrdp/xrdp-sesman.pid; please update the unit file accordingly.
Mar  6 14:40:21 raspberrypi systemd[1]: Reloading.
Mar  6 14:40:22 raspberrypi systemd[1]: /lib/systemd/system/xrdp.service:8: PIDFile= references path below legacy directory /var/run/, updating /var/run/xrdp/xrdp.pid → /run/xrdp/xrdp.pid; please update the unit file accordingly.
Mar  6 14:40:22 raspberrypi systemd[1]: /lib/systemd/system/xrdp-sesman.service:9: PIDFile= references path below legacy directory /var/run/, updating /var/run/xrdp/xrdp-sesman.pid → /run/xrdp/xrdp-sesman.pid; please update the unit file accordingly.
Mar  6 14:40:23 raspberrypi systemd[1]: Started evcc.
Mar  6 14:40:24 raspberrypi evcc[10993]: [main  ] INFO 2024/03/06 14:40:24 evcc 0.124.7
Mar  6 14:40:24 raspberrypi evcc[10993]: [main  ] INFO 2024/03/06 14:40:24 using config file: /etc/evcc.yaml
Mar  6 14:40:24 raspberrypi evcc[10993]: [main  ] INFO 2024/03/06 14:40:24 starting ui and api at :7070
Mar  6 14:40:25 raspberrypi evcc[10993]: [db    ] INFO 2024/03/06 14:40:25 using sqlite database: /var/lib/evcc/evcc.db
```

### What type of operating system are you running?

Linux

### Version

0.124.7

Semmy100, 2024-03-06:
Kann ich bestätigen, neue Version läßt sich nicht aufrufen.

andig, 2024-03-06:
Fehlerbeschreibung fehlt.

MMA150475, 2024-03-06:
Fehlerbeschreibung: 

Die Oberfläche lässt sich nicht mehr aufrufen. EVCC belibt genau dort stehen wo das Logfile endet.

andig, 2024-03-06:
Mhhm. Dann bitte mal mit `log: trace` starten und zusätzlich `--profile` analog https://github.com/evcc-io/evcc/discussions/12700#discussioncomment-8686351 starten.

MMA150475, 2024-03-06:
Wenn ichs richtig verstanden habe muss ich im Browser (auf meinem Arbeits-PC) einfach /debug/pprof dranhängen? Das sieht dann so aus: 
http://192.168.178.105:7070/#/debug/pprof

Aber da kommt dann trotzdem nix - oder muss ich das auf dem pi machen? 

Hier das Log vom Start mit --profile

pi@raspberrypi:~ $ evcc --profile -c evcc.yaml
[main  ] INFO 2024/03/06 16:08:25 evcc 0.124.7
[main  ] INFO 2024/03/06 16:08:25 using config file: evcc.yaml
[main  ] INFO 2024/03/06 16:08:25 starting ui and api at :7070
[db    ] INFO 2024/03/06 16:08:26 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2024/03/06 16:08:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT * FROM `settings` 17 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[sunspec] TRACE 2024/03/06 16:08:26 modbus: send 00 01 00 00 00 06 47 03 9c 40 00 02
[sunspec] TRACE 2024/03/06 16:08:26 modbus: recv 00 01 00 00 00 07 47 03 04 53 75 6e 53
[sunspec] TRACE 2024/03/06 16:08:26 modbus: send 00 02 00 00 00 06 47 03 9c 42 00 02
[sunspec] TRACE 2024/03/06 16:08:26 modbus: recv 00 02 00 00 00 07 47 03 04 00 01 00 42
[sunspec] TRACE 2024/03/06 16:08:26 modbus: send 00 03 00 00 00 06 47 03 9c 86 00 02
[sunspec] TRACE 2024/03/06 16:08:26 modbus: recv 00 03 00 00 00 07 47 03 04 00 67 00 32
[sunspec] TRACE 2024/03/06 16:08:26 modbus: send 00 04 00 00 00 06 47 03 9c ba 00 02
[sunspec] TRACE 2024/03/06 16:08:26 modbus: recv 00 04 00 00 00 07 47 03 04 00 71 00 3c
[sunspec] TRACE 2024/03/06 16:08:26 modbus: send 00 05 00 00 00 06 47 03 9c f8 00 02
[sunspec] TRACE 2024/03/06 16:08:26 modbus: recv 00 05 00 00 00 07 47 03 04 00 78 00 1a
[sunspec] TRACE 2024/03/06 16:08:26 modbus: send 00 06 00 00 00 06 47 03 9d 14 00 02
[sunspec] TRACE 2024/03/06 16:08:26 modbus: recv 00 06 00 00 00 07 47 03 04 00 7b 00 18
[sunspec] TRACE 2024/03/06 16:08:26 modbus: send 00 07 00 00 00 06 47 03 9d 2e 00 02
[sunspec] TRACE 2024/03/06 16:08:26 modbus: recv 00 07 00 00 00 07 47 03 04 00 a0 00 44
[sunspec] TRACE 2024/03/06 16:08:26 modbus: send 00 08 00 00 00 06 47 03 9d 74 00 02
[sunspec] TRACE 2024/03/06 16:08:26 modbus: recv 00 08 00 00 00 07 47 03 04 03 22 00 3e
[sunspec] TRACE 2024/03/06 16:08:26 modbus: send 00 09 00 00 00 06 47 03 9d b4 00 02
[sunspec] TRACE 2024/03/06 16:08:26 modbus: recv 00 09 00 00 00 07 47 03 04 ff ff 00 00
[sunspec] TRACE 2024/03/06 16:08:26 modbus: send 00 0a 00 00 00 06 47 03 9c 44 00 42
[sunspec] TRACE 2024/03/06 16:08:26 modbus: recv 00 0a 00 00 00 87 47 03 84 4b 4f 53 54 41 4c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 50 49 4b 4f 20 49 51 35 2e 35 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 30 31 2e 32 37 2e 30 39 39 33 32 00 00 00 00 00 39 32 30 39 32 52 48 44 30 30 30 30 45 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 47 5a 5a
[sunspec] TRACE 2024/03/06 16:08:26 modbus: send 00 0b 00 00 00 06 47 03 9d 30 00 08
[... cut ...]
[sunspec] TRACE 2024/03/06 16:08:26 modbus: recv 00 24 00 00 00 2b 47 03 28 00 03 44 43 5f 53 54 52 49 4e 47 5f 33 20 20 20 20 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[db    ] TRACE 2024/03/06 16:08:26 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[db    ] TRACE 2024/03/06 16:08:26 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[go-e  ] TRACE 2024/03/06 16:08:26 GET http://192.168.178.116/api/status?filter=alw
[go-e  ] TRACE 2024/03/06 16:08:26 {"alw":true}
[db    ] TRACE 2024/03/06 16:08:26 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>

MMA150475, 2024-03-06:
Sorry, da kam noch was:

Mar  6 16:25:13 raspberrypi systemd[1]: Started evcc.
Mar  6 16:25:14 raspberrypi evcc[14431]: [main  ] INFO 2024/03/06 16:25:14 evcc 0.124.7
Mar  6 16:25:14 raspberrypi evcc[14431]: [main  ] INFO 2024/03/06 16:25:14 using config file: /etc/evcc.yaml
Mar  6 16:25:14 raspberrypi evcc[14431]: [main  ] INFO 2024/03/06 16:25:14 starting ui and api at :7070
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] INFO 2024/03/06 16:25:15 using sqlite database: /var/lib/evcc/evcc.db
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT * FROM `settings` LIMIT 1 -1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT * FROM `settings` 25 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT * FROM `configs` LIMIT 1 -1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT #011fk_configs_details#011%") -1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT #011fk_devices_details#011%") -1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [db    ] TRACE 2024/03/06 16:25:15 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%#011device_id#011%") -1 <nil>
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: send 00 01 00 00 00 06 47 03 9c 40 00 02
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: recv 00 01 00 00 00 07 47 03 04 53 75 6e 53
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: send 00 02 00 00 00 06 47 03 9c 42 00 02
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: recv 00 02 00 00 00 07 47 03 04 00 01 00 42
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: send 00 03 00 00 00 06 47 03 9c 86 00 02
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: recv 00 03 00 00 00 07 47 03 04 00 67 00 32
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: send 00 04 00 00 00 06 47 03 9c ba 00 02
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: recv 00 04 00 00 00 07 47 03 04 00 71 00 3c
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: send 00 05 00 00 00 06 47 03 9c f8 00 02
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: recv 00 05 00 00 00 07 47 03 04 00 78 00 1a
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: send 00 06 00 00 00 06 47 03 9d 14 00 02
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: recv 00 06 00 00 00 07 47 03 04 00 7b 00 18
Mar  6 16:25:15 raspberrypi evcc[14431]: [sunspec] TRACE 2024/03/06 16:25:15 modbus: send 00 07 00 00 00 06 47 03 9d 2e 00 02
[... cut ...]
Mar  6 16:25:54 raspberrypi kernel: [37533.081152] Under-voltage detected! (0x00050005)
Mar  6 16:25:58 raspberrypi kernel: [37537.241202] Voltage normalised (0x00000000)
Mar  6 16:26:05 raspberrypi kernel: [37543.481322] Under-voltage detected! (0x00050005)
Mar  6 16:26:15 raspberrypi evcc[14431]: [modbus] TRACE 2024/03/06 16:26:15 modbus: closing connection due to idle timeout: 1m0.017713091s
Mar  6 16:26:17 raspberrypi kernel: [37555.961383] Voltage normalised (0x00000000)
Mar  6 16:26:23 raspberrypi kernel: [37562.201446] Under-voltage detected! (0x00050005)

stan23, 2024-03-06:
Hier das gleiche Problem: der Webserver scheint nicht mehr zu laufen.
Bei mir ist es ein LXC auf Proxmox.

**0.124.7**
```
marco@evccCT:~$ evcc --profile --config /etc/evcc.yaml
[main  ] INFO 2024/03/06 16:34:56 evcc 0.124.7
[main  ] INFO 2024/03/06 16:34:56 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/06 16:34:56 starting ui and api at :7070
[db    ] INFO 2024/03/06 16:34:56 using sqlite database: /home/marco/.evcc/evcc.db
[db    ] TRACE 2024/03/06 16:34:56 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/03/06 16:34:56 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/06 16:34:56 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/06 16:34:56 SELECT * FROM `settings` 8 <nil>
[mqtt  ] INFO 2024/03/06 16:34:56 connecting evcc-519037875 at tcp://ioBrokerVM.f17a:1883
[db    ] TRACE 2024/03/06 16:34:56 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/03/06 16:34:56 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/03/06 16:34:57 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/03/06 16:34:57 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/06 16:34:57 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/06 16:34:57 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/03/06 16:34:57 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/03/06 16:34:57 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/06 16:34:57 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/03/06 16:34:57 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/03/06 16:34:57 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/03/06 16:34:57 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "device_id	%") -1 <nil>
[mqtt  ] DEBUG 2024/03/06 16:34:57 tcp://ioBrokerVM.f17a:1883 connected
[db    ] TRACE 2024/03/06 16:34:57 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[db    ] TRACE 2024/03/06 16:34:57 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[go-e  ] TRACE 2024/03/06 16:34:57 GET http://go-eCharger_112810.f17a/api/status?filter=alw
[go-e  ] TRACE 2024/03/06 16:34:57 {"alw":false}
[db    ] TRACE 2024/03/06 16:34:57 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[mqtt  ] TRACE 2024/03/06 16:34:57 recv meter/energy/gridPowerCombined: '-185'
[mqtt  ] TRACE 2024/03/06 16:34:57 recv meter/energy/gridPowerCombined: '-198'
[mqtt  ] TRACE 2024/03/06 16:35:09 recv meter/energy/gridPowerCombined: '-202'
[mqtt  ] TRACE 2024/03/06 16:35:21 recv meter/energy/gridPowerCombined: '-193'
[mqtt  ] TRACE 2024/03/06 16:35:33 recv meter/energy/gridPowerCombined: '-191'
[mqtt  ] TRACE 2024/03/06 16:35:45 recv meter/energy/gridPowerCombined: '-189'
```
Danach kommt im Log alle 10 bis 15 Sekunden ein neuer Wert, sonst nichts mehr.

Bin schon 3 Mal zwischen 0.124.6 und.7 hin- und her und es ist reproduzierbar.

In **0.124.6** sieht das Logs anfangs gleich aus, und dann geht es mit Tronity weiter:
```
marco@evccCT:~$ evcc --profile --config /etc/evcc.yaml
[main  ] INFO 2024/03/06 16:48:26 evcc 0.124.6
[main  ] INFO 2024/03/06 16:48:26 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/06 16:48:26 starting ui and api at :7070
[db    ] INFO 2024/03/06 16:48:27 using sqlite database: /home/marco/.evcc/evcc.db
[db    ] TRACE 2024/03/06 16:48:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/03/06 16:48:27 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/06 16:48:27 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/06 16:48:27 SELECT * FROM `settings` 8 <nil>
[mqtt  ] INFO 2024/03/06 16:48:27 connecting evcc-1058233568 at tcp://ioBrokerVM.f17a:1883
[db    ] TRACE 2024/03/06 16:48:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[... cut ...]
**0.124.7**
```
marco@evccCT:~$ sudo ss -tulpn | grep LISTEN
tcp   LISTEN 0      128          0.0.0.0:22        0.0.0.0:*    users:(("sshd",pid=132,fd=3))   
tcp   LISTEN 0      100        127.0.0.1:25        0.0.0.0:*    users:(("master",pid=282,fd=13))
tcp   LISTEN 0      128             [::]:22           [::]:*    users:(("sshd",pid=132,fd=4))   
tcp   LISTEN 0      100            [::1]:25           [::]:*    users:(("master",pid=282,fd=14))
marco@evccCT:~$ sudo lsof -i:7070
marco@evccCT:~$ 
```

stan23, 2024-03-06:
@Semmy100 & @MMA150475:
habt ihr beide Tronity in Benutzung?

Semmy100, 2024-03-06:
Ja, ich benutze Tronity.

MMA150475, 2024-03-06:
Ja, ich benutze Tronity seit Stelantis so rumzickt.

stan23, 2024-03-06:
Gut, dann haben wir jetzt 3 Installationen bei denen die 0.124.7 mit Tronity nicht läuft.

MMA150475, 2024-03-06:
Ja, wenn man Tronity ausklammert gehts auf einmal wieder. Dann erscheint das Fahrzeug wieder als Gastfahrzeug.

andig, 2024-03-06:
@MMA150475 

> Mar 6 16:25:54 raspberrypi kernel: [37533.081152] Under-voltage detected! (0x00050005)

Dein Raspi hat erstmal ein Problem mit der Stromversorgung. Da kann alles passieren, hier OT.

Tronity schaue ich mir an.

andig, 2024-03-06:
Geht

    evcc vehicle

bei Euch? Wann genau hängt sich das auf? Es braucht bitte ein --profile/ /debug/pprof, siehe Anfrage oben.

MMA150475, 2024-03-07:
@andig
 
OT - ja, ist mir bekannt - mir ist das Netzteil abgeraucht, und ich kam noch nicht dazu ein neues einzubauen. Läuft trotzdem erstaunlich stabil über die USB Versorgung, bisher keine Neustarts oder sonstige Effekte gesehen die darauf zurückzuführen wären. 
Wird demnächst sowieso durch einen neueren leistungsfähigeren Raspi ersetzt.

andig, 2024-03-07:
Bei Tronity mussten wir etwas ändern, es ist allerdings nicht zu erkennen, wie/warum das einen negativen Effekt haben sollte. Mir fehlt die Möglichkeit, das zu reproduzieren. Der Ball liegt hei euch :/

MMA150475, 2024-03-07:
Wenn du mir genau sagst wie ich das mit dem /debug/pprof machen muss dann kann ich das heute abend mal machen. Gestern hab ichs nicht hinbekommen, trotz der Anleitung in dem anderen Thread.
Ich bediene den Raspi normalerweise remote, und rufe die Oberfläche von meinem Arbeitsplatz-PC aus auf. Ich kann auch direkt drauf gehen wenn das erforderlich wäre.

stan23, 2024-03-07:
> Geht
> 
> ```
> evcc vehicle
> ```
> 
> bei Euch? Wann genau hängt sich das auf? 

Nein geht nicht.
Es hängt genau wie das log vor der ersten Zeile mit Tronity.

**0.124.6**
```
marco@evccCT:~$ evcc vehicle
[main  ] INFO 2024/03/07 11:44:03 evcc 0.124.6
[main  ] INFO 2024/03/07 11:44:03 using config file: /etc/evcc.yaml
[db    ] INFO 2024/03/07 11:44:03 using sqlite database: /home/marco/.evcc/evcc.db
[db    ] TRACE 2024/03/07 11:44:03 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT * FROM `settings` 8 <nil>
[mqtt  ] INFO 2024/03/07 11:44:03 connecting evcc-1626445299 at tcp://ioBrokerVM.f17a:1883
[db    ] TRACE 2024/03/07 11:44:03 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/03/07 11:44:03 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "device_id	%") -1 <nil>
[tronity] TRACE 2024/03/07 11:44:03 POST https://api.tronity.tech/authentication
[mqtt  ] DEBUG 2024/03/07 11:44:03 tcp://ioBrokerVM.f17a:1883 connected
[db    ] TRACE 2024/03/07 11:44:03 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[tronity] TRACE 2024/03/07 11:44:03 {"client_id":"***","client_secret":"***","grant_type":"app"}
--
{"id":"7716e61d-d343-4d69-a7b6-32cfdaac7be4","access_token":"eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJzdWIiOiI3NzE2ZTYxZC1kMzQzLTRkNjktYTdiNi0zMmNmZGFhYzdiZTQiLCJhcHBJZCI6IjQ1MWY5ODViLTBiYmUtNDc5Yi04MjA3LWMyYTI5MzA0YTM3NCIsInZlaGljbGVJZCI6bnVsbCwic2NvcGVzIjpbXSwicm9sZSI6InVzZXIiLCJpYXQiOjE3MDk4MDgyNDMsImV4cCI6MTcwOTg5NDY0MywiYXVkIjoiaHR0cHM6Ly9wbGF0Zm9ybS50cm9uaXR5LmlvIiwiaXNzIjoicGxhdGZvcm0udHJvbml0eS5pbyJ9.Qbz7PLs9iwv8HK-7UpXAQwCSqD4a7dIZjy6d7C7D6Eg","expires_in":3600,"token_type":"bearer"}
[tronity] TRACE 2024/03/07 11:44:03 GET https://api.tronity.tech/tronity/vehicles
[tronity] TRACE 2024/03/07 11:44:03 {"data":[{"id":"63aecd4ccf3be20019b02a7c","scopes":["read_charge","read_battery","read_vin","read_vehicle_info","tronity_charges","tronity_charging","tronity_control_charging","tronity_factsheet","tronity_location","tronity_odometer","tronity_range","tronity_soc","tronity_vehicle_data"],"createdAt":"2022-12-30T11:36:44.402Z","updatedAt":"2022-12-30T11:36:44.402Z","vin":"VR3UHZKXZNT109814","displayName":"e-208","manufacture":"Peugeot","model":"e-208","year":2022,"kWh":45,"liter":null,"offline":false,"valid":true}],"count":1,"total":1,"page":1,"pageCount":1}
[tronity] TRACE 2024/03/07 11:44:03 GET https://api.tronity.tech/tronity/vehicles/63aecd4ccf3be20019b02a7c/last_record
[tronity] TRACE 2024/03/07 11:44:03 {"odometer":24506,"range":172,"level":63,"charging":"Charging","plugged":true,"chargerPower":11,"chargeRemainingTime":7800,"latitude":49.02885,"longitude":11.47115,"status":null,"timestamp":1709805885000,"lastUpdate":1709805885000}
Soc:           63%
Capacity:      50.0kWh
Charge status: C
Range:         172km
Odometer:      24506km
Features:      []
marco@evccCT:~$ 
```

**0.124.7**
```
marco@evccCT:~$ evcc vehicle
[main  ] INFO 2024/03/07 11:45:50 evcc 0.124.7
[main  ] INFO 2024/03/07 11:45:50 using config file: /etc/evcc.yaml
[db    ] INFO 2024/03/07 11:45:50 using sqlite database: /home/marco/.evcc/evcc.db
[... cut ...]
[db    ] TRACE 2024/03/07 11:45:50 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "device_id	%") -1 <nil>
[mqtt  ] DEBUG 2024/03/07 11:45:50 tcp://ioBrokerVM.f17a:1883 connected
[db    ] TRACE 2024/03/07 11:45:50 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
```
und da bleibt die Ausgabe stehen.

> Es braucht bitte ein --profile/ /debug/pprof, siehe Anfrage oben.

Wie kann ich das aufzeichnen wenn der Webserver nicht läuft?

Würden dir meine Tronity-Zugangsdaten helfen?

andig, 2024-03-07:
> Würden dir meine Tronity-Zugangsdaten helfen?

Schick gerne deine Config an info@evcc.io

andig, 2024-03-08:
Fixed in 5f89c91b90d39cd7ee69cb92a828bd5b64d0dfed

closed 2024-03-08 by andig

Semmy100, 2024-03-08:
Und wie installiert man dann das gefixde?

stan23, 2024-03-09:
> Und wie installiert man dann das gefixde?

Hier unter nightly:
https://docs.evcc.io/docs/installation/linux/

Man kann die (passenden) Sourcen auch händisch eintragen:

```
marco@evccCT:~$ cat /etc/apt/sources.list.d/evcc-unstable.list 
deb [signed-by=/usr/share/keyrings/evcc-unstable-archive-keyring.gpg] https://dl.cloudsmith.io/public/evcc/unstable/deb/debian bullseye main
deb-src [signed-by=/usr/share/keyrings/evcc-unstable-archive-keyring.gpg] https://dl.cloudsmith.io/public/evcc/unstable/deb/debian bullseye main
```

Zurück zu stable kommt man indem man diese unstable.list löscht und ein `apt update; apt upgrade` durchführt.

Die 0.124.8+1709972561 läuft bei mir wieder, sehr schön :)