evcc-io/evcc #12716: [influx] write failed (attempts 0): 400 Bad Request: field type conflict

mdkeil, 2024-03-06:
### Describe the bug

Bei fast jedem Neustart von EVCC habe ich einmalig einen Fehler im Log seitens Influx, welcher die Funktion aber nicht zu beeinträchtigen scheint.

Keine Ahnung, seit wann der Fehler auftritt, geht aber schon einige Versionen zurück; denke aber <= 0.123.x

**Info:** influxdb 1.8 in einem separaten Docker-Container

### Steps to reproduce

evcc Neustart

### Configuration details

```yaml
nicht erforderlich
```

### Log details

```text
[main  ] INFO 2024/03/06 15:07:16 evcc 0.124.7
[main  ] INFO 2024/03/06 15:07:16 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/06 15:07:16 starting ui and api at :7070
[db    ] INFO 2024/03/06 15:07:16 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/03/06 15:07:16 connecting evcc-14306230 at tcp://192.168.1.2:1883
[lp-1  ] WARN 2024/03/06 15:07:16 poll mode '{connected 10m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] WARN 2024/03/06 15:07:18 vehicle 'evDummy' does not support automatic detection
[site  ] INFO 2024/03/06 15:07:18 site config:
[site  ] INFO 2024/03/06 15:07:18   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/06 15:07:18     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/06 15:07:18     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/06 15:07:18     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/03/06 15:07:18   vehicles:
[site  ] INFO 2024/03/06 15:07:18     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/03/06 15:07:18     vehicle 2: range ✓ finish ✗ status ✓ climate ✓ wakeup ✗
[site  ] INFO 2024/03/06 15:07:18     vehicle 3: range ✓ finish ✗ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/03/06 15:07:18 loadpoint 1:
[lp-1  ] INFO 2024/03/06 15:07:18   mode:        off
[lp-1  ] INFO 2024/03/06 15:07:18   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/03/06 15:07:18   meters:      charge ✓
[lp-1  ] INFO 2024/03/06 15:07:18     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2024/03/06 15:07:18 loadpoint 2:
[lp-2  ] INFO 2024/03/06 15:07:18   mode:        off
[lp-2  ] INFO 2024/03/06 15:07:18   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2024/03/06 15:07:18   meters:      charge ✓
[lp-2  ] INFO 2024/03/06 15:07:18     charge:    power ✓ energy ✓ currents ✗
[influx] TRACE 2024/03/06 15:07:18 write bufferSoc=map[value:90] (map[])
[influx] TRACE 2024/03/06 15:07:18 write bufferStartSoc=map[value:100] (map[])
[influx] TRACE 2024/03/06 15:07:18 write prioritySoc=map[value:80] (map[])
[influx] TRACE 2024/03/06 15:07:18 write bufferSoc=map[value:90] (map[])
[influx] TRACE 2024/03/06 15:07:18 write bufferStartSoc=map[value:100] (map[])
[influx] TRACE 2024/03/06 15:07:18 write prioritySoc=map[value:80] (map[])
[influx] TRACE 2024/03/06 15:07:18 write residualPower=map[value:100] (map[])
[influx] TRACE 2024/03/06 15:07:18 write phasesConfigured=map[value:3] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write minCurrent=map[value:6] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write maxCurrent=map[value:16] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write planEnergy=map[value:0] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write priority=map[value:0] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write minCurrent=map[value:6] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write maxCurrent=map[value:16] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write enableThreshold=map[value:0] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write disableThreshold=map[value:0] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write phasesConfigured=map[value:3] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write phasesEnabled=map[value:3] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write phasesActive=map[value:3] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write chargerPhysicalPhases=map[value:<nil>] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write chargerIcon=map[value:<nil>] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write vehicleOdometer=map[value:0] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write planEnergy=map[value:0] (map[loadpoint:Carport])
[... 55 lines cut ...]
[influx] TRACE 2024/03/06 15:07:18 write vehicleSoc=map[value:0] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write vehicleRange=map[value:0] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write vehicleTargetSoc=map[value:0] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write effectivePriority=map[value:0] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write effectivePlanSoc=map[value:0] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write effectiveMinCurrent=map[value:6] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write effectiveMaxCurrent=map[value:16] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write effectiveLimitSoc=map[value:100] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write chargeCurrent=map[value:0] (map[loadpoint:Carport])
[influx] TRACE 2024/03/06 15:07:18 write greenShareHome=map[value:1] (map[])
[influx] TRACE 2024/03/06 15:07:18 write greenShareLoadpoints=map[value:1] (map[])
[influx] TRACE 2024/03/06 15:07:18 write tariffGrid=map[value:0.2373] (map[])
[influx] TRACE 2024/03/06 15:07:18 write tariffFeedIn=map[value:0.0736] (map[])
[influx] TRACE 2024/03/06 15:07:18 write tariffPriceHome=map[value:0.0736] (map[])
[influx] TRACE 2024/03/06 15:07:18 write tariffPriceLoadpoints=map[value:0.0736] (map[])
[influx] ERROR 2024/03/06 15:07:19 write failed (attempts 0): 400 Bad Request: field type conflict
```

### What type of operating system are you running?

Docker container

### Version

v0.124.7

andig, 2024-03-06:
Leider können wir da nix machen da uns Influx nicht verrät, welches Feld betroffen ist- wirklich blöd: https://github.com/influxdata/influxdb/issues/24568. Daher schweren Herzens wontfix. DB löschen löst das Problem :/

andig, 2024-03-06:
PS.: Du könntest im Code von async auf sync API umstellen. Dann sollte die Fehlermeldung sofort kommen und es gäbe die Chance, das betroffene Feld zu finden. Aber auch das löst das Problem nicht da die Datentypänderung im Zweifel immer noch vorhanden ist...

closed 2024-03-06 by andig

mdkeil, 2024-03-06:
> Du könntest im Code von async auf sync API umstellen.

Wo und Wie genau? -- Wo ich helfen kann, das mache ich natürlich ;)

andig, 2024-03-06:
Coding- da musst Du bitte in die influxdb.go rein schauen...