evcc-io/evcc #12723: UI Fehler Code 404

123aerox, 2024-03-06:
### Describe the bug

Während ich die "Device Configuration" Seite offen habe fällt im Hintergrund folgender Fehler im 10sec Takt an. 
Request failed with status code 404.
GET http://192.168.1.21/api//config/devices/meter/elgris meter/status

### Steps to reproduce

1.evcc öffnen 
2.Seite "Device Configuration" öffnen
3.zurück auf die Hauptseite und Fehler entdecken

### Configuration details

```yaml
network:
  schema: https
  host: 192.168.1.21
  port: 80

interval: 10s # control cycle interval

sponsortoken: *****

telemetry: true

plant: *****

log: debug
 
meters:
- name: elgris meter
  type: template
  usage: grid
  template: sma-energy-meter
  host: 192.168.1.45

- name: SMA1.5
  type: template
  template: sma-inverter
  usage: pv
  host: 192.168.1.56
  password: *****

chargers:
- name: Wallbe Eco S
  type: template
  template: phoenix-ev-eth
  modbus: tcpip
  id: 255
  host: 192.168.1.54 # Hostname
  port: 502 # Port

vehicles:
- name: Spring
  type: template
  template: dacia
  title: Dacia Spring 
  user: *****
  password: *****
  vin: *****
  cache: 5m 
  capacity: 27.4
  phases: 1
  cloud: false
  mode: now

loadpoints:
- title: Wallbox Garage
  charger: Wallbe Eco S
  vehicle: Spring
  soc:
    poll:
      mode: connected
      interval: 5m
    estimate: true

site:
  title: Mein Zuhause
  meters:
    grid: elgris meter
    pv: 
     - SMA1.5

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: fixed
    price: 0.3267 # EUR/kWh
  feedin:
    type: fixed
    price: 0.3267 # [currency]/kWh
  co2:
    type: grünstromindex
    zip: 86502
```

### Log details

```text
[main  ] INFO 2024/03/06 20:56:04 evcc 0.124.7
[main  ] INFO 2024/03/06 20:56:04 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/06 20:56:04 starting ui and api at :80
[db    ] INFO 2024/03/06 20:56:04 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] WARN 2024/03/06 20:56:07 poll mode '{connected 5m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] DEBUG 2024/03/06 20:56:08 charge total import: 4436.000kWh
[site  ] INFO 2024/03/06 20:56:08 site config:
[site  ] INFO 2024/03/06 20:56:08   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/03/06 20:56:08     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/06 20:56:08     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/06 20:56:08   vehicles:
[site  ] INFO 2024/03/06 20:56:08     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/03/06 20:56:08 loadpoint 1:
[lp-1  ] INFO 2024/03/06 20:56:08   mode:        off
[lp-1  ] INFO 2024/03/06 20:56:08   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/03/06 20:56:08   meters:      charge ✓
[lp-1  ] INFO 2024/03/06 20:56:08     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/03/06 20:56:08 phase timer inactive
[lp-1  ] DEBUG 2024/03/06 20:56:08 pv timer inactive
[lp-1  ] INFO 2024/03/06 20:56:08 vehicle updated: unknown -> Dacia Spring
[lp-1  ] DEBUG 2024/03/06 20:56:08 set charge mode: now
[site  ] WARN 2024/03/06 20:56:08 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/03/06 20:56:08 ----
[lp-1  ] DEBUG 2024/03/06 20:56:08 charge power: 0W
[site  ] DEBUG 2024/03/06 20:56:08 pv power: 0W
[site  ] DEBUG 2024/03/06 20:56:08 grid meter: 272W
[site  ] DEBUG 2024/03/06 20:56:08 grid powers: [22 235 14]W
[site  ] DEBUG 2024/03/06 20:56:08 grid currents: [0.452 1.29 0]A
[site  ] DEBUG 2024/03/06 20:56:08 site power: 272W
[lp-1  ] DEBUG 2024/03/06 20:56:08 vehicle odometer: 28969km
[lp-1  ] DEBUG 2024/03/06 20:56:08 charge voltages: [228 228 228]V
[lp-1  ] DEBUG 2024/03/06 20:56:08 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/06 20:56:08 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/06 20:56:08 charge total import: 4436.000kWh
[lp-1  ] DEBUG 2024/03/06 20:56:08 charger status: B
[lp-1  ] DEBUG 2024/03/06 20:56:08 charge start energy: 4.44e+03kWh
[lp-1  ] INFO 2024/03/06 20:56:08 car connected
[lp-1  ] DEBUG 2024/03/06 20:56:08 pv timer elapse
[lp-1  ] DEBUG 2024/03/06 20:56:08 pv timer inactive
[lp-1  ] DEBUG 2024/03/06 20:56:08 charge total import: 4436.000kWh
[lp-1  ] DEBUG 2024/03/06 20:56:09 vehicle status: B
[lp-1  ] DEBUG 2024/03/06 20:56:09 vehicle soc: 90%
[lp-1  ] DEBUG 2024/03/06 20:56:09 vehicle range: 145km
[lp-1  ] DEBUG 2024/03/06 20:56:09 limitSoc reached: 90.0% > 90%
[site  ] DEBUG 2024/03/06 20:56:18 ----
[lp-1  ] DEBUG 2024/03/06 20:56:18 charge power: 0W
[site  ] DEBUG 2024/03/06 20:56:18 pv power: 0W
[site  ] DEBUG 2024/03/06 20:56:18 grid meter: 253W
[site  ] DEBUG 2024/03/06 20:56:18 grid powers: [21 217 14]W
[site  ] DEBUG 2024/03/06 20:56:18 grid currents: [0.449 1.22 0]A
[site  ] DEBUG 2024/03/06 20:56:18 site power: 253W
[lp-1  ] DEBUG 2024/03/06 20:56:18 vehicle odometer: 28969km
[lp-1  ] DEBUG 2024/03/06 20:56:18 charge voltages: [228 228 228]V
[lp-1  ] DEBUG 2024/03/06 20:56:18 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/06 20:56:18 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/06 20:56:18 charge total import: 4436.000kWh
[lp-1  ] DEBUG 2024/03/06 20:56:18 charger status: B
[lp-1  ] DEBUG 2024/03/06 20:56:18 next soc poll remaining time: 4m50s
[lp-1  ] DEBUG 2024/03/06 20:56:18 limitSoc reached: 90.0% > 90%
[site  ] DEBUG 2024/03/06 20:56:28 ----
[lp-1  ] DEBUG 2024/03/06 20:56:28 charge power: 0W
[site  ] DEBUG 2024/03/06 20:56:28 pv power: 0W
[site  ] DEBUG 2024/03/06 20:56:28 grid meter: 258W
[site  ] DEBUG 2024/03/06 20:56:28 grid powers: [21 221 14]W
[site  ] DEBUG 2024/03/06 20:56:28 grid currents: [0.449 1.24 0]A
[site  ] DEBUG 2024/03/06 20:56:28 site power: 258W
[lp-1  ] DEBUG 2024/03/06 20:56:28 charge voltages: [228 228 228]V
[lp-1  ] DEBUG 2024/03/06 20:56:28 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/06 20:56:28 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/06 20:56:28 charge total import: 4436.000kWh
[lp-1  ] DEBUG 2024/03/06 20:56:28 charger status: B
[lp-1  ] DEBUG 2024/03/06 20:56:28 next soc poll remaining time: 4m40s
[lp-1  ] DEBUG 2024/03/06 20:56:28 limitSoc reached: 90.0% > 90%
[site  ] DEBUG 2024/03/06 20:56:38 ----
[lp-1  ] DEBUG 2024/03/06 20:56:38 charge power: 0W
[site  ] DEBUG 2024/03/06 20:56:38 pv power: 0W
[site  ] DEBUG 2024/03/06 20:56:38 grid meter: 256W
[site  ] DEBUG 2024/03/06 20:56:38 grid powers: [22 219 14]W
[site  ] DEBUG 2024/03/06 20:56:38 grid currents: [0.452 1.24 0]A
[site  ] DEBUG 2024/03/06 20:56:38 site power: 256W
[lp-1  ] DEBUG 2024/03/06 20:56:38 charge voltages: [228 228 228]V
[lp-1  ] DEBUG 2024/03/06 20:56:38 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/06 20:56:38 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/06 20:56:38 charge total import: 4436.000kWh
[lp-1  ] DEBUG 2024/03/06 20:56:38 charger status: B
[lp-1  ] DEBUG 2024/03/06 20:56:38 next soc poll remaining time: 4m30s
[lp-1  ] DEBUG 2024/03/06 20:56:38 limitSoc reached: 90.0% > 90%
[site  ] DEBUG 2024/03/06 20:56:48 ----
[lp-1  ] DEBUG 2024/03/06 20:56:48 charge power: 0W
[site  ] DEBUG 2024/03/06 20:56:48 pv power: 0W
[site  ] DEBUG 2024/03/06 20:56:48 grid meter: 254W
[site  ] DEBUG 2024/03/06 20:56:48 grid powers: [21 219 14]W
[site  ] DEBUG 2024/03/06 20:56:48 grid currents: [0.449 1.22 0]A
[site  ] DEBUG 2024/03/06 20:56:48 site power: 254W
[lp-1  ] DEBUG 2024/03/06 20:56:48 charge voltages: [228 228 228]V
[lp-1  ] DEBUG 2024/03/06 20:56:48 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/06 20:56:48 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/06 20:56:48 charge total import: 4436.000kWh
[lp-1  ] DEBUG 2024/03/06 20:56:48 charger status: B
[lp-1  ] DEBUG 2024/03/06 20:56:48 next soc poll remaining time: 4m20s
[lp-1  ] DEBUG 2024/03/06 20:56:48 limitSoc reached: 90.0% > 90%
[site  ] DEBUG 2024/03/06 20:56:58 ----
[lp-1  ] DEBUG 2024/03/06 20:56:58 charge power: 0W
[site  ] DEBUG 2024/03/06 20:56:58 pv power: 0W
[site  ] DEBUG 2024/03/06 20:56:58 grid meter: 254W
[site  ] DEBUG 2024/03/06 20:56:58 grid powers: [21 218 14]W
[site  ] DEBUG 2024/03/06 20:56:58 grid currents: [0.452 1.22 0]A
[site  ] DEBUG 2024/03/06 20:56:58 site power: 254W
[lp-1  ] DEBUG 2024/03/06 20:56:58 charge voltages: [228 228 228]V
[lp-1  ] DEBUG 2024/03/06 20:56:58 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/06 20:56:58 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/06 20:56:58 charge total import: 4436.000kWh
[lp-1  ] DEBUG 2024/03/06 20:56:58 charger status: B
[lp-1  ] DEBUG 2024/03/06 20:56:58 next soc poll remaining time: 4m10s
[lp-1  ] DEBUG 2024/03/06 20:56:58 limitSoc reached: 90.0% > 90%
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.124.7

123aerox, 2024-03-08:
Ich benutze den Microsoft Edge. Nein nichts dazwischen. Direkte Kommunikation mit dem offiziellen Dockercontainer.

naltatis, 2024-03-09:
Ursache ist hier, dass du Leerzeichen in den `name` Feldern verwendest. Das sind eigentlich technische Schlüssel (werden nie angezeigt) in denen wir nur begrenzt Sonderzeichen erwarten. Da es aber (bis auf den Config UI Fall den du beschreibst) sonst scheinbar aktuell überall funktioniert hab ich in dem PR (oben) Unterstützung dafür eingebaut.

123aerox, 2024-03-09:
Danke, da habe ich dummerweise nicht drauf geachtet. Ja, sonnst hat es immer tadellos geklappt deshalb ist mir das noch nicht aufgefallen 😂👍

andig, 2024-03-11:
Closed in https://github.com/evcc-io/evcc/commit/042a472aa9044bd036fc066dca361794e50901c6