evcc-io/evcc #12728: OCCP - ABB TerraAC - 1.24.7 - Laden startet nicht wieder nach pausieren - (evtl. #12459)

sreichholf, 2024-03-07:
### Describe the bug

Wenn ich das laden mit 1.24.7 anhalte und danach wieder starten möchte - z.B. durch einen wechsel von "Aus" auf "Schnell" -  kommt mit meiner TerraAC kein SetChargingProfile mehr (nie wieder egal was ich manuell als Lademodus einstelle).

Meine TerraAC läuft bereits auf Firmware 1.8.0, das war mit der .6 aber noch alles i.O.

### Steps to reproduce

1. Ladevorgang z.B. durch auswahl von "Schnell" starten
2. Ladevorgang z.B. durchauswahl von "Aus" oder auch "PV" (bei zu wenig PV Leistung) anhalten
3. Ladevorgang erneut starten durch auswahl von z.B. "Schnell".

### Configuration details

```yaml
ABB TerraAC über ocpp

`chargers:
  - name: TerraAC
    type: ocpp
    connecttimeout: 30m
    timeout: 20m
    getconfiguration: true
    bootnotification: true`
```

### Log details

```text
`[site  ] DEBUG 2024/03/07 07:26:00 site power: -166W
[lp-1  ] DEBUG 2024/03/07 07:26:00 charge total import: 0.000kWh
[lp-1  ] DEBUG 2024/03/07 07:26:00 charger status: C
[lp-1  ] DEBUG 2024/03/07 07:26:00 charge start energy: 0kWh
[lp-1  ] INFO 2024/03/07 07:26:00 start charging ->
[lp-1  ] DEBUG 2024/03/07 07:26:00 wake-up timer: stop
[lp-1  ] DEBUG 2024/03/07 07:26:02 soc estimated: 57.00% (vehicle: 57.00%)
[lp-1  ] DEBUG 2024/03/07 07:26:02 vehicle soc: 57%
[lp-1  ] DEBUG 2024/03/07 07:26:03 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/07 07:26:03 vehicle range: 207km
[lp-1  ] DEBUG 2024/03/07 07:26:03 pv charge current: 6.24A = 6A + 0.241A (-166W @ 3p)
[ocpp  ] TRACE 2024/03/07 07:26:03 sent JSON message to TACW************: [2,"1298119047","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":2,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6.2}]}}}]
[ocpp  ] TRACE 2024/03/07 07:26:03 received JSON message from TACW************: [3, "1298119047", {"status": "Accepted"}]
[lp-1  ] DEBUG 2024/03/07 07:26:03 max charge current: 6.24A
[ocpp  ] TRACE 2024/03/07 07:26:16 received JSON message from TACW************: [2, "2579192", "MeterValues", {"connectorId": 1, "transactionId": 2, "meterValue": [{"timestamp": "2024-03-07T06:26:17.000Z", "sampledValue": [{"value": "224.40", "context": "Sample.Periodic", "format": "Raw", "measurand": "Voltage", "phase": "L1-N", "unit": "V"}, {"value": "6.05", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "phase": "L1", "unit": "A"}, {"value": "1342", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "phase": "L1", "unit": "W"}, {"value": "224.20", "context": "Sample.Periodic", "format": "Raw", "measurand": "Voltage", "phase": "L2-N", "unit": "V"}, {"value": "6.00", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "phase": "L2", "unit": "A"}, {"value": "1331", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "phase": "L2", "unit": "W"}, {"value": "224.10", "context": "Sample.Periodic", "format": "Raw", "measurand": "Voltage", "phase": "L3-N", "unit": "V"}, {"value": "6.02", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "phase": "L3", "unit": "A"}, {"value": "1334", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "phase": "L3", "unit": "W"}, {"value": "43", "context": "Sample.Periodic", "format": "Raw", "measurand": "Energy.Active.Import.Register", "unit": "Wh"}, {"value": "6", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Offered", "unit": "A"}, {"value": "18.07", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "unit": "A"}, {"value": "4007", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "unit": "W"}]}]}]
[ocpp  ] TRACE 2024/03/07 07:26:16 sent JSON message to TACW************: [3,"2579192",{}]
[site  ] DEBUG 2024/03/07 07:26:30 ----
[lp-1  ] DEBUG 2024/03/07 07:26:30 charge power: 4007W
[site  ] DEBUG 2024/03/07 07:26:30 pv power: 166W
[site  ] DEBUG 2024/03/07 07:26:30 site power: 3841W
[lp-1  ] DEBUG 2024/03/07 07:26:30 charge total import: 0.043kWh
[lp-1  ] DEBUG 2024/03/07 07:26:30 charger status: C
[lp-1  ] DEBUG 2024/03/07 07:26:30 soc estimated: 57.05% (vehicle: 57.00%)
[lp-1  ] DEBUG 2024/03/07 07:26:30 vehicle soc: 57%
[lp-1  ] DEBUG 2024/03/07 07:26:30 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/07 07:26:30 vehicle range: 207km
[lp-1  ] DEBUG 2024/03/07 07:26:30 pv charge current: 0.674A = 6.24A + -5.57A (3841W @ 3p)
[ocpp  ] TRACE 2024/03/07 07:26:30 sent JSON message to TACW************: [2,"41708667","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":2,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/03/07 07:26:30 received JSON message from TACW************: [3, "41708667", {"status": "Accepted"}]
[lp-1  ] DEBUG 2024/03/07 07:26:30 max charge current: 6A
[ocpp  ] TRACE 2024/03/07 07:26:53 received JSON message from TACW************: [2, "6466982", "MeterValues", {"connectorId": 1, "transactionId": 2, "meterValue": [{"timestamp": "2024-03-07T06:26:54.000Z", "sampledValue": [{"value": "224.80", "context": "Sample.Periodic", "format": "Raw", "measurand": "Voltage", "phase": "L1-N", "unit": "V"}, {"value": "6.05", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "phase": "L1", "unit": "A"}, {"value": "1343", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "phase": "L1", "unit": "W"}, {"value": "224.60", "context": "Sample.Periodic", "format": "Raw", "measurand": "Voltage", "phase": "L2-N", "unit": "V"}, {"value": "6.00", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "phase": "L2", "unit": "A"}, {"value": "1333", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "phase": "L2", "unit": "W"}, {"value": "224.50", "context": "Sample.Periodic", "format": "Raw", "measurand": "Voltage", "phase": "L3-N", "unit": "V"}, {"value": "6.02", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "phase": "L3", "unit": "A"}, {"value": "1335", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "phase": "L3", "unit": "W"}, {"value": "84", "context": "Sample.Periodic", "format": "Raw", "measurand": "Energy.Active.Import.Register", "unit": "Wh"}, {"value": "6", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Offered", "unit": "A"}, {"value": "18.07", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "unit": "A"}, {"value": "4011", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "unit": "W"}]}]}]
[ocpp  ] TRACE 2024/03/07 07:26:53 sent JSON message to TACW************: [3,"6466982",{}]
[lp-1  ] DEBUG 2024/03/07 07:26:55 set charge mode: off
[site  ] DEBUG 2024/03/07 07:26:55 ----
[lp-1  ] DEBUG 2024/03/07 07:26:55 charge power: 4011W
[site  ] DEBUG 2024/03/07 07:26:55 pv power: 166W
[site  ] DEBUG 2024/03/07 07:26:55 site power: 3845W
[lp-1  ] DEBUG 2024/03/07 07:26:55 charge total import: 0.084kWh
[lp-1  ] DEBUG 2024/03/07 07:26:55 charger status: C
[lp-1  ] DEBUG 2024/03/07 07:26:55 soc estimated: 57.09% (vehicle: 57.00%)
[lp-1  ] DEBUG 2024/03/07 07:26:55 vehicle soc: 57%
[... 43 lines cut ...]
[lp-1  ] DEBUG 2024/03/07 07:28:01 charger status: B
[lp-1  ] DEBUG 2024/03/07 07:28:01 max charge current: 16A
[lp-1  ] DEBUG 2024/03/07 07:28:01 charger enable
[lp-1  ] DEBUG 2024/03/07 07:28:01 wake-up timer: start
[ocpp  ] TRACE 2024/03/07 07:28:06 received JSON message from TACW************: [2, "9623474", "MeterValues", {"connectorId": 1, "transactionId": 2, "meterValue": [{"timestamp": "2024-03-07T06:28:07.000Z", "sampledValue": [{"value": "226.20", "context": "Sample.Periodic", "format": "Raw", "measurand": "Voltage", "phase": "L1-N", "unit": "V"}, {"value": "0.00", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "phase": "L1", "unit": "A"}, {"value": "0", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "phase": "L1", "unit": "W"}, {"value": "225.70", "context": "Sample.Periodic", "format": "Raw", "measurand": "Voltage", "phase": " [... cut ...] ext": "Sample.Periodic", "format": "Raw", "measurand": "Current.Offered", "unit": "A"}, {"value": "0.00", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "unit": "A"}, {"value": "0", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "unit": "W"}]}]}]
[ocpp  ] TRACE 2024/03/07 07:28:43 sent JSON message to TACW************: [3,"8452964",{}]`
```

### What type of operating system are you running?

Docker container

### Version

1.24.7

andig, 2024-03-10:
Vmtl. ein Seiteneffekt von https://github.com/evcc-io/evcc/pull/12459

andig, 2024-03-10:
Siehe auch https://github.com/evcc-io/evcc/issues/12821. Könntest Du bitte auch nochmal verifizieren, ob 0.124.6 noch funktioniert?

/cc @cheerfulalex ähnlicher Nebeneffekt wie Du ihn auch schon beobachtet hattest.

sreichholf, 2024-03-11:
Funktioniert einwandfrei, darauf bin ich direkt zurück. Jaz gestern stressfrei geladen.

andig, 2024-03-11:
Closed in https://github.com/evcc-io/evcc/commit/1580857163d006d744319aaf79570175161ecb50

cheerfulalex, 2024-03-11:
Ich glaube hier wird etwas vermischt: 

solange man im "pv"-Modus bleibt und nichts über Weboberfläche umschaltet tut SetChargingProfile = 0A für das Pausieren wie gewünscht. Auto anschließen, abziehen -> alles wunderbar. 

Wenn ich jetzt aber über die Weboberfläche von "pv" -> "aus" -> "schnell" umschaltete wird kein seites EVCC kein SetCargingProifle mit 16A geschickt. Genau genommen es wird gar nichts geschickt. Und das ist der eigentlicher Fehler. 

Ich schulde nocht die Log dazu, aber bei mir sieht es mit https://github.com/evcc-io/evcc/pull/12459 genauso aus.

sreichholf, 2024-03-11:
Da ich an besagtem Tag keine Sonne hatte, kann es gut sein, dass es im PV Modus einwandfrei funktioniert hätte.
Ansonsten deckt sich mein Log ja mit deinen Ausführungen, dass schlicht nichts gesendet wird.

premultiply, 2024-03-11:
> Wenn ich jetzt aber über die Weboberfläche von "pv" -> "aus" -> "schnell" umschaltete wird seites EVCC kein SetCargingProifle mit 16A geschickt. Genau genommen es wird gar nichts geschickt. Und das ist der eigentliche Fehler.

@andig Das sollten wir dann aber vor einem Revert zuerst checken und ggf. fixen.

andig, 2024-03-11:
Das eigentliche Problem ist https://github.com/evcc-io/evcc/issues/12821. Und auch hier steckt ein Bug drin der gelöst werden müsste. Also erstmal revert.

premultiply, 2024-07-14:
Das Problem war hier nur, dass evcc nach

    [ocpp  ] TRACE 2024/03/07 07:26:55 sent JSON message to TACW************: [2,"3510183110","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":2,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]

kein `SetChargingProfile` mehr gesendet hat um die Ladefreigabe auch wieder zu erteilen.

Behoben in https://github.com/evcc-io/evcc/pull/14868

andig, 2024-07-24:
Danke @premultiply den hatte ich vom Radar verloren.

closed 2024-07-24 by andig