evcc-io/evcc #12729: MQTT Climater activation ignored

betaphi, 2024-03-07:
### Describe the bug

I have configured my Tesla Model Y manually via MQTT/Teslamate as this gives me the possibility to add support for `climater`, which is currently not supported by the Teslamate template.

Unfortunately when the climate is activated, evcc does not activate the load point. Of course I have verified that the climate status is correctly published via the correct MQTT topic.

Also, `evcc vehicle` correctly reflects the status as `Climate Active: true`.
Other similar issues have pointed to some 'polling' values not being correct, which could solve those issues. However, I would believe that in the case of using MQTT 'polling' should not be the source of the issue, or is it?

Output of `evcc vehicle` while the car was connected and the climate was on:
```
[main  ] INFO 2024/03/07 07:18:52 evcc 0.124.6
[main  ] INFO 2024/03/07 07:18:52 using config file: /etc/evcc.yaml
[db    ] INFO 2024/03/07 07:18:52 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/03/07 07:18:52 connecting evcc-1018362524 at tcp://homeassistant:1883
[mqtt  ] DEBUG 2024/03/07 07:18:53 tcp://homeassistant:1883 connected
Soc:            98%
Capacity:       60.0kWh
Charge status:  A
Range:          403km
Climate active: true
OnIdentified:   Mode:pv
Features:       [CoarseCurrent]
```

Regarding the attached log output, the activation of the pre-heating was published via MQTT at **07:07:21** and the car was disconnected from the charger at **07:11:21**.

### Steps to reproduce

1. Configure `climater` via MQTT.
2. Activate climate while the EV is connected to the LP and charging is already finished.
3. See that the LP will not be enabled.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
    #  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

mqtt:
  broker: homeassistant:1883
  topic: evcc
  user: evcc
  password: #####

log: debug
levels:
  cache: error

# unique installation id
plant: #####

interval: 10s # control cycle interval

sponsortoken: #####

vehicles:
- name: ev1
  type: custom
  title: Model Y
  capacity: 60
  features: ["coarsecurrent"]
  onIdentify:
    mode: pv
      #    minSoc: 10
      #    targetSoc: 80
  soc:
[... 352 lines cut ...]
evcc-1  | [lp-1  ] DEBUG 2024/03/07 07:11:21 set charge mode: off
evcc-1  | [lp-1  ] DEBUG 2024/03/07 07:11:21 set charge mode: pv
evcc-1  | [lp-1  ] DEBUG 2024/03/07 07:11:21 set session soc limit: 0
evcc-1  | [lp-1  ] DEBUG 2024/03/07 07:11:21 set session energy limit: 0
evcc-1  | [lp-1  ] WARN 2024/03/07 07:11:21 charger out of sync: expected disabled, got enabled
evcc-1  | [lp-1  ] DEBUG 2024/03/07 07:11:21 charger disable
evcc-1  | [lp-1  ] DEBUG 2024/03/07 07:11:21 wake-up timer: stop
evcc-1  | [site  ] DEBUG 2024/03/07 07:11:21 ----
evcc-1  | [lp-1  ] DEBUG 2024/03/07 07:11:21 charge power: 0W
evcc-1  | [site  ] DEBUG 2024/03/07 07:11:21 grid meter: 1728W
evcc-1  | [site  ] DEBUG 2024/03/07 07:11:21 pv power: 0W
evcc-1  | [site  ] DEBUG 2024/03/07 07:11:21 site power: 1728W
evcc-1  | [lp-1  ] DEBUG 2024/03/07 07:11:21 charge currents: [0 0 0]A
evcc-1  | [lp-1  ] DEBUG 2024/03/07 07:11:21 charge total import: 219.126kWh
evcc-1  | [lp-1  ] DEBUG 2024/03/07 07:11:21 charger status: A
evcc-1  | [site  ] DEBUG 2024/03/07 07:11:31 ----
```

### What type of operating system are you running?

Linux

### Version

 0.124.6

andig, 2024-03-07:
Bitte evcc trace log wo climater active sichtbar ist.

betaphi, 2024-03-07:
> Bitte evcc trace log wo climater active sichtbar ist im Kontext nicht erfolgter Freigabe.

Siehe gewünschte Trace Log. Man sieht klar, dass das MQTT Topic empfangen wird. Eine Aktion folgt darauf nicht.

```
evcc-1  | [site  ] DEBUG 2024/03/07 17:48:18 ----
evcc-1  | [go-e  ] TRACE 2024/03/07 17:48:18 GET http://10.1.211.1/api/status?filter=alw,car,eto,nrg,wh,trx,cards
evcc-1  | [go-e  ] TRACE 2024/03/07 17:48:18 {"alw":false,"eto":219126,"car":3,"wh":0,"trx":null,"nrg":[234.9799957,233.1199951,234.9799957,2.480000019,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"n/a","energy":0,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false}]}
evcc-1  | [lp-1  ] DEBUG 2024/03/07 17:48:18 charge power: 0W
evcc-1  | [site  ] DEBUG 2024/03/07 17:48:18 grid meter: 1196W
evcc-1  | [site  ] DEBUG 2024/03/07 17:48:18 pv power: 0W
evcc-1  | [site  ] DEBUG 2024/03/07 17:48:18 site power: 1196W
evcc-1  | [lp-1  ] DEBUG 2024/03/07 17:48:18 charge currents: [0 0 0]A
evcc-1  | [lp-1  ] DEBUG 2024/03/07 17:48:18 charge total import: 219.126kWh
evcc-1  | [lp-1  ] DEBUG 2024/03/07 17:48:18 charger status: B
evcc-1  | [lp-1  ] DEBUG 2024/03/07 17:48:18 limitSoc reached: 75.0% > 70%
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:19 recv homeassistant/sensor/eg_og_power_consumption: '1194'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:19 recv homeassistant/sensor/eg_og_power_consumption: '1195'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:20 recv homeassistant/sensor/eg_og_power_consumption: '1197'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:21 recv homeassistant/sensor/eg_og_power_consumption: '1195'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:21 recv homeassistant/sensor/eg_og_power_consumption: '1193'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:22 recv homeassistant/sensor/eg_og_power_consumption: '1194'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:22 recv homeassistant/sensor/eg_og_power_consumption: '1192'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:22 recv homeassistant/sensor/eg_og_power_consumption: '1195'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:23 recv homeassistant/sensor/eg_og_power_consumption: '1192'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:23 recv homeassistant/sensor/eg_og_power_consumption: '1195'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:23 recv homeassistant/sensor/eg_og_power_consumption: '1194'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:24 recv homeassistant/sensor/eg_og_power_consumption: '1193'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:24 recv homeassistant/sensor/eg_og_power_consumption: '1194'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:24 recv homeassistant/sensor/eg_og_power_consumption: '1195'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:25 recv homeassistant/sensor/eg_og_power_consumption: '1194'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:26 recv homeassistant/sensor/eg_og_power_consumption: '1195'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:26 recv homeassistant/sensor/eg_og_power_consumption: '1196'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:26 recv homeassistant/sensor/eg_og_power_consumption: '1195'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:26 recv homeassistant/sensor/eg_og_power_consumption: '1193'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:27 recv homeassistant/sensor/eg_og_power_consumption: '1195'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:27 recv homeassistant/sensor/eg_og_power_consumption: '1196'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:27 recv homeassistant/sensor/eg_og_power_consumption: '1195'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:28 recv teslamate/cars/1/plugged_in: 'true'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:28 recv teslamate/cars/1/is_climate_on: 'true'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:28 recv teslamate/cars/1/rated_battery_range_km: '307.01'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:28 recv homeassistant/sensor/eg_og_power_consumption: '1197'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:28 recv homeassistant/sensor/eg_og_power_consumption: '1194'
evcc-1  | [site  ] DEBUG 2024/03/07 17:48:28 ----
evcc-1  | [go-e  ] TRACE 2024/03/07 17:48:28 GET http://10.1.211.1/api/status?filter=alw,car,eto,nrg,wh,trx,cards
evcc-1  | [go-e  ] TRACE 2024/03/07 17:48:28 {"alw":false,"eto":219126,"car":3,"wh":0,"trx":null,"nrg":[234.6699982,233.1199951,234.9799957,2.480000019,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"n/a","energy":0,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false}]}
evcc-1  | [lp-1  ] DEBUG 2024/03/07 17:48:28 charge power: 0W
evcc-1  | [site  ] DEBUG 2024/03/07 17:48:28 grid meter: 1194W
evcc-1  | [site  ] DEBUG 2024/03/07 17:48:28 pv power: 0W
evcc-1  | [site  ] DEBUG 2024/03/07 17:48:28 site power: 1194W
evcc-1  | [lp-1  ] DEBUG 2024/03/07 17:48:28 charge currents: [0 0 0]A
evcc-1  | [lp-1  ] DEBUG 2024/03/07 17:48:28 charge total import: 219.126kWh
evcc-1  | [lp-1  ] DEBUG 2024/03/07 17:48:28 charger status: B
[... cut ...]
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:49:06 recv homeassistant/sensor/eg_og_power_consumption: '1191'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:49:07 recv homeassistant/sensor/eg_og_power_consumption: '1189'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:49:07 recv homeassistant/sensor/eg_og_power_consumption: '1192'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:49:07 recv homeassistant/sensor/eg_og_power_consumption: '1191'
evcc-1  | [mqtt  ] TRACE 2024/03/07 17:49:07 recv homeassistant/sensor/eg_og_power_consumption: '1192'
evcc-1  | [site  ] DEBUG 2024/03/07 17:49:08 ----
```

betaphi, 2024-03-10:
@andig May I ask why the issue was closed immediately after opening it and even it remains closed after I have posted the requested log?

andig, 2024-03-10:
Closed since incomplete.

betaphi, 2024-03-10:
@andig What exactly is incomplete here? You have asked for a trace log, and I have provided it in my answer 3 days ago. What else do you need?

andig, 2024-03-10:
Trace log shows no climater afaikt. This fits with climater not even implemented in Teslamate template. Happy to take a PR for adding it.

betaphi, 2024-03-10:
@andig May I ask you to read the bug report carefully instead of discarding it over and over again? The first sentence of the bug report reads:

> I have configured my Tesla Model Y manually via MQTT/Teslamate as this gives me the possibility to add support for climater, which is currently not supported by the Teslamate template.

This means, I have of course configured EVCC to receive the relevant MQTT topic, which can also be verified in my configuration file, that I have attached.

Also, the provided trace log clearly shows that the topic is correctly received.

`evcc-1  | [mqtt  ] TRACE 2024/03/07 17:48:28 recv teslamate/cars/1/is_climate_on: 'true'`

Unfortunately there is no further reaction to this receiving of the topic. This is the whole point of filing the bug.

When looking at the code that I believe to be relevant to this, it seems that it is necessary to change the `soc polling mode` to `connected` in order for `climater` to work at all. Maybe you can confirm this. Especially when no polling is needed at all (for example in case MQTT is used), this is especially unintuitive and should at least get mentioned in the respective documentation.

andig, 2024-03-10:
> When looking at the code that I believe to be relevant to this, it seems that it is necessary to change the soc polling mode to connected in order for climater to work at all. 

See `vehicleClimatePollAllowed`