evcc-io/evcc #12738: No connection to Huawei SmartLogger after updating to 0.124.7

lokipo, 2024-03-07:
### Describe the bug

After updating from 0.124.4 to 0.124.7 I cannot connect to my Huawei SmartLogger any longer as I am getting a modbus Error along the lines of "pv 1 power: read failed: modbus: exception '3' (illegal data value), function '3'"

### Steps to reproduce

1. Update to 0.124.7
2. Start evcc
3. get error
...

### Configuration details

```yaml
meters:
- type: template
  template: huawei-smartlogger 
  usage: pv
  # Modbus TCP
  modbus: tcpip
  id: 1  
  host: xxx.xyz 
  port: 502  
  name: pv1
```

### Log details

```text
[site  ] DEBUG 2024/03/07 10:55:15 ----
[lp-1  ] DEBUG 2024/03/07 10:55:15 charge power: 0W
[lp-2  ] DEBUG 2024/03/07 10:55:15 charge power: 5920W
[lp-3  ] DEBUG 2024/03/07 10:55:15 charge power: 0W
[lp-4  ] DEBUG 2024/03/07 10:55:15 charge power: 0W
[site  ] ERROR 2024/03/07 10:55:16 pv 1 power: read failed: modbus: exception '3' (illegal data value), function '3'
[site  ] DEBUG 2024/03/07 10:55:16 pv power: 0W
[site  ] DEBUG 2024/03/07 10:55:16 site power: 5920W
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.7

andig, 2024-03-07:
Bitte `modbus: trace` für letzte funktionierende und aktuelle Version im Vergleich.

lokipo, 2024-03-07:
0.124.7 (broken):

[modbus] TRACE 2024/03/07 14:13:21 modbus: send 00 03 00 00 00 06 01 03 7d 40 00 02
[modbus] TRACE 2024/03/07 14:13:21 modbus: recv 00 03 00 00 00 03 01 83 03
[site  ] ERROR 2024/03/07 14:13:21 pv 1 power: read failed: modbus: exception '3' (illegal data value), function '3'

0.124.4 (working):

[modbus] TRACE 2024/03/07 14:15:23 modbus: send 00 03 00 00 00 06 00 03 9e 4d 00 02
[modbus] TRACE 2024/03/07 14:15:23 modbus: recv 00 03 00 00 00 07 00 03 04 00 00 75 83
[modbus] TRACE 2024/03/07 14:15:23 modbus: send 00 04 00 00 00 06 00 03 9e 70 00 02
[modbus] TRACE 2024/03/07 14:15:23 modbus: recv 00 04 00 00 00 07 00 03 04 00 0d f5 5a
[site  ] DEBUG 2024/03/07 14:15:23 pv power: 30083W
[site  ] DEBUG 2024/03/07 14:15:23 site power: -30083W

RTTTC, 2024-03-07:
Could be the reason https://github.com/evcc-io/evcc/pull/12628

andig, 2024-03-07:
/cc @cmusik kannst Du was dazu sagen? Muss Dein PR reverted werden?

lokipo, 2024-03-07:
Changed the modbus-id to 11:

[modbus] TRACE 2024/03/07 16:00:07 modbus: send 00 02 00 00 00 06 0b 03 7d 40 00 02
[modbus] TRACE 2024/03/07 16:00:07 modbus: recv 00 02 00 00 00 03 0b 83 03
[site  ] ERROR 2024/03/07 16:00:07 pv 1 power: read failed: modbus: exception '3' (illegal data value), function '3'

cmusik, 2024-03-07:
Ok, then it looks like your powermeter has a different id. I thought that 11 is the default value, but it is possible to configure that. Please check in the smartlogger in the deployment wizard -> power meter which id is set for your power meter. If that is not working, then it would be helpful to know your setup.

cmusik, 2024-03-07:
@andig Es wäre gut das besser in der Dokumentation kenntlich zu machen. Hätte ich eigentlich auch schon vorher machen sollen, aber nun ist das Kind in den Brunnen gefallen. Kannst du mir einen Tipp geben an welcher Stelle ich das machen müsste um mehr Text in die Dokumentation zu bekommen?

lokipo, 2024-03-07:
Hi,

my meter seems to be running on "COM2-11", see attached screenshot.
[image]

cmusik, 2024-03-07:
That is identical to my setup. Which version of the smartlogger software are you running?

lokipo, 2024-03-07:
I am running V300R001C00SPC030

cmusik, 2024-03-07:
hmm, I think that is quite old. Are you able to update it to a newer version? I'm running the latest V300R023C10SPC311. Before that I was running V300R023C10SPC170 where this also worked. But maybe we will need to bring back the old method for older versions.

lokipo, 2024-03-07:
Updated to V300R023C10SPC315:

[modbus] TRACE 2024/03/07 21:16:54 modbus: send 00 02 00 00 00 06 0b 03 7d 40 00 02
[modbus] TRACE 2024/03/07 21:16:54 modbus: recv 00 02 00 00 00 03 0b 83 03
[site  ] ERROR 2024/03/07 21:16:54 pv 1 power: read failed: modbus: exception '3' (illegal data value), function '3'

cmusik, 2024-03-07:
Which power sensor are you using?

cmusik, 2024-03-08:
I think I have overseen that you have `usage: pv` in your configuration. In the old configuration there was no use of this parameter and so it was just ignored. Now it tries to get all the values from the inverters, power meters and batteries. So please update your configuration to `usage: grid` and extend the other parts.

lokipo, 2024-03-08:
Hi,

I changed the config to:

meters:
- type: template
  template: huawei-smartlogger
  usage: grid
  modbus: tcpip
  id: 11
  host: xxx.yyy.de
  port: 502
  name: pv1

Which is now working. What I do not understand is the meaning of "grid". I thought that I had to use "pv" in order to tell evcc that is meter is measuring what my solar-array is producing and that I will add another meter (2 Fronius-meters in fact, one measuring the internal usage and one measuring the grid-power going in or solar-power going out) later (still waiting on parts) to be able to measure if there is any spare power.

andig, 2024-03-08:
@cmusik I'm a litte confused. Is this a problem with the updated template?

cmusik, 2024-03-08:
@andig I would say not really. It changed and this caused the problem. The smartlogger has metrics for different devices. In the old template we only queried the data from the logger itself which are accumulated from all devices. But there you have only a few values. To get other values (e.g. battery SOC) you need to query a different unit id. And this is something that everyone has to check self as the unit ids are not static. You can change them and also they are assigned in different orders. But this also applies to also to the dongle configuration.

@lokipo you need different usage types to get all data. In the old template only the accumulated values from the smartlogger were used and they are not complete for all parts. A whole configuration needs grid and pv and optionally the battery if it is present.

lokipo, 2024-03-08:
@cmusik thank you for the clarification. So I need to read "grid" for checking what exactly? And how do I read "pv" as this is not yet working which is why I changed to "grid".

cmusik, 2024-03-08:
Here is an example of my configuration:

```
meters:
- name: grid1
  type: template
  template: huawei-smartlogger
  usage: grid
  id: 11
  host: smartlogger.fritz.box
  port: 502

- name: battery
  type: template
  template: huawei-smartlogger
  usage: battery
  id: 3
  host: smartlogger.fritz.box
  port: 502
  capacity: 10

- name: pv_10ktl
  type: template
  template: huawei-smartlogger
  usage: pv
  id: 3
  host: smartlogger.fritz.box
  port: 502
  capacity: 10

site:
  title: mysite
  meters:
    grid: grid1
    pv:
      - pv_10ktl
    battery:
      - battery
  residualPower: 100
```

The unit id will be probably different then mine, so you have to check your smartlogger. Hope this helps!

lokipo, 2024-03-08:
@cmusik 

That explains a lot, and clears my issue, thank you.
I know have a working config using both my grid-meter and both my inverters.

cmusik, 2024-03-08:
Normally I would say that there is one grid and multiple inverters. So please try to set the 33KTL to `usage: pv`.